AravisProject/aravis #505: Selecting network interface to be sued by Aravis

Louis-Demers, 2021-04-30:
Running on macOS, when I work with GigE cameras, I always make sure that I’m connected with ethernet, but often, I forget to disconnect my WIFI interface. Despite my LAN being higher priority than the WIFI, invariably, I get timeouts and the camera stops working. My only option, is to disconnect WIFI, and restart the application. It would be more robust/reliable if there was a way to steer the communications through the LAN interface. Flir Spinaker SDK seems to allow this by allowing to select a specific network interface to use to connect to the camera.

EmmanuelP, 2021-04-30:
Thanks for the report.

There is 2 things we can do:

- if Aravis receives multiple discovery answers for the same device, it should prefer wired interfaces
- we should provide an API to force the interface like FLIR spinaker SDK

What is the API for this setting in Spinaker ?

justneedanacc, 2021-06-02:
> There is 2 things we can do:
> 
>     * if Aravis receives multiple discovery answers for the same device, it should prefer wired interfaces
>     * we should provide an API to force the interface like FLIR spinaker SDK

Being able to specify the network interfaces would be great. 
In `arv_gv_discover_socket_list_send_discover_packet` there is the line `broadcast_address = g_inet_address_new_from_string ("255.255.255.255");`
You don't know over what interface this gets send out. Under Windows its based on interface metrics and under Linux its also going to get send out from one interface and not all. Using a direct broadcast like 192.168.1.255 (for a 192.168.1.0/24 net) would be better.

My system has four 10Gbit interfaces and two 1Gbit interfaces. All of those interfaces have Network cameras connected. In adition the system has other interfaces without cameras. Being able to specify interfaces in Aravis would be really great in this situation.

EmmanuelP, 2021-06-02:
> In arv_gv_discover_socket_list_send_discover_packet there is the line broadcast_address = g_inet_address_new_from_string ("255.255.255.255");
> You don't know over what interface this gets send out. Under Windows its based on interface metrics and under Linux its also going to get send out from one interface and not all. Using a direct broadcast like 192.168.1.255 (for a 192.168.1.0/24 net) would be better. Using 255.255.255.255 is really hard to predict in multi interface setups.

Aravis explicitely sends the discovery packet to every network interfaces by specifying their address.

https://github.com/AravisProject/aravis/blob/02f21f6327f5cfb2f3df85c81084c4711ec56d5a/src/arvgvinterface.c#L171

So far it seems to work fine. We could use the broadcast address we already store in ArvNetworkInterface:

https://github.com/AravisProject/aravis/blob/02f21f6327f5cfb2f3df85c81084c4711ec56d5a/src/arvnetworkprivate.h#L92

Pull request appreciated...

EmmanuelP, 2021-06-09:
I have forgotten to mention the fact it is possible to instantiate a camera by using its IP or MAC address. That may help to use the device using the right network.

jsarrett, 2023-09-28:
I am having a related issue, that would be solved if this bug were fixed.  However @EmmanuelP  I don't think that both of the proposed fixes are equal.  In general, there's no easy way to tell if an interface is wired or wireless.  Every method is platform specific.

Binding Aravis to a specific IP address however is a very useful feature.  It's orthogonal to weather you specify a camera IP or not.  A user might want to only discover cameras on a specific subnet.

In general hosts only have one interface per broadcast domain, which is why the discover mechanism usually avoids this.  It's not required though, some hosts will have two or more interfaces in the same IP subnet. So this is still a real live bug/feature request.

commit 775ef23c (2025-02-12): gv: discovery interface selection

Fixes #505
commit 2c5730cd (2025-02-12): gv: discovery interface selection

Fixes #505
commit 3a214b89 (2025-02-12): gv: discovery interface selection

Fixes #505
closed 2025-02-12 by EmmanuelP