AravisProject/aravis #506: Failed to enable Auto Exposure ROI on FLIR Blackfly 162S2C

lherman-cs, 2021-05-10:
**Describe the bug**

Aravis GStreamer plugin doesn't accept the camera's auto exposure ROI settings. 

I was trying to set FLIR-BFS-U3-16S2C-CS auto exposure ROI from the GStreamer plugin:

```sh
gst-launch-1.0 aravissrc camera-name=FLIR-0140D428 offset-y=150 features="AutoAlgorithmSelector=Ae AutoExposureRoiEnable_Val=true AutoExposureRoiWidth_Val=228" ! video/x-raw, width=1440, height=810, format=UYVY, framerate=30/1 ! videoconvert ! fpsdisplaysink
```

Aravis seems to receive the parameters properly, but it errors out with a write error to the camera. Following are the logs from "all:3":

[log.txt](https://github.com/AravisProject/aravis/files/6449545/log.txt)

Also, `AutoExposureRoiHeight_Val` seems to be an invalid key, whereas `AutoExposureRoiWidth_Val` exists. Is this expected? Is there a tool to list available nodes? 

**To Reproduce**
```sh
gst-launch-1.0 aravissrc camera-name=FLIR-0140D428 offset-y=150 features="AutoAlgorithmSelector=Ae AutoExposureRoiEnable_Val=true AutoExposureRoiWidth_Val=228" ! video/x-raw, width=1440, height=810, format=UYVY, framerate=30/1 ! videoconvert ! fpsdisplaysink
```

**Expected behavior**
Expected the camera to be configured properly.

**Camera description:**
 - Manufacturer: FLIR
 - Model: BFS-U3-16S2C-CS (Blackfly)
 - Interface: USB-3 

**Platform description:**
 - Aravis version: ARAVIS_0_8_5
 - OS: Ubuntu 20.04
 - Hardware: x86_64

EmmanuelP, 2021-05-10:
> Also, AutoExposureRoiHeight_Val seems to be an invalid key, whereas AutoExposureRoiWidth_Val exists. Is this expected? Is there a tool to list available nodes?

Yes, arv-tool-0.8

Please try to set auto exposure with it.

The log file content does not look like a regular debug output/

I'm closing this issue. Please reopen if you think there is still something to fix in Aravis.

lherman-cs, 2021-05-10:
My bad, I uploaded the wrong log. Below is the correct one:

[out.log](https://github.com/AravisProject/aravis/files/6449828/out.log)

lherman-cs, 2021-05-10:
@EmmanuelP would you mind reopening the issue? I don't have access to reopen. Thanks

EmmanuelP, 2021-05-10:
I don't see any error or failure in the log.

Again, please use arv-tool first.

lherman-cs, 2021-05-10:
@EmmanuelP toward the end, the camera responded with:

```
[GcRegisterNode::_set_integer_value] address = 0xc3424, value = 0x168
packet_type  = cmd
command      = write-memory-cmd
size         = 12
id           = 32
address      = 0x00000000000c3424
00000000 55 33 56 43 00 40 02 08 0c 00 20 00 24 34 0c 00  U3VC.@.... .$4..
00000010 00 00 00 00 68 01 00 00                          ....h...        
packet_type  = (null)
command      = write-memory-ack
size         = 4
id           = 32
written      =          0 (0x00000000)
00000000 55 33 56 43 06 80 03 08 04 00 20 00 00 00 00 00  U3VC...... .....

[[UvDevice::write_memory] Unexpected answer (0x8006) <-- this error
```

lherman-cs, 2021-05-10:
I tried using`arv-tool`, however the configurations are not reflected on Spinview. Is this expected? Or aravis has a different preset from Spinview? If it has a different preset, how can we confirm if the new settings are applied properly?

arv-tool command:

```
./arv-tool-0.8 control AutoAlgorithmSelector=Ae AasRoiEnable=true AasRoiWidth=300 AasRoiHeight=100 AasRoiOffsetX=300 AasRoiOffsetY=40
```

EmmanuelP, 2021-05-10:
The command you have written above should output the actual feature value. What was the output of your command ?