AravisProject/aravis #509: arv-tool-0.6 cannot get device from docker container

hetao-git, 2021-05-16:
**Is your feature request related to a problem? Please describe.**
I have been using arv-tool-0.6 to detect Basler camera for long time and it has been working fine in Ubuntu. I recently moved my application to docker container, but when I ran arv-tool-0.6 inside the container, it couldn't detect the camera. When I ran the same command on the host, it could detect the camera. 

**Describe the solution you'd like**
I would like to use arv-tool-0.6 to detect cameras from the docker container.
I think we probably do 2 things: 1)Make some network configuration change on the host? 2)Install additional tools to the docker image?

**Describe alternatives you've considered*.
Maybe I can try to run shell script to host from docker container, but it doesn't sound a clean solution.

**Additional context**
All bin files and lib files have been included to the same aravis folder and it was working on any Ubuntu host.

EmmanuelP, 2021-05-17:
Hi,

Thanks for the report.

I guess Aravis could be modified to restrict the ports it uses to a specified range. But this is a really low priority task, as it is already possible to configure the container to make Aravis work (--network=host for example).

Unless someone proposes a merge request for that with an acceptable level of complexity, it will likely not happen.

Cheers.

hetao-git, 2021-05-18:
Thanks for the comment EmmanueIP!

Just would like to give a little bit more update. I could ping the camera from the docker container, so that means the container can access the network, but the Aravis couldn't?

1)When you said: I guess Aravis could be modified to restrict the ports it uses to a specified range.
You mean we can modify aravis source code to specify a port range. But how will this port range help discover the camera? Also we are using Aravis to process videos, not just discover cameras. 

2)When you say configure the container to make Aravis work (--network=host), you mean you can specify --network=host when we start the docker container? such as: docker run --rm -d --network host --name my_nginx nginx. Our team actually doesn't recommend to start the docker in a different way. But I am not sure if there is other way around.

EmmanuelP, 2021-05-18:
> 1)When you said: I guess Aravis could be modified to restrict the ports it uses to a specified range.
You mean we can modify aravis source code to specify a port range. But how will this port range help discover the camera? Also we are using Aravis to process videos, not just discover cameras.

Aravis finds cameras by broadcasting a UDP discovery packet, and then listening to the discovery acknowledges.

You can also access a camera directly by using its IPV4 address as the camera name.

> 2)When you say configure the container to make Aravis work (--network=host), you mean you can specify --network=host when we start the docker container? such as: docker run --rm -d --network host [...]

I'm not a docker expert. I thought using --network=host will solve your issue, but if it is not the case that means there is another blocking point.

hetao-git, 2021-05-18:
How can we access the camera using IP address? Right now we are running the command like: arv-tool-0.6 -n Basler-12345678 control BalanceWhiteAuto=Off ....(and some other parameters). But it returned "No device found", and it was supposed to return something like:
Basler-12345678 (192.10.0.1)
BalanceWhiteAuto = Off
........

EmmanuelP, 2021-05-18:
Try: `arv-tool-0.6 --name=10.11.1.19`

If it doesn't work, please attach the output of: `arv-tool-0.6 --name=10.11.1.19 --debug=all:3`

hetao-git, 2021-05-18:
arv-tool-0.6 --name=10.11.1.19 didn't work on container, and host either. The output was:

[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 240.10.9.9
[GvDiscoverSocket::new] Add interface 240.10.10.2
No device found

EmmanuelP, 2021-05-18:
I'm not completely sure camera instantiation by ip address is working with aravis 0.6.

Could you try Aravis 0.8 ?

hetao-git, 2021-05-18:
I don't have 0.8 right now. But when I ran the same command on host, I saw the camera was found

EmmanuelP, 2021-05-18:
So, I can not help further. It is a container configuration issue.

hetao-git, 2021-05-18:
Yeah,  I will see how I can change network configurations.
Thanks for your help!

hetao-git, 2021-05-18:
Actually, maybe UDP is the issue? Our container cannot use UDP?

deadprogram, 2021-06-03:
For anyone who might be struggling with running Aravis in a container, you need to give the container access to the host network, as well as to provide setcap so the higher speed streaming will work.

We use something like this:

```
docker run --network host --privileged -it your-container-name:latest
```

EmmanuelP, 2021-07-05:
Not an aravis issue, I'm closing this report.

Thanks.

closed 2021-07-05 by EmmanuelP