AravisProject/aravis #511: Failure to get list of Pleora devices

Louis-Demers, 2021-06-07:
**Describe the bug**

		arv_update_device_list () and  arv_get_n_devices () fila to enumerate all Pleora iPort devices

**To Reproduce**
 on my network, I have numerous Flir and Pleora devices, `arv-tool-0.8 -d all` produces the following output
```
[15:06:37.364] 🅸 interface> Found 0 USB3Vision device (among 2 USB devices)
[15:06:37.364] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1
[15:06:37.365] 🅸 interface> [GvDiscoverSocket::new] Add interface 10.0.0.6
[15:06:37.365] 🆆 interface> [ArvGVInterface::send_discover_packet] Error: Error sending message: Can't assign requested address
[15:06:37.366] 🅸 interface> [GvInterface::discovery] Device 'Point Grey Research-Blackfly BFLY-PGE-31S4M-19274080' found (interface 10.0.0.6) user_id '' - MAC '00:b0:9d:26:19:60'
[15:06:37.368] 🅸 interface> [GvInterface::discovery] Device 'FLIR-Blackfly S BFS-PGE-31S4M-19263386' found (interface 10.0.0.6) user_id '' - MAC '2c:dd:a3:25:ef:9a'
[15:06:37.368] 🅸 interface> [GvInterface::discovery] Device 'FLIR-Blackfly S BFS-PGE-31S4M-19263381' found (interface 10.0.0.6) user_id '' - MAC '2c:dd:a3:25:ef:95'
[15:06:37.368] 🅸 interface> [GvInterface::discovery] Device 'FLIR-Blackfly S BFS-PGE-31S4M-19050440' found (interface 10.0.0.6) user_id '' - MAC '2c:dd:a3:22:af:c8'
[15:06:37.385] 🅸 interface> [GvInterface::discovery] Device 'Pleora Technologies Inc.-Analog Pro PT01-AN1IP0x-32yG-' found (interface 10.0.0.6) user_id '' - MAC '00:11:1c:04:58:7f'
[15:06:37.385] 🅸 interface> [GvInterface::discovery] Device 'Pleora Technologies Inc.-Analog Pro PT01-AN1IP0x-32yG-' found (interface 10.0.0.6) user_id '' - MAC '00:11:1c:04:65:56'
[15:06:37.388] 🅸 interface> [GvInterface::discovery] Device 'Pleora Technologies Inc.-Analog Pro PT01-AN1IP0x-32yG-' found (interface 10.0.0.6) user_id '' - MAC '00:11:1c:04:65:57'
FLIR-Blackfly S BFS-PGE-31S4M-19050440 (10.0.2.53)
FLIR-Blackfly S BFS-PGE-31S4M-19263381 (10.0.2.52)
FLIR-Blackfly S BFS-PGE-31S4M-19263386 (10.0.2.54)
Pleora Technologies Inc.-Analog Pro PT01-AN1IP0x-32yG- (10.0.2.61)
Point Grey Research-Blackfly BFLY-PGE-31S4M-19274080 (10.0.2.55)
```
It seems somewhat random which one will get presented. Strangely it does discover them properly but does not enumerate them all.
[log.txt](https://github.com/AravisProject/aravis/files/6611459/log.txt)

**Expected behavior**

to se a complete list of all devices 

**Camera description:**
 - Pleora Technologies inc.
 - Analog Pro PT01-AN1IP0x-32yG-
 - GigE
  
**Platform description:**
 - Aravis version 0.8.6 and 0.8.10
 - MacOS 10.15.7 and 11.4
 - Hardware MacBook Pro[86_64] and MacMini [M1]

EmmanuelP, 2021-06-07:
Hi,

Thanks for the report.

That is because the Pleora does not seem to return a serial number. As the device list is stored in a hash table using a key built using device manufacturer, model and serial number, Aravis does only remember one device.

As a workaround, it is possible to instantiate a camera using the MAC address or the IP address.

Could you attach the output of arv-tool-0.8 -d all:3 please ? Preferably with only the pleora device connected.

Louis-Demers, 2021-06-07:
My instinct would have been to use the MAC address for the hash table as this is supposed to be unique and not rely on all manufacturer implementing the serial number or other identifier.

I'll look into getting the camera directly. But that means I can no longer present the user with a possible selection. Not critical however. 

Here is the log with my 3 Pleora devices produced by  arv-tool-0.8 -d all:3. 
[3pleroa.txt](https://github.com/AravisProject/aravis/files/6611584/3pleroa.txt)

Louis-Demers, 2021-06-10:
Hi,  I'll certainly give it a try and give you back some feedback.

Louis-Demers, 2021-06-10:
Hi again, a quick test shows that it works.

```
[build] src/arv-tool-0.8            
FLIR-Blackfly S BFS-PGE-31S4M-19050440 (10.0.2.53)
FLIR-Blackfly S BFS-PGE-31S4M-19263381 (10.0.2.52)
FLIR-Blackfly S BFS-PGE-31S4M-19263386 (10.0.2.54)
Pleora Technologies Inc.-Analog Pro PT01-AN1IP0x-32yG-00:11:1c:04:58:7f (10.0.2.63)
Pleora Technologies Inc.-Analog Pro PT01-AN1IP0x-32yG-00:11:1c:04:65:56 (10.0.2.62)

```
Thanks a lot.

PS: it seems the Pleora device needs to be powered up with the ethernet cable attached. Simply reconnecting the ethernet cable while powered up does not make it grab an IP and communicate...

Louis-Demers, 2021-06-12:
> Since the discovery data is written and sent by the device, there is also no certainty the MAC data is correct.
> In the linked pull request, I have implemented a workaround where Aravis uses the MAC address as the serial number if it is empty. Could you please test this patch ?

Hi,

I tested it and it very works well.  I’ve added the output to the issue on GitHub.

Louis Demers Eng.
***@***.***

PR #513 (2021-07-02): gvinterface: use MAC as serial number fallback
Some devices have a NULL string as the serial number. Use the MAC
address a a serial number fallabck in this case.

Fix #511
commit 8f75d1f6 (2021-07-02): gvinterface: use MAC as serial number fallback

Some devices have a NULL string as the serial number. Use the MAC
address a a serial number fallabck in this case.

Fix #511
closed 2021-07-02 by EmmanuelP