AravisProject/aravis #514: Can't get image data from the camera

dyoo-dev, 2021-06-10:
Hello, 
Thank you for this great project. 
I am using LUCID Triton 0.4MP camera on CentOS 7.
When I run the following command, it detects the camera, but can't get image data.  
```
$ arv-camera-test-0.8 -n "10.0.0.123"
```

This is the debug log. I don't know what I should look at. Any thought or opinion would be great. 

```
/bin/arv-camera-test-0.8 -n "10.0.0.123"
Looking for camera '10.0.0.123'
[17:04:01.695] 🅸 device> [GvDevice::new] Interface address = 10.0.0.2
[17:04:01.695] 🅸 device> [GvDevice::new] Device address = 10.0.0.123
[17:04:01.700] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:Lucid_Vision_Labs_TRI004S-C_rev1_0_288aca1993d678e65944c943b7a64cb0.zip;800d9370;f202' at 0x200
[17:04:01.700] 🅸 device> [GvDevice::load_genicam] Xml address = 0x800d9370 - size = 0xf202 - Lucid_Vision_Labs_TRI004S-C_rev1_0_288aca1993d678e65944c943b7a64cb0.zip
[17:04:02.142] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[17:04:02.147] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:mstns'
[17:04:02.182] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.182] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.182] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.182] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.182] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.182] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.183] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.183] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.183] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.183] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.185] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.185] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.185] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.185] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.185] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.185] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.185] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.186] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.187] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.187] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.187] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.187] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.188] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.188] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.188] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.188] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.188] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.189] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.189] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.189] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.189] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[17:04:02.189] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[... 97 lines cut ...]
>=  2000;       0
<      0;       0
min     ;     n/a
max     ;     n/a
last max
at:     ;       0
Counter =        0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_failures             = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_timeouts             = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_aborteds             = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_missing_frames       = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_received_packets     = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_missing_packets      = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_error_packets        = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_ignored_packets      = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_resend_requests      = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_resent_packets       = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
[17:04:12.406] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
[17:04:12.406] 🅸 stream> [Stream::finalize] Flush 50 buffer[s] in input queue
[17:04:12.406] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue
```

hendrenja, 2021-06-10:
@dyoo-dev - I wish you the best of luck with this one. We have found Lucid devices work with a very slim set of third-party GenICam/GEV producers. We have an open ticket with Lucid Support regarding this exact behavior. My understanding is that they have done something non-standard in the stream transport. 

Vendors typically provide a Common Transport Interface (CTI) library that can be used to dynamically load the vendor-specific GenTL producer functionality. Lucid has a prototype library available for Windows but there is not a solution for Linux users. 

We resorted to building a Lucid-specific camera app using their ArenaSDK.

EmmanuelP, 2021-07-05:
Did you try to force the camera framerate, using -f option ?

dyoo-dev, 2021-07-15:
@EmmanuelP @haaninjo Thank you for your responses. I found that the firewall of CentOs blocked the image capture.

EmmanuelP, 2021-09-16:
Thanks for the follow up.

closed 2021-09-16 by EmmanuelP

NihalSoans, 2022-02-07:
I was having the same issue.
@[hendrenja](https://github.com/hendrenja)'s comment helped me find the issue I was having. Setting the node "GevGVSPExtendedIDMode" to 'Off' worked for me.

rgov, 2023-06-11:
@NihalSoans It looks like extended IDs have been supported since 2020 ([commit](https://github.com/AravisProject/aravis/commit/ccb0841455218884a9d3cfc573c4339615f9eea4)). Any idea why disabling it for Lucid cameras was necessary for you?