AravisProject/aravis #515: GVSP fails when client has Multiple NICs

gilesknap, 2021-06-14:
**Describe the bug**
GVSP fails to connect to the client when the client is running on a machine with multiple network interfaces.

**To Reproduce**
I am running the EPICS AreaDetector Client https://github.com/areaDetector/ADAravis to connect to an AVT Manta. All GVCP communications works fine but when I initiate acquisition the client sees no GVSP packets. I have done a network trace and can see that the client passes the ephemeral port for GVSP to the camera, however the IP address it passes along with this is for the wrong Network interface and is not routable from the camera.

If I run the same code on a different client with a single NIC then the streaming works correctly.

**Camera description:**
 - Manufacturer AVT 
 - Model Manta G-235 PoE
 - Interface: Ethernet UDP

**Platform description:**
 - Aravis version
 - OS: Ubuntu 20.04
 - Hardware x86_64

EmmanuelP, 2021-07-05:
Could you attach the output of `arv-tool-0.8 -d all:3` ?

gilesknap, 2021-07-12:
Thanks for getting back to me.

The results are:
```
arv-tool-0.8 -d all:3
Found 0 USB3Vision device (among 5 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 172.23.168.18
[GvDiscoverSocket::new] Add interface 172.17.0.1
[GvDiscoverSocket::new] Add interface 10.46.0.0
No device found
```
The 172.23.168.18 interface is the one through which GSVP is correctly talking to the camera. The camera is on a different subnet which is why it is showing as not found. The two subnets have a direct route through a switch.

gilesknap, 2021-07-12:
When I tested this with a packet sniffer I saw that there was a control packet  set with destination address 10.46.0.0.

eudoxos, 2021-08-08:
I *think* I was hitting something similar with #232 , does it look similar? (I fortunately no longer need that functionality). How are other GigEVision libs handling this? Is it possible to query the camera for IP address it sees when the client talks to it?

EmilioPeJu, 2023-11-23:
Hi @EmmanuelP , I think I have an approach that would fix this issue for most cases.

At the moment the source of the interface IP comes from calling `arv_gv_interface_camera_locate` (https://github.com/AravisProject/aravis/blob/main/src/arvgvinterface.c#L535C1-L535C31)
This function seems to have 2 phases:
    - firstly, it compares the masked IP value of each interface with the masked IP value of the camera, if this condition is met, it uses said IP
    - if the last step didn’t work, it tries to send a basic control packet to the camera over each interface, it uses the IP of the interface that got the answer back. This is the part were the wrong IP is obtained, not sure why.

My suggestion is that before the second phase, we could try sending the control packet only to the interface associated to the default gateway, if that doesn't answer, then do the second phase with all the interfaces.

I think it is reasonable to prioritise that interface and it fixes the problem, though I couldn't figure out exactly why the second phase can sometimes select the wrong interface.

Please let me know your thoughts.

gilesknap, 2023-11-29:
Thanks @EmilioPeJu.

@EmmanuelP would you accept a PR that implements Emilio's suggestion?

I the issue is simply that if there is an additional interface that can also contact the camera via a NAT then then there is a chance that the library will bind to that for the GVSP stream. But NATs will not allow new connections from outside.

This sounds unusual but it is normally the case when running inside a container or even running natively on a server that has kubernetes components installed.

Emilios solution only changes the order of the choosing the interfaces such that the default one is first - this should not affect any other users.

EmmanuelP, 2023-11-29:
> @EmmanuelP would you accept a PR that implements Emilio's suggestion?

I don't grasp the full implications of the proposal. A pull request would at least help me to test it. So please go ahead.