AravisProject/aravis #516: USB3Vision write_memory error (write protect

Nxt-1, 2021-06-15:
First described on the forums [here](https://aravis-project.discourse.group/t/usb-permissions/272). Using an Allied Vision 1800 U-158m using GStreamer. The system is running Ubuntu 18.04 on a RPi 4.

Afting installing 0.8.9 and 0.8.10 (they are both installed at this point I belive) and running `arv-camera-test-0.8` I get the following output:
```
Looking for the first available camera
Failed to read the current device configuration: USB3Vision read_memory error (access-denied)
```

As request on the forum, the output of `lsusb -v -d1ab2:`
```
Bus 002 Device 002: ID 1ab2:0001  
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x1ab2 
  idProduct          0x0001 
  bcdDevice            1.00
  iManufacturer           2 Allied Vision
  iProduct                3 ALVIUM 1800 U-158m
  iSerial                 4 00394
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           94
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              224mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass        239 Miscellaneous Device
      bFunctionSubClass       5 USB3 Vision
      bFunctionProtocol       0 
      iFunction               1 USB3 Vision Device
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      0 
      iInterface              0 
      ** UNRECOGNIZED:  14 24 01 00 00 01 00 00 00 01 00 05 06 07 08 09 0a 04 0b 0c
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      2 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength           22
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000002
      Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000c
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   3
      Lowest fully-functional device speed is SuperSpeed (5Gbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat        2047 micro seconds
Device Status:     0x000c
  (Bus Powered)
  U1 Enabled
  U2 Enabled
```

EmmanuelP, 2021-06-16:
Hi,

Thanks for the report.

Did you try `arv-camera-test-0.8` as root ?

Nxt-1, 2021-06-16:
I did not run it as root. I tested it again with sudo priviliges yet the output remains:
```
~$ sudo arv-camera-test-0.8 
Looking for the first available camera
Failed to read the current device configuration: USB3Vision read_memory error (access-denied)
```

EmmanuelP, 2021-07-05:
Please attach the output of `arv-camera-test-0.8 -d all`.

tijssen, 2021-08-14:
Hi there, sorry to revive this issue, but I'm facing this exact same issue with another Allied Vision USB 3 camera!

I am using the latest Aravis 0.8.16, Linux Mint 19 (Ubuntu 20.04 base) with an Allied Vision Mako U-051B. I've tried all the suggestions posed thusfar and every time I get the same ```read_memory error```. I'm certain I've set up the udev rules correctly and I've tried rebooting. When I try to access the camera through ```arv-viewer-0.8```, I can see the indicator light on the camera blinking continuously, indicating that there is some sort of data stream between the camera and the computer. However, the screen on my computer remains black. Trying any other way of interacting with the camera through Aravis (e.g. ```arv-camera-test-0.8``` or using one of the Python tests) results in the same error described above.

My output for ```arv-camera-test-0.8 -d all``` is as follows (using ```sudo``` doesn't change anything):
```
~$ arv-camera-test-0.8 -d all
Looking for the first available camera
[15:24:24.229] 🅸 interface> Found 1 USB3Vision device (among 6 USB devices)
[15:24:24.230] 🅸 device> [UvDevice::new] Vendor  = Allied Vision
[15:24:24.230] 🅸 device> [UvDevice::new] Product = Mako Mako U-051B
[15:24:24.230] 🅸 device> [UvDevice::new] S/N     = 671089271
[15:24:24.235] 🅸 device> [UvDevice::new] Using control endpoint 1, interface 0
[15:24:24.235] 🅸 device> [UvDevice::new] Using data endpoint 3, interface 1
[15:24:24.236] 🅸 device> Get genicam
[15:24:24.236] 🅸 device> MANUFACTURER_NAME =        'Allied Vision'
[15:24:24.239] 🅸 device> MAX_DEVICE_RESPONSE_TIME = 0x00000064
[15:24:24.239] 🅸 device> DEVICE_CAPABILITY        = 0x0000000000000f09
[15:24:24.239] 🅸 device> SRBM_ADDRESS =             0x0000000000010000
[15:24:24.239] 🅸 device> MANIFEST_TABLE_ADDRESS =   0x0000000000100000
[15:24:24.242] 🅸 device> U3VCP_CAPABILITY =         0x00000001
[15:24:24.242] 🅸 device> MAX_CMD_TRANSFER =         0x00000400
[15:24:24.242] 🅸 device> MAX_ACK_TRANSFER =         0x00000400
[15:24:24.242] 🅸 device> SIRM_OFFSET =              0x0000000000020000
[15:24:24.248] 🅸 device> SIRM_INFO =                  0x0a000000
[15:24:24.249] 🅸 device> SIRM_CONTROL =               0x00000000
[15:24:24.249] 🅸 device> SIRM_REQ_PAYLOAD_SIZE =      0x0000000000075300
[15:24:24.249] 🅸 device> SIRM_REQ_LEADER_SIZE =       0x00000038
[15:24:24.249] 🅸 device> SIRM_REQ_TRAILER_SIZE =      0x00000020
[15:24:24.249] 🅸 device> SIRM_MAX_LEADER_SIZE =       0x00000000
[15:24:24.249] 🅸 device> SIRM_PAYLOAD_SIZE =          0x00000000
[15:24:24.249] 🅸 device> SIRM_PAYLOAD_COUNT =         0x00000000
[15:24:24.249] 🅸 device> SIRM_TRANSFER1_SIZE =        0x00000000
[15:24:24.249] 🅸 device> SIRM_TRANSFER2_SIZE =        0x00000000
[15:24:24.249] 🅸 device> SIRM_MAX_TRAILER_SIZE =      0x00000000
[15:24:24.250] 🅸 device> MANIFEST_N_ENTRIES =       0x0000000000000001
[15:24:24.250] 🅸 device> MANIFEST ENTRY
                         00000000 00 00 00 01 00 04 01 01 00 00 20 00 00 00 00 00  .......... .....
                         00000010 5c 8a 00 00 00 00 00 00 91 7f 7b 9a 7f 53 af 7d  \.........{..S.}
                         00000020 03 60 09 91 69 01 b0 b9 3a 97 8c 9b 00 00 00 00  .`..i...:.......
                         00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[15:24:24.250] 🅸 device> genicam address =          0x0000000000200000
[15:24:24.250] 🅸 device> genicam size    =          0x0000000000008a5c
[15:24:24.283] 🅸 device> zip file =                 AVT_USB3_Vision.xml
[15:24:24.311] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.312] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.312] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.312] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.312] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.313] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.313] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.313] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.313] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.314] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.314] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.314] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.314] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.314] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.314] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.315] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.315] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.315] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.315] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.315] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.315] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.315] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.316] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.316] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.316] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[... cut ...]
[15:24:24.344] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.344] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.344] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.344] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.344] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:24:24.344] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
Failed to read the current device configuration: USB3Vision read_memory error (access-denied)
```
Thank you!

tijssen, 2021-08-14:
Oh and it might also be worth mentioning that I've tried this on Linux Mint 18 (base Ubuntu 18.04) as well, as well as earlier versions of Aravis.

On the other OS things behaved exaclty the same as described above. Earlier versions of Aravis (I believe 0.6.*) would behave different and sometimes got further but eventually they would come to a point that they'd freeze and I'd have to force quit the application. Testing earlier versions is a bit of work but if it'd help I could test more earlier versions.

tijssen, 2021-08-16:
Hey @EmmanuelP , have you had a look at this yet? If you want me to test anything else, let me know!

tijssen, 2021-08-18:
Hi there, me again. So I've been messing around a bit more and Aravis 0.8.6 on Ubuntu 21.04 works just fine with the aforementioned camera. So it seems like it is some sort of issue with a version of one of the dependencies (my suspicions go towards gstreamer and glibc6, perhaps?) packaged with Ubuntu 20.04 and below. Not an issue with Aravis, after all. Thanks for your time!

EmmanuelP, 2021-08-18:
Glad to hear you have solved your issue. I'm sorry to not have been more helpful.

Cheers.

closed 2021-08-18 by EmmanuelP