AravisProject/aravis #517: Problems with Basler usb3vison camera : daA1280-54uc

GauthierPerrod, 2021-06-15:
**Describe the bug**
Two bugs : at first couldn't frame because of some reasons.

``` Incomplete image received, dropping (received 1310868 / expected 3686400)```
I fixed using this https://github.com/AravisProject/aravis/issues/314
(256*256 things)

Seconds issue was that gstaravissrc wasn't working:

``` GstPipeline:pipeline0/GstAravis:aravis0: Could not set caps on camera "Basler-23484193": node 'AcquisitionFrameRateEnable
' not found ``` 
I fixed it by removing the followings lines :
```
+                       // if (local_error == NULL)                                                                                           
+                       //      arv_camera_set_boolean (camera, "AcquisitionFrameRateEnable", TRUE, &local_error);  ```
(l990-991)
in /src/arvcamera.c

**To Reproduce**
- Using a daA1280-54uc Basler on a jetson tx2  / jetson nx

**Expected behavior**
Unable to use arv-viewer and gstaravissrc

**Camera description:**
 - Basler
 - daA1280-54uc
 - usb3vison

**Platform description:**
 - Aravis version 0.8
 - OS: Ubuntu 18.04
 - Jetson nx / tx2

**Additional context**
My solution is working but it does'nt seem to be very reliable

EmmanuelP, 2021-07-05:
Could you attach the output of `arv-tool-0.8 genicam` ?

EmmanuelP, 2021-09-16:
Frame rate setting should have improved in the latest release. Please test Aravis 0.8.17, and reopen this issue if it is not fixed.

closed 2021-09-16 by EmmanuelP

akshayash001, 2022-11-10:
can anyone know that Basler daA1280-54uc support xilinx drivers ,linux drivers or any bare metal drivers?