AravisProject/aravis #518: Gstreamer cannot understand the camera pixel format: 0x1080009

Danilich1994, 2021-06-17:
**Describe the bug**
Aravis Viewer outputs "Gstreamer cannot understand the camera pixel format: 0x1080009" into terminal after play button press in GUI.

**To Reproduce**
`./arv-viewer-0.8`

**Expected behavior**
Start of image stream from camera.

**Camera description:**
 - Manufacturer MStar Vision
 - Model MV-A7200CU130
 - Interface [e.g. USB3] USB3

**Platform description:**
 - Aravis version 0.8.10
 - OS: [e.g. Fedora 20] Ubuntu Desktop 21.04
 - Hardware [e.g. x86_64] arm64

**Additional context**
Output of `./arv-viewer-0.8 -d all`
[arv-viewer.txt](https://github.com/AravisProject/aravis/files/6669891/arv-viewer.txt)

EmmanuelP, 2021-07-02:
What is the output of `arv-tool-0.8 features` ?

KeterM, 2021-07-20:
Dear sir, I have the same problem with a different camera.

**Describe the bug**
Aravis Viewer outputs "Gstreamer cannot understand the camera pixel format: 0x1080009" into terminal after play button press in GUI. I noticed that the bug happens before, as soon as I select the camera: on the GUI there is no pixel format in the combo box and the binning parameters are undefined. 

**Camera description:**
Manufacturer Daheng Imaging
Model MER2-202-60GC
Interface [Ethernet]

**Platform description:**
Aravis version 0.8.10
OS: Ubuntu Desktop 20.10
Hardware [e.g. x86_64] x86

**Additional context**
Output of  `arv-viewer-0.8 -d all`
[arv-viewer.txt](https://github.com/AravisProject/aravis/files/6847105/arv-viewer.txt)
Output of `arv-tool-0.8 features`
[arv_tool_features.txt](https://github.com/AravisProject/aravis/files/6847106/arv_tool_features.txt)

EmmanuelP, 2021-07-21:
Hi @KeterM 

It is most probably not related, as your camera is a GigEVision one, while the original bug report is about an USB3Vision. Please open another issue.

Thanks.

EmmanuelP, 2021-07-21:
> It is most probably not related, as your camera is a GigEVision one, while the original bug report is about an USB3Vision. Please open another issue.

Well after rereading the debug output, it is may be related. Nevertheless, please still open another issue.

closed 2021-07-21 by EmmanuelP