AravisProject/aravis #522: arv-viwer and docker

PauloFavero, 2021-06-29:
Hi guys,

I have installed gstreamer and aravis inside a docker container. The arv-tool works well but when launching the arv-viewer these errors appears:

Gtk-Message: 17:55:29.036: Failed to load module "canberra-gtk-module"
Gtk-Message: 17:55:29.040: Failed to load module "canberra-gtk-module"

```
(arv-viewer-0.8:12): GLib-GIO-CRITICAL **: 17:55:29.397: g_dbus_proxy_new_sync: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
```

I wrote this Dockerfile: 

```
FROM ubuntu:18.04

RUN apt-get update && apt-get install -y build-essential \
    pkg-config \
    python3 \
    python3-pip \
    sudo \
    udev \
    cmake \
    gcc-multilib \
    g++-multilib \
    libcanberra-gtk-module && \
    pip3 install pip --upgrade \
    pip3 install meson \
    pip3 install ninja

WORKDIR /app
COPY aravis-0.8.10.tar.xz ./

RUN tar -xf aravis-0.8.10.tar.xz

ENV DEBIAN_FRONTEND noninteractive

# Install gstreamer
RUN apt-get install -y  libglib2.0-dev libxml2-dev libnotify-dev libaudit-dev libgstreamer* \
    gstreamer1.0-tools \
    gstreamer1.0-alsa \
    gstreamer1.0-plugins-base \
    gstreamer1.0-plugins-good \
    gstreamer1.0-plugins-bad \
    gstreamer1.0-plugins-ugly \
    gstreamer1.0-libav

RUN apt-get update && \
    apt-get install -y \
        intltool \
        libgirepository1.0-dev \
        gtk-doc-tools \
        libglib2.0-dev \
        libglibmm-2.4-dev \
        libcanberra-gtk-module

# Build aravis
# https://download-fallback.gnome.org/sources/aravis/0.8/pip
WORKDIR /app/aravis-0.8.10

RUN meson build && cd build && ninja && ninja install
RUN ldconfig

# User configuration
ARG USER_NAME=${USER_NAME}
ARG USER_PASSWORD=${USER_PASSWORD}
ARG USER_UID=${USER_UID}
ARG USER_GID=${USER_UID}

RUN groupadd -g ${USER_GID} ${USER_NAME}

RUN useradd -d /home/${USER_NAME} -s /bin/bash \
    -m ${USER_NAME} -u ${USER_UID} -g ${USER_GID} -G sudo \
    && echo "${USER_NAME}:${USER_PASSWORD}" | \
    chpasswd && adduser ${USER_NAME} sudo && usermod -aG sudo ${USER_NAME} && usermod -a -G video ${USER_NAME}

USER ${USER_NAME}
```

The docker-compose file to build and run this container is the following:

```
version: '3.5'
  
services:
  gstreamer:
    container_name: gstreamer
    build:
      context: .
      dockerfile: Dockerfile
      args:
        - USER_NAME=aravis
        - USER_PASSWORD=aravis
        - USER_UID=1000
        - USER_GID=1000
    devices: 
      - "/dev:/dev"
    #   - /dev/ttyACM0:/dev/ttyACM0
    network_mode: "host"
    volumes: 
      - $HOME/.Xauthority:/root/.Xauthority
      - /tmp/.X11-unix/:/tmp/.X11-unix
    environment:
      - GST_PLUGIN_PATH=/usr/local/lib/x86_64-linux-gnu/gstreamer-1.0
      - DISPLAY=$DISPLAY
    privileged: true
    command: arv-viewer-0.8
```
Also the interface is not properly rendered when activating a camera. 

[image]

Does anyone know what could be the problem?

Thanks,
Paulo

james-lan, 2021-12-22:
I'm encountering this with a Chameleon3, and the fake camera, and on multiple NON-DOCKER distros: Arch x86-64 (Intel CPU/graphics) built from git source, Kubuntu 21.10 (Intel CPU/graphics) from packages. (I think another ubuntu flavor (Xubuntu?) Both under X11 and Wayland. I have a few more systems I can try. ((Suse, Gentoo) if that would be helpful. It did work one time, I think after I switched from Wayland back to X11. 

It also happened on the Kubuntu system with the 'fake' camera, so it's not the camera. 

If wanted I can test on a few different distros and Nvidia GPU, but I expect the results to be similar. 

Gstreamer throws no errors even with debug on, until trying to stop the stream. GTK doesn't seem to report much useful even with -gdk-debug=all. Vendor tool can see and display it, so it's not the camera. If you want I can run another log and post it here? 

Example to confirm it here with the test camera built from git. 
[image]

Let me know what would be useful to have to track it down.

james-lan, 2021-12-22:
Same thing with Gentoo (Intel CPU/Nvidia GPU) under X11 with the fake camera. (built from gentoo portage packages)

EmmanuelP, 2022-05-11:
Probably a dup of #345 . Please reopen if the proposed solution of #345 does not work.

closed 2022-05-11 by EmmanuelP