AravisProject/aravis #525: localtime_r needs -D_POSIX_C_SOURCE in mingw64

eudoxos, 2021-07-12:
[localtime_r manpage](https://linux.die.net/man/3/localtime_r) explains the feature macros. Otherwise `localtime_r` is not defined mingw64 (see also e.g. https://github.com/m-ab-s/media-autobuild_suite/issues/1211). This can be fixed by adding `-D_POSIX_C_SOURCE` to `library_c_args` in `src/meson.build` though I am not sure whether this can be done unconditionally.

https://github.com/AravisProject/aravis/blob/76254b7faf59901166ec78a73e94b03a0cd8bb77/src/arvdebug.c#L156

EmmanuelP, 2021-07-12:
Hi @eudoxos 

Does #526 fix your issue ?

eudoxos, 2021-07-12:
Perfect, please merge that!

EmmanuelP, 2021-07-12:
Unfortunately, it looks like localtime is not threadsafe. I have to find another solution.

EmmanuelP, 2021-07-12:
#526 is updated and use glib API now.

commit a6104c01 (2021-07-12): debug: use localtime instead of localtime_r

The former one doesn't need _POSIX_C_SOURCE, which is not defined when aravis is
compiled using mingw64.

Fix #525
commit 31536988 (2021-07-12): debug: don't use localtime_r

It requires _POSIX_C_SOURCE, which is not defined when aravis is compiled using
mingw64.

Fix #525
PR #526 (2021-07-13): debug: use localtime instead of localtime_r
The former one doesn't need _POSIX_C_SOURCE, which is not defined when aravis is
compiled using mingw64.

Fix #525
commit 637bc796 (2021-07-13): debug: don't use localtime_r

It requires _POSIX_C_SOURCE, which is not defined when aravis is compiled using
mingw64.

Fix #525
closed 2021-07-13 by EmmanuelP