AravisProject/aravis #527: Please consider change tagging convention

kloczek, 2021-07-13:
At the moment is used tagging patters `ARAVIS_<version_with_underscores>`.
To allow easier map git tag to packages version it would be better use tagging convention which is used by majority of all OSS projects which is `<version.with.dots>`.

EmmanuelP, 2021-07-13:
The problem is github insists in publishing source archives directly built from git. While for now the content of github built tarballs is (or almost is) the same as the tarballs I'm building, it will be probably not always the case.

The correct tarballs is the one with the correct naming, ie aravis-x.y.z.tar.xz

closed 2021-07-13 by EmmanuelP

kloczek, 2021-07-13:
Please read again ticket descriptio. I'm asking about consider change **tagging convention**.
This has nothing to do with tar ball content.
Maybe if I'll show you line from my aravis.spec this will tell you about what I'm taking:
```spec
Source0:        %{VCS}/releases/download/ARAVIS_%(tr "." "_" <<<%{version})/%{name}-%{version}.tar.xz
```
If tagging conveion would be as `<version.with.dots>` that line may look like below:
```spec
Source0:        %{VCS}/releases/download/%{version}/%{name}-%{version}.tar.xz
```
I'm asking about use KISS principle .. only this and nothing more.

EmmanuelP, 2021-07-13:
> Please read again ticket description

Please consider I'm reading carefully the ticket opened on github.

While I agree the path to the official tarballs will be simpler, it will also make the name of the unwanted ones similar to the official one, which is something I'd like to avoid. But I may change my mind.

kloczek, 2021-07-13:
Why you are trying to avoid that?

kloczek, 2021-07-13:
Thak you very much for the change :)