AravisProject/aravis #529: arv_camera_acquisition() takes long time

dyoo-dev, 2021-07-15:
**Describe the bug**
Hello, 
I take a single image from a camera with **arv_camera_acqusition()**. 
0.4MP camera takes 350ms, and 1.6MP camera takes over 1sec to take a single image.
I think it should be very short. Why is it taking so long time? 

**Camera description:**
 - Manufacturer: LUCID
 - Model: Triton 0.4MP and 1.6MP
 - Interface [e.g. USB3]: ethernet

**Platform description:**
 - Aravis version: 0.8
 - OS: [e.g. Fedora 20]: Ubuntu 18.04 and CentOS 7

EmmanuelP, 2021-07-21:
Please attach the output of `ARV_DEBUG=all ./tests/arv-acquisition-test`

dyoo-dev, 2021-07-22:
This is the output. Please take a look. Thanks

```
[10:32:06.301] ❷ interface> Found 0 USB3Vision device (among 9 USB devices)
[10:32:06.301] ❷ interface> [GvDiscoverSocket::new] Add interface 127.0.0.1
[10:32:06.301] ❷ interface> [GvDiscoverSocket::new] Add interface 192.168.167.132
[10:32:06.301] ❷ interface> [GvDiscoverSocket::new] Add interface 10.0.0.120
[10:32:06.301] ❷ interface> [GvDiscoverSocket::new] Add interface 172.19.0.1
[10:32:06.301] ❷ interface> [GvDiscoverSocket::new] Add interface 172.20.0.1
[10:32:06.301] ❷ interface> [GvDiscoverSocket::new] Add interface 172.17.0.1
[10:32:06.306] ❷ interface> [GvInterface::discovery] Device 'Lucid Vision Labs-TRI004S-C-211800832' found (interface 10.0.0.120) user_id '' - MAC '1c:0f:af:a9:a7:49'
[10:32:06.306] ❷ device> [GvDevice::new] Interface address = 10.0.0.120
[10:32:06.306] ❷ device> [GvDevice::new] Device address = 10.0.0.42
[10:32:06.310] ❷ device> [GvDevice::load_genicam] xml url = 'Local:Lucid_Vision_Labs_TRI004S-C_rev1_0_288aca1993d678e65944c943b7a64cb0.zip;800d9370;f202' at 0x200
[10:32:06.310] ❷ device> [GvDevice::load_genicam] Xml address = 0x800d9370 - size = 0xf202 - Lucid_Vision_Labs_TRI004S-C_rev1_0_288aca1993d678e65944c943b7a64cb0.zip
[10:32:06.984] ❶ device> [GvDevice::read_memory] Ack reception timeout
[10:32:06.984] ❶ device> [GvDevice::read_memory] Ack reception timeout
[10:32:07.486] ❶ device> [GvDevice::read_memory] Ack reception timeout
[10:32:07.486] ❶ device> [GvDevice::read_memory] Ack reception timeout
[10:32:07.801] ❷ device> [GvDevice::load_genicam] Zipped xml data
[10:32:07.802] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:mstns'
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.804] ❷ dom> [Genicam::create_element] Unknown tag (String)
[10:32:07.818] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.818] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.818] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.818] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.818] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.818] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.819] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.820] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.820] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.820] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.820] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.820] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.820] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.820] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.821] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.822] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[10:32:07.822] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[... 87 lines cut ...]
<      0;       0
min     ;     n/a
max     ;     n/a
last max
at:     ;       0
Counter =        0
[... cut ...]
[10:32:07.910] ❷ stream> [GvStream::finalize] n_ignored_packets      = 0
[10:32:07.910] ❷ stream> [GvStream::finalize] n_resend_requests      = 0
[10:32:07.910] ❷ stream> [GvStream::finalize] n_resent_packets       = 0
[10:32:07.910] ❷ stream> [GvStream::finalize] n_resend_ratio_reached = 0
[10:32:07.910] ❷ stream> [GvStream::finalize] n_duplicated_packets   = 0
[10:32:07.910] ❷ stream> [Stream::finalize] Flush 0 buffer[s] in input queue
[10:32:07.910] ❷ stream> [Stream::finalize] Flush 0 buffer[s] in output queue
Image successfully acquired
```

EmmanuelP, 2021-07-26:
Most of the time is spent loading the Genicam data from the device (1.5s). A cache would help a lot in this example.

```
[10:32:06.310] ❷ device> [GvDevice::load_genicam] xml url = 'Local:Lucid_Vision_Labs_TRI004S-C_rev1_0_288aca1993d678e65944c943b7a64cb0.zip;800d9370;f202' at 0x200
...
[10:32:07.802] ❷ dom> [GcFeature::set_attribute] Unknown attribute 'xmlns:mstns'
```

dyoo-dev, 2021-07-27:
@EmmanuelP Thank you for your reply. Can you explain more about the cache? 

Also, I got the following logs sometimes. Usually, the snapshot takes 500ms, but whenever I got this, the snapshot took over 1 sec.  Why do I have these messages?  
```
[10:32:06.984] ❶ device> [GvDevice::read_memory] Ack reception timeout
[10:32:06.984] ❶ device> [GvDevice::read_memory] Ack reception timeout
[10:32:07.486] ❶ device> [GvDevice::read_memory] Ack reception timeout
[10:32:07.486] ❶ device> [GvDevice::read_memory] Ack reception timeout
```

EmmanuelP, 2021-07-27:
> @EmmanuelP Thank you for your reply. Can you explain more about the cache?

It could be possible to store the retrieved genicam data in a dedicated directory, and use this cached data the next time aravis connects to the camera. But it is not implemented yet.

> Also, I got the following logs sometimes. Usually, the snapshot takes 500ms, but whenever I got this, the snapshot took over 1 sec. Why do I have these messages?

There are some packet lost during data transmission, and the default time-out is a bit large (500ms). Unfortunately there is no setting for this value, except one at compilation time, when you select the fast-heartbeat option.