AravisProject/aravis #530: Gstreamer cannot understand the camera pixel format: 0x1080009 [GigEVision]

KeterM, 2021-07-21:
Dear sir, I have the same problem with a different camera.

**Describe the bug**
Aravis Viewer outputs "Gstreamer cannot understand the camera pixel format: 0x1080009" into terminal after play button press in GUI. I noticed that the bug happens before, as soon as I select the camera: on the GUI there is no pixel format in the combo box and the binning parameters are undefined. 

**Camera description:**
Manufacturer Daheng Imaging
Model MER2-202-60GC
Interface [Ethernet]

**Platform description:**
Aravis version 0.8.11 
OS: Ubuntu Desktop 20.10
Hardware [e.g. x86_64] x86

**Additional context**
Output of  `arv-viewer-0.8 -d all`
[arv-viewer.txt](https://github.com/AravisProject/aravis/files/6847105/arv-viewer.txt)
Output of `arv-tool-0.8 features`
[arv_tool_features.txt](https://github.com/AravisProject/aravis/files/6847106/arv_tool_features.txt)

_Originally posted by @KeterM in https://github.com/AravisProject/aravis/issues/518#issuecomment-883244640_

EmmanuelP, 2021-07-21:
Thanks for the new issue.

Could you attach the output of `arv-camera-test-0.8 -d all` ?

Did you try to select another pixel format in the viewer ?

Do you have several network connections ? (like WIFI + wired ethernet)

KeterM, 2021-07-21:
Thank you Emmanuel for your effort.
Here the output (I stopped after a while) of the command `arv-camera-test-0.8 -d all`
[arv-camera-test.txt](https://github.com/AravisProject/aravis/files/6854509/arv-camera-test.txt)

I can't select another pixel format: the combo box is empty.
My pc and the camera are both wired connected, through the same switch.

EmmanuelP, 2021-07-21:
The issue is your camera only supports bayer pixel formats, but you don't have the gstreamer bayer plugin installed.

It should be provided by the gst-plugins-bad package. Please install it and retry to use arv-viewer.

KeterM, 2021-07-21:
Yes! I confirm that after the **gst-plugins-bad** package installation, now it works!
The pixel format combo box is not empty, there is only the BayerBG8 format.

closed 2021-07-21 by KeterM