AravisProject/aravis #540: -Wenum-conversion not understood by older compilers (gcc 9.3.0)

eudoxos, 2021-07-25:
https://github.com/AravisProject/aravis/blob/c139db0127a8a0d10e9c7cf87c81c9fd5f963030/meson.build#L34

I am using `x86_64-linux-gnu-gcc` cross-compiler under Linux, it is version `9.3.0`, and the `-Wenum-conversion` is not understood (it is understood by the native gcc 9.3.0, curiously):
```
x86_64-linux-gnu-gcc: error: unrecognized command line option ‘-Wenum-conversion’; did you mean ‘-Wno-conversion’?
```

EmmanuelP, 2021-07-25:
Hi,

The point of cc.get_supported_arguments is to filter the options to the one actually supported by the compiler. Which means -Wenum-conversion should not be passed to gcc if it is not supported. Are you trying to use meson cross compilation ?

eudoxos, 2021-07-25:
> Are you trying to use meson cross compilation ?

Yes, exactly. As described here: https://github.com/eudoxos/aravis/blob/winbuild/README.md#cross-compilation-for-windows

EmmanuelP, 2021-07-25:
Unless I have misunderstood crossroad, you are not using meson cross compilation support. I was asking because if meson cross compilation is used, there is a distinction in meson between the native compiler and the cross one, which could explain why the unsupported option is used during compilation.

https://mesonbuild.com/Cross-compilation.html

eudoxos, 2021-07-25:
`crossroad meson` will add the `--cross-file` to (native) `meson` according to the current platform; the wrapping is [here](https://git.tuxfamily.org/crossroad/crossroad.git/tree/src/in-crossroad.py#n334) and [this is the cross-file toolchain-w64.meson](https://git.tuxfamily.org/crossroad/crossroad.git/tree/platforms/meson/toolchain-w64.meson). So it should be running the cross-compiler.. Strange that it fails in this case. I will try to look closer what meson calls.

eudoxos, 2021-07-26:
After a closer look, I understand better now. The failure is when calling (native) `g-ir-scanner` which passes all flags for the cross-compiler to the native compiler. And as it happens, the cross-compiler does support `Wenum-conversion` (hence it does not get filtered out by `get_supported_arguments`) whereas the native gcc internally called by `g-ir-scanner` does not and fails. 

I was misled by the original post seeing `x86_64-linux-gnu-gcc` (which is native, normally called just `gcc` ;) ) as opposed to `x86_64-w64-mingw32-gcc` (which is the cross-compiler).

eudoxos, 2021-07-26:
It seems to be meson's upstream issue [here](https://github.com/mesonbuild/meson/blob/dbf2ace6ca1ce39aa01497f815b65856079cc581/mesonbuild/modules/gnome.py#L449). It takes compile arguments from the dependency (which is `aravis_library` in this case) but calls native compiler with them. I will see if they can do something about it.

eudoxos, 2021-07-26:
My idea was to use only common warning options for cross-compilation:
```meson
  warning_c_args=cc.get_supported_arguments(warning_c_args)
  if meson.is_cross_build()
    native_cc = meson.get_compiler('c', native: true)
    warning_c_args = native_cc.get_supported_arguments(warning_c_args)
  endif
```
but that is not enough as `g-ir-scanner` also runs linker, which does not find host libraries, being native linker.

So I'd say for now that introspection is unsupporter for cross-compilation, unless a solution is found.

EmmanuelP, 2021-07-26:
Did you try to only add '-Wenum-conversion' if the build is native ?

EmmanuelP, 2021-07-26:
What I'm proposing is different:

```meson
if not meson.is_cross_build()
  warning_c_args += '-Wenum-conversion`
endif
```

eudoxos, 2021-07-26:
That will only for for my particular combination of compilers, unlike the second filter for warning_c_args which I proposed above.  But anyway, I'd make introspection off by default for cross-compilation, until it is resolved (on meson side, or wherever the issue stems from).