AravisProject/aravis #541: Python Required

jiapei100, 2021-07-26:
**Is your feature request related to a problem? Please describe.**
It looks by strictly following **meson manual**, there is no such a python wrapper built out??? I wonder how I can use **Aravis** via **Python**.

**Describe the solution you'd like**
Successfully call the following couple of lines:
```console
➜  ~ python
Python 3.8.10 (default, Jun  2 2021, 10:49:15) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Aravis
```

**Describe alternatives you've considered**
- Tested under the test folder already, the direct command:
```console
➜  python ls
arv-buffer-test.py  arv-camera-test.py  arv-enum-test.py  arv-evaluator-test.py  arv-feature-test.py  arv-genicam-test.py  arv-single-buffer-test.py
➜  python python arv-camera-test.py 
Traceback (most recent call last):
  File "arv-camera-test.py", line 14, in <module>
    gi.require_version ('Aravis', '0.8')
  File "~/.local/lib/python3.8/site-packages/gi/__init__.py", line 126, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Aravis not available
```
- Tried [python-aravis](https://github.com/SintefManufacturing/python-aravis), but failed to install that python package.

**Additional context**
N/A

EmmanuelP, 2021-07-27:
Hi,

In order to be able to `import Aravis` directly, you have to proceed to a system wide installation in `/usr`. By default, meson installs everything in `/usr/ocal`.

Alternatively, you can tell python the path to the Aravis modules, using GI_TYPELIB_PATH and LD_LIBRARY_PATH environment variables.

https://stackoverflow.com/questions/19156974/how-to-get-python-to-recognize-find-homebrew-installed-gi-repository-pygobjects

`

closed 2021-07-27 by EmmanuelP

jiapei100, 2021-07-27:
@EmmanuelP 
Thank you..

I successfully installed **Aravis** again, using `➜  aravis-0.8.15 meson --prefix /usr --buildtype=plain builddir`.
And, I did `➜  export GI_TYPELIB_PATH="/usr/lib/x86_64-linux-gnu/girepository-1.0" `

As for `LD_LIBRARY_PATH`, it does have included `/usr/lib/x86_64-linux-gnu` .
Still, `import Aravis` does NOT work.

Any more suggestions?

EmmanuelP, 2021-07-27:
On some systems, you have to run `ldconfig` as root after a library installation.

jiapei100, 2021-07-27:
@EmmanuelP 
Tried...

```
➜  ~ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal
➜  ~ python --version
Python 3.8.10
➜  ~ python
Python 3.8.10 (default, Jun  2 2021, 10:49:15) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import Aravis
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'Aravis'
>>> 
```

EmmanuelP, 2021-07-27:
Aravis python binding is provided by gobject introspection. You can not directly import Aravis. Here are the needed steps:

```
import gi
gi.require_version ('Aravis', '0.8')
from gi.repository import Aravis
```

jiapei100, 2021-07-27:
@EmmanuelP 
Great ... Thank you ...
Any more documentation for Aravis Python? And, any sample code of **Aravis Python**?

Thank you ....

EmmanuelP, 2021-07-27:
Not much documentation I'm afraid. There is this page about Aravis openCV and Python, but a it outdated.

jiapei100, 2021-07-28:
@EmmanuelP 
The test code under **python** folder helped a lot.. However, how to convert from `gi.repository.Aravis.Buffe` to `numpy.ndarray` or `opencv cv::Mat` ??

jiapei100, 2021-07-28:
@EmmanuelP 
And, how to call these 2 functions?

```console
camera.set_gain_auto()
camera.set_exposure_time_auto()
```

Both of these 2 functions require 2 parameters:

```console
xception has occurred: TypeError
Aravis.Camera.set_gain_auto() takes exactly 2 arguments (1 given)
  File "....../arv-camera-test.py", line 34, in <module>
    camera.set_gain_auto()
```

jiapei100, 2021-07-28:
What's more, @EmmanuelP 
How to let Aravis automatically find multiple cameras?

EmmanuelP, 2021-07-28:
Hi,

Github issues is not the right place for these questions. Please use the Aravis discourse instance.

https://aravis-project.discourse.group/c/using-aravis/8