AravisProject/aravis #543: cannot run arv-viewer

PythonGuython, 2021-07-28:
**Hardware**
Raspberry pi 4

**Environment**
Raspbian OS

I'm able to successfully build Aravis - I have validated that the fake camera works. However, I cannot get arv-viewer to work. I have installed all the packages required, the only error I get while building is for the C compiler:

`Compiler for C supports arguments -Wenum-conversion: NO (cached)`

Other than that, meson lists all dependences as installed.

I have enabled usb, viewer, and gst-plugin in `meson configure`
and I rebuilt with `ninja reconfigure`

Am I missing something or is this an error? I apologize if I am posting this in the wrong place.

EmmanuelP, 2021-07-28:
Hi,

> However, I cannot get arv-viewer to work.

If you don't say what goes wrong, it will be difficult to help you. Be more specific please.

PythonGuython, 2021-07-28:
after building, I able to run the bash executables `arv-camera-test-0.8`, a`rv-fake-gv-camera-0.8` and `arv-tool-0.8`, but there is no `arv-viewer` executable. Using `meson configure -D` I modified my meson configuration to be as follows:

```
Core properties:
  Source dir /home/pi/aravis
  Build dir  /home/pi/aravis/build
                                                                                                            
Main project options:                                                                                       
                                                                                                            
  Core options        Current Value           Possible Values                                               Description
  ------------        -------------           ---------------                                               -----------
  auto_features       auto                    [enabled, disabled, auto]                                     Override value of all 'auto' features
  backend             ninja                   [ninja, vs, vs2010, vs2015, vs2017, vs2019, xcode]            Backend to use
  buildtype           debug                   [plain, debug, debugoptimized, release, minsize, custom]      Build type to use
  debug               true                    [true, false]                                                 Debug
  default_library     shared                  [shared, static, both]                                        Default library type
  force_fallback_for  []                                                                                    Force fallback for those subprojects
  install_umask       0022                    [preserve, 0000-0777]                                         Default umask to apply on permissions of installed files
  layout              mirror                  [mirror, flat]                                                Build directory layout
  optimization        0                       [0, g, 1, 2, 3, s]                                            Optimization level
  strip               false                   [true, false]                                                 Strip targets on install
  unity               off                     [on, off, subprojects]                                        Unity build
  unity_size          4                       >=2                                                           Unity block size
  warning_level       1                       [0, 1, 2, 3]                                                  Compiler warning level to use
  werror              false                   [true, false]                                                 Treat warnings as errors
  wrap_mode           default                 [default, nofallback, nodownload, forcefallback]              Wrap mode
  cmake_prefix_path   []                                                                                    List of additional prefixes for cmake to search
  pkg_config_path     []                                                                                    List of additional paths for pkg-config to search
                                                                                                            
  Backend options     Current Value           Possible Values                                               Description
  ---------------     -------------           ---------------                                               -----------
  backend_max_links   0                       >=0                                                           Maximum number of linker processes to run or 0 for no limit
                                                                                                            
  Base options        Current Value           Possible Values                                               Description
  ------------        -------------           ---------------                                               -----------
  b_asneeded          true                    [true, false]                                                 Use -Wl,--as-needed when linking
  b_colorout          always                  [auto, always, never]                                         Use colored output
  b_coverage          false                   [true, false]                                                 Enable coverage tracking.
  b_lto               false                   [true, false]                                                 Use link time optimization
  b_lundef            true                    [true, false]                                                 Use -Wl,--no-undefined when linking
  b_ndebug            false                   [true, false, if-release]                                     Disable asserts
  b_pch               true                    [true, false]                                                 Use precompiled headers
  b_pgo               off                     [off, generate, use]                                          Use profile guided optimization
  b_pie               false                   [true, false]                                                 Build executables as position independent
  b_sanitize          none                    [none, address, thread, undefined, memory, address,undefined] Code sanitizer to use
[... cut ...]
Program gtkdoc-scan found: YES (/usr/bin/gtkdoc-scan)
Program xsltproc found: YES (/usr/bin/xsltproc)
Configuring gtkdocentities.ent using configuration
Program gtkdoc-scan found: YES (/usr/bin/gtkdoc-scan)
Program gtkdoc-scangobj found: YES (/usr/bin/gtkdoc-scangobj)
Program gtkdoc-mkdb found: YES (/usr/bin/gtkdoc-mkdb)
Program gtkdoc-mkhtml found: YES (/usr/bin/gtkdoc-mkhtml)
Program gtkdoc-fixxref found: YES (/usr/bin/gtkdoc-fixxref)
Build targets in project: 44

Found ninja-1.10.1 at /usr/bin/ninja

```

EmmanuelP, 2021-07-29:
'arv(viewer-0.8` is built into the `viewer` directory. I would be surprised if it was not there.

EmmanuelP, 2021-07-29:
Be aware the performance of the viewer will be pretty terrible on a raspberry.

PythonGuython, 2021-07-30:
Maybe it is built, I'm quite sure how to tell. Here is a screenshot of the files in the `viewer` directory:

[image]

I went into the `build` directory within the `viewer` directory and found my meson log:

```
Build started at 2021-07-27T20:40:27.294719
Main binary: /usr/bin/python3
Build Options: 
Python system: Linux
The Meson build system
Version: 0.56.1
Source dir: /home/pi/aravis/viewer
Build dir: /home/pi/aravis/viewer/build
Build type: native build

ERROR: First statement must be a call to project
```

I'm not sure what could be causing this. Maybe it has something to do with where I installed aravis? I installed it on my desktop. Not sure if that is relevant

EmmanuelP, 2021-07-30:
> Maybe it has something to do with where I installed aravis? I installed it on my desktop.

I don't understand what it means.

What is the content of `build/meson-logs/meson-log.txt` ? Where `build` is the build directory.

PythonGuython, 2021-07-30:
> Maybe it has something to do with where I installed aravis? I installed it on my desktop.
> 
> I don't understand what it means.

actually I installed it in `/home/pi`. As in I navigated to `/home/pi`, cloned the repository and built from there. I'm not sure if there is a convention for install location of libraries. As you can probably tell, I'm new to building from source.

Here is the content of `build/meson-logs/meson-log.txt`

```
Build started at 2021-07-28T16:39:41.438086
Main binary: /usr/bin/python3
Build Options: -Dviewer=enabled -Dbackend=ninja -Dgst-plugin=enabled -Dusb=enabled
Python system: Linux
The Meson build system
Version: 0.56.1
Source dir: /home/pi/aravis
Build dir: /home/pi/aravis/build
Build type: native build
Project name: aravis
Project version: 0.8.16
C compiler for the host machine: cc (gcc 8.3.0 "cc (Raspbian 8.3.0-6+rpi1) 8.3.0")
C linker for the host machine: cc ld.bfd 2.31.1
None of 'AR' are defined in the environment, not changing global flags.
C++ compiler for the host machine: c++ (gcc 8.3.0 "c++ (Raspbian 8.3.0-6+rpi1) 8.3.0")
C++ linker for the host machine: c++ ld.bfd 2.31.1
C compiler for the build machine: cc (gcc 8.3.0 "cc (Raspbian 8.3.0-6+rpi1) 8.3.0")
C linker for the build machine: cc ld.bfd 2.31.1
None of 'AR' are defined in the environment, not changing global flags.
C++ compiler for the build machine: c++ (gcc 8.3.0 "c++ (Raspbian 8.3.0-6+rpi1) 8.3.0")
C++ linker for the build machine: c++ ld.bfd 2.31.1
Build machine cpu family: arm
Build machine cpu: armv7l
Host machine cpu family: arm
Host machine cpu: armv7l
Target machine cpu family: arm
Target machine cpu: armv7l
Using cached compile:
Cached command line:   

Code:
 extern int i;
int i;

Cached compiler stdout:
 
Cached compiler stderr:
 
Compiler for C supports arguments -Wduplicated-branches: YES (cached)
Using cached compile:
Cached command line:   

Code:
 extern int i;
int i;

Cached compiler stdout:
 
Cached compiler stderr:
 
Compiler for C supports arguments -Wimplicit-fallthrough: YES (cached)
Using cached compile:
Cached command line:   

Code:
 extern int i;
int i;

Cached compiler stdout:
 
Cached compiler stderr:
[... 435 lines cut ...]
PKG_CONFIG_PATH: 
Got pkgconfig variable girdir : 
PKG_CONFIG_PATH: 
PKG_CONFIG_PATH: 
Got pkgconfig variable girdir : 
Build targets in project: 44

Found ninja-1.10.1 at /usr/bin/ninja
Running compile:
Working directory:  /home/pi/aravis/build/meson-private/tmptx25wypb
Command line:  c++ /home/pi/aravis/build/meson-private/tmptx25wypb/testfile.cpp -o /home/pi/aravis/build/meson-private/tmptx25wypb/output.obj -pipe -c -D_FILE_OFFSET_BITS=64 -O0 -fpermissive --print-search-dirs 

Code:
 
Compiler stdout:
 install: /usr/lib/gcc/arm-linux-gnueabihf/8/
programs: =/usr/lib/gcc/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../../arm-linux-gnueabihf/bin/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../../arm-linux-gnueabihf/bin/arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../../arm-linux-gnueabihf/bin/
libraries: =/usr/lib/gcc/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../../arm-linux-gnueabihf/lib/arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../../arm-linux-gnueabihf/lib/arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../../arm-linux-gnueabihf/lib/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/8/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/8/../../../:/lib/arm-linux-gnueabihf/8/:/lib/arm-linux-gnueabihf/:/lib/:/usr/lib/arm-linux-gnueabihf/8/:/usr/lib/arm-linux-gnueabihf/:/usr/lib/

Compiler stderr:
 
None of 'STRIP' are defined in the environment, not changing global flags.
None of 'STRIP' are defined in the environment, not changing global flags.

```

EmmanuelP, 2021-07-30:
> actually I installed it in /home/pi.

The installation part is another step you probably have not done yet (It happens when you call `ninja install`). But anyway, if the compilation went fine without error, you should have the viewer binary in the viewer build directory.

What is the content of `/home/pi/aravis/viewer/build` ?

PythonGuython, 2021-07-30:
That fixed the issue - figured it was something simple I was missing... Thank you for the help

closed 2021-07-30 by PythonGuython

alexvatti, 2022-02-25:
yocto+poky+zeus: cross compiliton / build for arm:
Program g-ir-scanner found: NO
src/meson.build:297:2: ERROR: Program(s) ['g-ir-scanner'] not found or not executable

Can you help me to resolve the issue

EmmanuelP, 2022-02-25:
@alexvatti Please open a new issue.