AravisProject/aravis #546: timeout & error on stream reception (e.g. when blocked by firewall)?

eudoxos, 2021-07-28:
I am able to run the viewer in Windows now, and can get images form the camera quite nicely - around 70MB/s (1G network) with a direct table, the CPU core (1 out of 4) being at about 60%.

The error count is quite high, but does it indicate a problem at all? I actually see many more of them when I run under Linux (different machine, though, that might be an issue).

[image]

A thing which would be nice to have is to see some obvious error message when the stream packets don't arrive within some timeout. Since I am not used to Windows, I forget about the private/public network distinction, and forget about the firewall blocking all inbound traffic by default (so the stream packets never reach aravis, even though they do show up in Wireshark)

EmmanuelP, 2021-07-29:
> The error count is quite high, but does it indicate a problem at all?

3016 errors for 2518 completed buffers, that is pretty terrible. The number of missing packets seems to be quite high, and may be the packet resend mechanism on Windows is not as good as on linux. It relies heavily on get_real_time, I don't know if this function is accurate enough on Windows.

Do you get the same error rate when using `arv-camera-test` ?

> I actually see many more of them when I run under Linux

Using my laptop, using fedora 34, I can get a stream that uses almost all the 1G bandwidth, with a very low error rate. The performance is also quite good on a macOS laptop from 2015.

> A thing which would be nice to have is to see some obvious error message when the stream packets don't arrive within some timeout.

Yes, a in-app notification would be nice, and would also be useful to indicate a `save to file` failure. Unfortunately, there is no simple API in gtk for that.

eudoxos, 2021-07-29:
> Do you get the same error rate when using `arv-camera-test` ?

This is for direct 1:1 cable connection:
```
Looking for the first available camera
vendor name            = Point Grey Research
model name             = Blackfly BFLY-PGE-20E4M
device serial number   = 16162100
image width            = 1600
image height           = 1200
horizontal binning     = 1
vertical binning       = 1
exposure               = 23358.6 ?s
gain                   = 3 dB
payload                = 1920000 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 3200 ns
gv packet size         = 1400 bytes
 34 frames/s -    65.3 MiB/s - 2 errors
 44 frames/s -    84.5 MiB/s
 39 frames/s -    74.9 MiB/s - 4 errors
 34 frames/s -    65.3 MiB/s - 9 errors
 43 frames/s -    82.6 MiB/s
 43 frames/s -    82.6 MiB/s - 1 error
n_completed_buffers    = 237
n_failures             = 16
n_underruns            = 0
n_timeouts             = 16
n_aborteds             = 0
n_missing_frames       = 0
n_size_mismatch_errors = 0
n_received_packets     = 340761
n_missing_packets      = 19335
n_error_packets        = 32
n_ignored_packets      = 0
n_resend_requests      = 940
n_resent_packets       = 37
n_resend_ratio_reached = 13
n_resend_disabled      = 32
n_duplicated_packets   = 2
n_transferred_bytes    = 466738552
n_ignored_bytes        = 0
```

and this is going through GBit switch:

```
Looking for the first available camera
vendor name            = Point Grey Research
model name             = Blackfly BFLY-PGE-20E4M
device serial number   = 16162100
image width            = 1600
image height           = 1200
horizontal binning     = 1
vertical binning       = 1
exposure               = 18609.3 ?s
gain                   = 1 dB
payload                = 1920000 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 3200 ns
gv packet size         = 1400 bytes
 30 frames/s -    57.6 MiB/s - 1 error
 39 frames/s -    74.9 MiB/s - 4 errors
 35 frames/s -    67.2 MiB/s - 8 errors
 43 frames/s -    82.6 MiB/s
 41 frames/s -    78.7 MiB/s - 2 errors
 34 frames/s -    65.3 MiB/s - 9 errors
 41 frames/s -    78.7 MiB/s - 3 errors
n_completed_buffers    = 263
n_failures             = 27
n_underruns            = 0
n_timeouts             = 27
n_aborteds             = 0
n_missing_frames       = 0
n_size_mismatch_errors = 0
n_received_packets     = 385300
n_missing_packets      = 32239
n_error_packets        = 12
n_ignored_packets      = 0
n_resend_requests      = 1584
n_resent_packets       = 61
n_resend_ratio_reached = 17
n_resend_disabled      = 12
n_duplicated_packets   = 0
n_transferred_bytes    = 527746108
n_ignored_bytes        = 0
```

EmmanuelP, 2021-07-29:
> > A thing which would be nice to have is to see some obvious error message when the stream packets don't arrive within some timeout.
> 
> Yes, a in-app notification would be nice, and would also be useful to indicate a `save to file` failure. Unfortunately, there is no simple API in gtk for that.

After some more thought about that, it is not that easy to detect a possible streaming issue, in particular if image acquisition is triggered by an external signal.

EmmanuelP, 2021-07-29:
The missing packet rate is a bit higher when there is a switch, but in bot cases, it is almost 10% of the stream packets.
You may want to try to increase the socket buffer size. May be an automatic socket buffer adjustment should be the default.

Is the CPU us of arv-camera-test in the same range as the one on linux ?

Do you know if there is an Windows API we could use to increase the stream thread priority ?

eudoxos, 2021-07-29:
Okay, so with `--frame-retention 1000` and `--packet-timeout 1000` I consistently have zero errors. I tried the thread priority (manually in task manager, also programmatically, but that always failed — I will anyway push the code in case someone ever finds it useful) and that had no visible effect. Yes, the Linux machine is on the same network.

Before it gets forgotten, this is a snippet to use high-precision clock under Windows (which also, IIRC, did not have any significant effect): https://github.com/AravisProject/aravis/pull/422#issuecomment-785988622 .

eudoxos, 2021-12-17:
This issue overlaps significantly with #593 which is resolved now, so I am closing it. We might create some platform-specific tips page later, once Windows is used more widely.

closed 2021-12-17 by eudoxos