AravisProject/aravis #549: Commit "use proper broadcast addresses in discovery for each network" broke device discovery

abrock, 2021-08-03:
**Describe the bug**

Commit 56cd617f0198c0981c64e6a96fe6dce128419699 broke camera discovery on my system.

**To Reproduce**

Try to discover GigE cameras using arv-tool. With the broken commit no packages show up in wireshark. With the commit before discovery (and later image capture) works as expected.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Camera description:**
 - Baumer
 - [VLXT-123M.I](https://www.baumer.com/de/de/produktubersicht/industriekameras-bildverarbeitung/industriekameras/lx-serie/vlxt-kameras/10-gige-kameras-mit-kupferkabel-10gbase-t-/vlxt-i/vlxt-123m-i/p/38562)
 - 10GigE

**Platform description:**
 - 598b396f3b11ca16432c3a87d09cb81de7ff866e
 - OS: [Ubuntu 20.04]
 - x86_64

**Additional context**

I did a git bisect to find the bad commit.
I have 18 cameras connected using a switch divided into VLANs. When I run arv-tool with the first bad commit I see no packages at all, when I use the commit before everything works as expected including image capture.

I attached wireshark captures, the output of "ip a" and the output of "route -n".
[aravis-bugreport.zip](https://github.com/AravisProject/aravis/files/6922456/aravis-bugreport.zip)

EmmanuelP, 2021-08-03:
Hi,

I guess in your setup there is no broadcast address associated to your vlan. Is that right ?

We need to re-add the global broadcast address the commit has suppressed.

abrock, 2021-08-03:
> I guess in your setup there is no broadcast address associated to your vlan. Is that right?

I don't know. Is that a property of the computer or the switch? How do I find out?

EmmanuelP, 2021-08-03:
I should confess that despite developping a library implementing an ethernet based protocol, I dont know much about network configuration. My understanding is the computer running Aravis should have an IP address corresponding to the camera Vlan. Using `ifconfig`, the block about this IP address should contain a broadcast address. If it not the case, the current code ill not be able to send a discovery broadcast packet to the cameras.

abrock, 2021-08-03:
I will post the output of ifconfig as soon as I have access to the computer again (probably next week since the power supply died today).

My understanding of VLANs is that from the perspective of the computer it looks as if all the different network interfaces are connected to different switches even though it's just one large switch.

EmmanuelP, 2021-08-04:
When you will have the chance to access to the computer, please run `./src/arv-tool-0.8 -d all` and paste the output here.

EmmanuelP, 2021-08-08:
The discovery of my TheImagingSource cameras is also broken. My guess is these devices only answer to discovery from the general broadcast address, not the interface one.

PR #552 (2021-08-11): gvinterface: use local brodcast, and fallback to directed on error
Fix #549
commit 5dea5cdf (2021-08-11): gvinterface: use local brodcast, and fallback to directed on error

Fix #549
closed 2021-08-11 by EmmanuelP