AravisProject/aravis #554: Change "Width, Height and PixelFormat" in GStreamer filter aravissrc

robertofem, 2021-08-15:
**Describe the bug**
I am using Gstreamer to build a pipeline that reads data from a camera and prints them to screen (later to save them to mp4).
I need to change more parameters than those available through aravissrc filter properties, like Height, Width and PixelFormat (I want RGB image and always get Mono).
I have a working pipeline with aravissrc, queue, videoconvert and xvimagesink. I try to get the aravis camera object (AravisCam) and device object to later use the Aravis functions (arv_device_set_integer_feature_value, arv_camera_set_region, etc.) to modify the settings of the camera (these configuration functions work when I use only aravis without gstreamer). However when I do that and check for returned value with ARV_IS_CAMERA it is never a correct camera). It does the same regardless where I put this piece of code after creating elements, after linking them in the pipeline and even after starting the pipeline. 

I have to include #include <arv.h> in the file in order to find ArvCamera object. I wonder if this is needed or there is a possibility to use Aravis objects just using the gst.h file. 

**To Reproduce**
I just add the following aravis code in a working pipeline after creating Gst elements and after linking them Neither works.

```
#include <gst/gst.h>
#include <arv.h> 
....
ArvCamera* cam;
g_object_get (arvsrc, "camera", &cam, NULL);
if (ARV_IS_CAMERA (cam)) g_printerr ("It is a camera.\n");
else g_printerr ("It is not a camera.\n");
```

Compilation command

```
cc arv2screen.c -o arv2screen `pkg-config --cflags --libs gstreamer-1.0` `pkg-config --cflags --libs aravis-0.6`
```

**Expected behavior**
To get a correct ArvCamera pointer of the internal camera handler used by the Gstreamer aravissrc plugin.

**Camera description:**
 - Manufacturer: iDS
 - Model: GV-5860CP-C-HQ-R2
 - Interface: Gige

**Platform description:**
 - Aravis version: 0.8.0 and 0.6.0 on along with Gstreamer 1.0
 - OS: Ubuntu 20.04
 - Hardware: x86_64

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2021-08-16:
The `ArvCamera` object returned by `g_object_get (arvsrc, "camera", &cam, NULL);` is only valid after the pipeline has started.

But aravissrc has also a `features` property that lets you set additional features as a space separated list of feature assignations. The string is sent as is to [arv_device_set_features_fro_string()](https://aravisproject.github.io/docs/aravis-0.8/ArvDevice.html#arv-device-set-features-from-string), just before the stream creation.

Does that solve your issue ?

EmmanuelP, 2021-08-16:
Please note that after the acquisition is started, there is some features that can no be changed, like `Width`, `Height` or `PixelFormat`.

robertofem, 2021-08-16:
Hello, thanks for the quick and accurate response (and for the aravis library). 

I am using gstreamer-1.0 downloaded with apt-get, which uses aravis 0.6, which does not have *featrures* property in it. But I can see the master version has it. I will compile aravis master, make gstreamer use the newest aravissrc and let you know if it works (I dont know if it is better to just install a newer gstreamer).

robertofem, 2021-08-28:
Hello,

I uninstalled aravis 0.6 and installed 0.8 compiling from source to have the "features" property. Now I get this error when running gst-launch-1.0 aravissrc ! videoconvert ! xvimagesink :+1: 

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Could not set caps on camera "": Value not found in <Enumeration> 'TriggerSelector'
Additional debug info:
../gst/gstaravis.c(369): gst_aravis_set_caps (): /GstPipeline:pipeline0/GstAravis:aravis0
Execution ended after 0:00:00.198166929
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

Aravis seems to work fine since "arv-camera-test-0.8" runs fine.
The new aravis 0.8.11 plugin gets recognized by "gst-inspect-1.0 aravis".

Is there any incompatibility in aravis 0.8.11 and the version o gstreamer-1.0 used (currently I am running gstreamer core 1.16.2, the one installed with ubuntu 20.04 and apt-get)? Should I update gstreamer?