AravisProject/aravis #558: FLIR Firefly - camera not found

drmcnelson, 2021-08-17:
Aravis fails to recognize the FLIR Firefly on Fedora

This is a popular camera, unfortunately only supported by FLIR with proprietary libraries on Ubuntu.  It would be a big help to be able use the Firely it with Aravis on Fedora.

EmmanuelP, 2021-08-18:
Hi @drmcnelson 

You have to tell us more about your issue.

- What model are you using ?
- What version of Aravis are you using ?
- What is the output of `arv-camera-test-0.8 -d all` ?

drmcnelson, 2021-08-18:
FFY-U3-04S2M-S

Aravis 0.8.16

# arv-camera-test-0.8 -d all

Looking for the first available camera
[12:10:01.828] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[12:10:01.828] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.0.31 (192.168.0.255)
No camera found: No supported device found

# lsusb

Bus 004 Device 004: ID 1e10:4000 Point Grey Research, Inc. U3V camera

# lsusb -v

(excerpted)

Bus 004 Device 004: ID 1e10:4000 Point Grey Research, Inc. U3V camera
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0         9
  idVendor           0x1e10 Point Grey Research, Inc.
  idProduct          0x4000 
  bcdDevice            0.00
  iManufacturer           1 FLIR
  iProduct                2 U3V camera
  iSerial                 3 19515069
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0074
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              152mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         3
      bFunctionClass        239 Miscellaneous Device
      bFunctionSubClass       5 USB3 Vision
      bFunctionProtocol       0 
      iFunction               4 USB3 Vision Device
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      0 
      iInterface              5 Control Interface
      ** UNRECOGNIZED:  14 24 01 00 00 01 00 00 00 01 00 06 07 08 09 0a 0b 03 0c 0c
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      1 
      iInterface             13 Event Interface
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass       239 Miscellaneous Device
      bInterfaceSubClass      5 USB3 Vision
      bInterfaceProtocol      2 
      iInterface             14 Streaming Interface
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst              15
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0016
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x00000006
      BESL Link Power Management (LPM) Supported
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat          32 micro seconds
Device Status:     0x000c
  (Bus Powered)
  U1 Enabled
  U2 Enabled

EmmanuelP, 2021-08-18:
It looks like aravis is compiled without usb support. Make sure you have the development package of libusb installed on your system.

drmcnelson, 2021-08-18:
Wonderful, thank you.  How do I recompile with libusb-devel?   I am not familiar with the meson build environment.

drmcnelson, 2021-08-18:
Never mind, I deleted the director and started fresh.   The two programs camera test and tool work now.  Thank you.

I want to trigger and capture single frames.  Where do I find documentation or example code to do that?

Is there already python support?   This has to get integrated with an existing Python based data acquisition system.

Thank you again

EmmanuelP, 2021-08-18:
> I want to trigger and capture single frames. Where do I find documentation or example code to do that?

The source code of `arv-camera-test` is small enough to be used as a starting point. Have a look at `src/arvcameratest.c`.
Or you can look at the source code of the different test program ins `tests`.

> Is there already python support? This has to get integrated with an existing Python based data acquisition system.

Python support is provided by `gobject-introspection`. Make sure you have it installed and correctly detected during Aravis build. It also has to be installed using `/usr` prefix, or else you will have to set some paths in order to be able to import the aravis module. There is also some examples of python use in `tests`.

closed 2021-08-18 by EmmanuelP

drmcnelson, 2021-08-18:
Okay, great thank you.  Have to ask again, how do you trigger a re-build in the meson/nemo environment?

drmcnelson, 2021-08-18:
Looks like  meson --reconfigure  and then meson configure prefix /usr ?

drmcnelson, 2021-08-18:
Got it,  it's  meson -Dprefix=/usr  build,  voila, thank you