AravisProject/aravis #559: Free buffers

drmcnelson, 2021-08-22:
A typical use of the library seems to be allocate some buffers, push them on to the stream and start acquisition.   The call to try_pop_buffer() then returns a filled buffer when available.

What seems to be needed is a way to free the buffers that were pushed.

In particular I am interested in having a call for this available in the python interface.

Thank you

EmmanuelP, 2021-08-23:
Hi,

There is normally no need to free buffers. You just have to push them back to the stream thread. They will be freed upon ArvStream destruction.

For future questions about Aravis use, please post them on Aravis discourse instance: https://aravis-project.discourse.group/

Cheers.

closed 2021-08-23 by EmmanuelP

drmcnelson, 2021-08-23:
Hi

What happens if the payload size changes?

It seems from code examples that the buffer size is matched to the payload,
and the payload matches the size of the ROI.

And the ROI could be smaller or larger from one moment to the next

EmmanuelP, 2021-08-23:
Le lun., août 23 2021 at 04:44:34 -0700, drmcnelson 
***@***.***> a écrit :
> What happens if the payload size changes?
> 

Aravis does not support a change in the payload size during an 
acquisition sequence.

	Emmanuel.

drmcnelson, 2021-08-23:
What about different acquisition sequences?

Meaning, finish one acq, change the ROI and start another.

EmmanuelP, 2021-08-23:
Le lun., août 23 2021 at 04:58:57 -0700, drmcnelson 
***@***.***> a écrit :
> What about different acquisition sequences?
> 
>  Meaning, finish one acq, change the ROI and start another.

Please have a look at `tests/arvroitest.c`.

	Emmanue.

drmcnelson, 2021-08-23:
Very good, thank you.  Is there a python equivalent?

It looks like del(stream) might do it.   Is that how you intended it should
be done?

EmmanuelP, 2021-08-23:
Le lun., août 23 2021 at 05:08:59 -0700, drmcnelson 
***@***.***> a écrit :
> Very good, thank you. Is there a python equivalent?
> 

No, sorry.

>  It looks like del(stream) might do it. Is that how you intended it 
> should
>  be done?

arv_stream_stop_thread (..., TRUE) will delete all the buffers it has 
reference to.

	Emmanuel.

drmcnelson, 2021-08-23:
Thank you, I will try that.

And thank you for making Aravis available,  It enables the next step in my
research.