AravisProject/aravis #563: arv_dom_node_list_get_length does not work on ArvDomNodeList*

romainreignier, 2021-09-01:
**Describe the bug**
While updating a program working on aravis 0.6, I have noticed that the function `arv_dom_node_list_get_length` does
not return the size of the list passed by argument but returns 0 with the output:

```
(process:5730): GLib-GObject-WARNING **: 15:28:53.873: invalid cast from 'ArvDomNodeChildList' to 'ArvDomNode'

** (process:5730): CRITICAL **: 15:28:53.874: arv_dom_node_get_parent_node: assertion 'ARV_IS_DOM_NODE (self)' failed

** (process:5730): CRITICAL **: 15:28:53.874: arv_dom_node_get_first_child: assertion 'ARV_IS_DOM_NODE (self)' failed
```

**To Reproduce**
Build this minimal code with aravis 0.8

```c
#include <stdio.h>
#include <string.h>

#include <arv.h>

int main(void) {
  ArvCamera *camera;
  ArvDevice *device;
  ArvGc *genicam;
  ArvGcNode *node;
  ArvDomNodeList *children;
  ArvDomNode *item;
  unsigned int length, i;

  arv_update_device_list();
#if ARAVIS_CHECK_VERSION(0, 8, 0)
  camera = arv_camera_new(NULL, NULL);
#else
  camera = arv_camera_new(NULL);
#endif
  if (camera != NULL) {
    device = arv_camera_get_device(camera);

    genicam = arv_device_get_genicam(device);

    node = arv_gc_get_node(genicam, "Root");

    children = arv_dom_node_get_child_nodes(ARV_DOM_NODE(node));
    length = arv_dom_node_list_get_length(children);
    for (i = 0; i < length; ++i) {
      item = arv_dom_node_list_get_item(children, i);
      printf("- %s\n", arv_dom_node_get_node_name(item));
    }

  } else {
    printf("No camera found\n");
  }

  arv_shutdown();
  return 0;
}
```

And run it with a `arv-fake-gv-camera` instance running.

**Expected behavior**
On aravis 0.6, the following output is obtained:
```
- pFeature
- pFeature
- pFeature
- pFeature
- pFeature
```

**Camera description:**
 - Aravis
 - Fake camera

**Platform description:**
 - Aravis 0.8.x (master branch today)
 - OS: Ubuntu 18.04
 - Hardware x86_64

**Additional context**
I have found this issue while porting the [ROS driver](https://github.com/fkie-forks/camera_aravis/blob/cec75c3538e92d37ffd6231ed97e25d98460c580/src/camera_aravis_nodelet.cpp#L1476) to aravis 0.8.

EmmanuelP, 2021-09-03:
That is what happen when code is not under CI control.

Please test https://github.com/AravisProject/aravis/tree/issue/563%2Fchild-list-fix

EmmanuelP, 2021-09-03:
Thanks for the report and the sample code, by the way. It really helps.

romainreignier, 2021-09-03:
I have tested the branch and it works as expected. Thank you.

commit 11250677 (2021-09-03): dom: child list is not a DOM node

Fix #563
closed 2021-09-03 by EmmanuelP