AravisProject/aravis #564: Add TriggerSoftware command to fake camera

romainreignier, 2021-09-01:
**Is your feature request related to a problem? Please describe.**
I would like to test an application using the Software trigger source on the fake camera because I do not have the real camera
with me.

**Describe the solution you'd like**
The fake camera would support `Software` `TriggerSource` and the `TriggerSoftware` command.

**Describe alternatives you've considered**
None

**Additional context**
I am volunteering to try implementing this feature but I am more a C++ than C developer and I have never worked with glib and Aravis before so any advice is welcomed.

From what I have seen, I first need to add some register values and a command in both `arvfakecamera.h/.c` and `arv-fake-camera.xml`.
Then wait on a `GCond` in `arv_fake_camera_wait_for_next_frame` in case the `TriggerSource` is set to `Software`.
And signal it in the thread of `arvgvfakecamera.c`.

Does this sound good for you?

EmmanuelP, 2021-09-03:
Hi Romain,

Software trigger would be a nice addition to the fake camera.

I must warn you the simulation code is not the best part of aravis code base. It was meant mostly as a debugging tool.

There is one thread that waits for incoming control packets and sends image data at the configured frame rate. You don't need to use GCond, as everything happen in this thread. Once you have implemented the code that handles `TriggerSource` and `TriggerSoftware`, you can implement a `arv_fake_camera_check_and_acknowledge_software_trigger()` function, that will check if the corresponding register and reset the value, and return true if a trigger was requested, and use this function in arv_gv_fake_camera `_thread`.

romainreignier, 2021-09-03:
Thank you for the indications, its seems simpler to only use data in one thread without synchronization.

But I am not sure to be able to implement it correctly. I will try and let you know.

romainreignier, 2021-09-03:
I have started to make some changes in my fork: https://github.com/AravisProject/aravis/compare/master...romainreignier:feature/add_trigger_software_to_fake_camera

But sending a `TriggerSoftware` command does not change the register on the fake camera.

For example, if I start the fake gv camera with debug info and send the command arv-tool, there is no write to the register `0x30c`:

```
./src/arv-tool-0.8 -d genicam:3 control TriggerSoftware
[...]
[14:12:09.614] 🅳 genicam> [GcCommand::execute] TriggerSoftware (0x1)
TriggerSoftware executed
```

```
./src/arv-fake-gv-camera-0.8 -d all
[14:12:06.330] 🅸 device> GVSP address = 127.0.0.1
[14:12:06.330] 🅸 device> GVCP address = 127.0.0.1:3956
[14:12:06.331] 🅸 device> Global discovery address = 255.255.255.255:3956
[14:12:06.331] 🅸 device> Listening to 2 sockets
[14:12:08.581] 🅸 device> [GvFakeCamera::handle_control_packet] Discovery command
[14:12:08.581] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:08.582] 🅸 device> [GvFakeCamera::handle_control_packet] Discovery command
[14:12:08.582] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:08.582] 🅸 device> [GvFakeCamera::handle_control_packet] Discovery command
[14:12:08.582] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:08.582] 🅸 device> [GvFakeCamera::handle_control_packet] Discovery command
[14:12:08.582] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:08.584] 🅸 device> [GvFakeCamera::handle_control_packet] Discovery command
[14:12:08.584] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.586] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 512 (512)
[14:12:09.586] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.587] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 65536 (512)
[14:12:09.587] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.587] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 66048 (512)
[14:12:09.587] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.587] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 66560 (512)
[14:12:09.587] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.587] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 67072 (512)
[14:12:09.588] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.588] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 67584 (512)
[14:12:09.588] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.588] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 68096 (512)
[14:12:09.588] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.588] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 68608 (512)
[14:12:09.588] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.588] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 69120 (512)
[14:12:09.589] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.589] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 69632 (512)
[14:12:09.589] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.589] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 70144 (512)
[14:12:09.589] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.589] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 70656 (512)
[14:12:09.589] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.589] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 71168 (512)
[14:12:09.590] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.590] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 71680 (512)
[14:12:09.590] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.590] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 72192 (512)
[14:12:09.590] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.590] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 72704 (512)
[14:12:09.590] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.590] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 73216 (512)
[14:12:09.591] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.591] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 73728 (512)
[14:12:09.591] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.591] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 74240 (512)
[14:12:09.591] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.591] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 74752 (512)
[14:12:09.591] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.592] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 75264 (512)
[... 20 lines cut ...]
[14:12:09.594] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.594] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 80896 (512)
[14:12:09.594] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.594] 🅸 device> [GvFakeCamera::handle_control_packet] Read memory command 81408 (96)
[14:12:09.594] 🅸 device> [GvFakeCamera::thread] Control packet received
[... cut ...]
[14:12:09.615] 🅸 device> [GvFakeCamera::thread] Control packet received
[14:12:09.615] 🅸 device> [GvFakeCamera::handle_control_packet] Write register command 2560 -> 0
[14:12:09.615] 🅸 device> [GvFakeCamera::handle_control_packet] Controller releases
[14:12:09.615] 🅸 device> [GvFakeCamera::thread] Control packet received
```

While for a `AcquisitionStart` request, I can see the `292` register being written:
```
[14:13:20.000] 🅸 device> [GvFakeCamera::handle_control_packet] Write register command 292 -> 1
```

Do you have an idea what is missing?

romainreignier, 2021-09-03:
Sorry, I have found the issue, there was a mismatch with the register name.