AravisProject/aravis #574: Pleora iPort-SB-GigE support

EmmanuelP, 2021-09-09:
@c10ud wrote:

I cannot see video coming from a Pleora iPort-SB-GigE device connected to a Sony Block

Here is the output of arv-tool-0.8 values: https://pastebin.com/eCLQEszp
And the output of arv-camera-test-0.8 -d all: https://pastebin.com/Jaku5CiJ
And the output of
GST_DEBUG=aravissrc:6 gst-launch-1.0 aravissrc camera-name="Pleora Technologies Inc.-iPORT-SB-GigE-PT01-SN4IP01-128x-00:11:1c:03:a1:70" ! video/x-raw,format=GRAY8,width=1280,height=720,framerate=30/1 ! videoconvert ! autovideosink

https://pastebin.com/P7JbSHZn

Thanks

EmmanuelP, 2021-09-09:
Hi @c10ud,

When you are not sure, it is better to open a new issue, with a link to the one you think it is related to.

In this case, I don't think it is related to #470 

```
[11:04:00.155] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 64
[11:04:00.345] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 55
[11:04:00.396] 🅸 device> [GvDevice::auto_packet_size] Packet size set to 64 bytes
 
** (arv-camera-test-0.8:23672): CRITICAL **: 11:04:00.416: arv_gc_integer_get_value: assertion 'error == NULL || *error == NULL' failed
gv packet size         = 0 bytes
```

The packet size is wrong. Sometimes it is linked to the newtork setup. How is the device connected to your machine ? Do you have a WIFI interface ? If so, did you try with WIFI disabled ?

c10ud, 2021-09-09:
Hello EmmanuelP, sorry for hijacking the other issue then: no wifi, it is an ubuntu 18.04 vm with two ethernet interfaces (one nat, one bridged)
the device is connected with one of those interfaces with a simple switch (and works with the Pleora-furnished test application -- on Windows host)

EmmanuelP, 2021-09-09:
Ok. Firewall may be ?

c10ud, 2021-09-09:
there is no firewall (that i know of).. windows firewall is deactivated for all interfaces, and the ubuntu ethernet interface is directly bridged on the physical ethernet port..

I am testing this because my end goal is trying aravis on a nvidia jetson board, if you think this could be related to the VM then I will hopefully make some tests on bare metal.

otherwise if you have some idea that I could try to see what is failing (sorry I just found aravis and started playing with it, great project by the way)

EmmanuelP, 2021-09-09:
Oh, I did not see it was from a virtual machine. So yes, please try from a real machine.

c10ud, 2021-09-09:
OK, I managed to try with the msys2 build in my host Windows and arv-camera-test-0.8 gives the same issue.

I managed to track back the issue to packet size not being negotiated correctly. in fact, test_packet_check() is failing because g_poll() is not receiving any event

with wireshark I see there's an exchange between the grabber and aravis, however I'm still too new to this to understand if it's replying back correctly..

c10ud, 2021-09-09:
[pleora_sony.zip](https://github.com/AravisProject/aravis/files/7137840/pleora_sony.zip)

Attached the wireshark capture of an unsuccessful arv-camera-test-0.8

Thanks

c10ud, 2021-09-10:
Hello Emmanuel,
I managed to make some progress by changing, in arvgvdevice.c:

//	arv_device_set_integer_feature_value (ARV_DEVICE (gv_device), "GevSCDA", g_htonl (*((guint32 *) address_bytes)), NULL);
	
	arv_device_write_register (ARV_DEVICE (gv_device),
					     ARV_GVBS_STREAM_CHANNEL_0_IP_ADDRESS_OFFSET,
					     g_ntohl(*((guint32 *) address_bytes)),
					     error);
	//arv_device_set_integer_feature_value (ARV_DEVICE (gv_device), "GevSCPHostPort",  port, NULL);

	arv_device_write_register (ARV_DEVICE (gv_device),
					     ARV_GVBS_STREAM_CHANNEL_0_PORT_OFFSET,
					     port,
					     error);

now I'm trying to understand why the test_packet_check() is still failing (it's receiving events now, and I see in wireshark it is communicating correctly with the device).

EmmanuelP, 2021-09-10:
Did you try to set the packet size and disable packet size check when running `arv-camera-test` ?

```
arv-camera-test-0.8 -j never -i 1500
```

c10ud, 2021-09-10:
Yes it was one of the tests I did before trying to dig deeper, however if didn't manage to force the packet size (it remained 64 bytes) and it said I was receiving 0 frame/s (0 MiB/s)

c10ud, 2021-09-13:
OK, some progress: converting the packet_size settings to register instead of memory writes lets me correctly start the stream with arv-camera-test-0.8!

Now I am trying to show some picture (with arv-viewer or gstreamer plugin directly), however it seems there is some issue with negotiating caps..

See sample logs for gstreamer: https://pastebin.pl/view/raw/a411fec3

I'm thinking this could be a similar issue to the one memwrite vs regwrite also for PixelFormat, or some issue with byte ordering...however I'm unable to understand how this setting is done on the device.. (I see on wireshark the device is replying with "malformed request" invalid parameter

EmmanuelP, 2021-09-13:
Did you have a look at this issue: https://github.com/AravisProject/aravis/issues/179 ?

c10ud, 2021-09-13:
ouch! I did not see that.. I will definitely look into it and see if it's
the same issue..

Thanks,
Riccardo

c10ud, 2021-09-14:
Hello Emmanuel, that indeed fixed the issue!

Thanks,
R

closed 2023-01-26 by EmmanuelP