AravisProject/aravis #578: USB3 Vision  Teledyne Lumerera

aminArlut, 2021-09-15:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Camera description:**
 - Manufacturer
 - Model
 - Interface [e.g. USB3]

**Platform description:**
 - Aravis version
 - OS: [e.g. Fedora 20]
 - Hardware [e.g. x86_64]

**Additional context**
Add any other context about the problem here.

aminArlut, 2021-09-15:
This is my first github issue, please let me know if any other information is needed.

EmmanuelP, 2021-09-16:
Hi,

Thanks for the report. For now, all the relevant informations are present.

I will have a look in case an obvious error can be fixed in Aravis, but as I don't have any device from this manufacturer, chances are odd.

aminArlut, 2021-09-16:
EmmanuelP, 

Thanks for the fast reply, I realize it will be difficult without the hardware. I tried digging around in the code myself, I know the failures start to creep in [here](https://github.com/AravisProject/aravis/blob/a76e1dabe89e0b132817d5bd32983bff8f4d1866/src/arvuvdevice.c#L469) if this means anything to you. 

FWIW, I talked with the Teledyne Lumenera team and they said that "Stemmer Common Vision Blocks for Linux" has been known to work for this camera. I installed that on a debian10 VM and I was able to verify I could pull frames off the camera. I installed the lastest aravis on the same debian10 image and ran into the same problems as above. 

I'd rather use aravis for my applications since it's the best genicam open source library in existence right now.

EmmanuelP, 2021-09-17:
> Thanks for the fast reply, I realize it will be difficult without the hardware. I tried digging around in the code myself, I know the failures start to creep in [...]

This line reads the SBRM_U3VCP_CAPABILITY register,at an address defined by the ABRM_SBRM_ADDRESS. 0x10000 seems a fine SBRM offset, so there is no reason a read to 0x10004 fails.

> FWIW, I talked with the Teledyne Lumenera team and they said that "Stemmer Common Vision Blocks for Linux" has been known to work for this camera. I installed that on a debian10 VM and I was able to [...]

What you may try is to use wireshark and capture the USB3 data during a connection to the camera using Stemmer Common Vision Blocks for Linux (which is in fact CVB for Ubuntu :( ). I may be able to figure out what is wrong.

aminArlut, 2021-09-17:
I wasn't able to upload the wireshark session on github, so I placed the session on [a google drive here](https://drive.google.com/drive/folders/1dWCGrzcUQf1Z4g2oU-laeWVEhanIbT6K?usp=sharing) titled "cvb_capture.pcapng". Let me know if you aren't able to access it or need another session.

EmmanuelP, 2021-09-18:
Thanks a lot. It would also be helpful if you could do the same capture using aravis.

aminArlut, 2021-09-20:
[aravis_capture.zip](https://github.com/AravisProject/aravis/files/7198393/aravis_capture.zip)

No problem, I provided the aravis capture here. Same debian10 VM and test conditions as taken with the cvb capture. Let me know if there is anything else you need.

victhor393, 2023-02-25:
Hi, I have a Lt29059, which suffers from this same bug. By changing the "size" argument of the function that reads ARV_SBRM_U3VCP_CAPABILITY to a type with size 8 (guint64), I got my camera to work with arv-viewer: https://github.com/AravisProject/aravis/blob/5e80a7950ef122626b64760e399e642f7a1b5dee/src/arvuvdevice.c#L512

This was the only difference that I saw between the CVB capture and the aravis capture provided.

EmmanuelP, 2023-02-26:
Hi @victhor393 

Good catch. Please test #768

victhor393, 2023-02-26:
Tested #768, it works for me.

EmmanuelP, 2023-02-27:
> Tested #768, it works for me.

Thanks.

PR #768 (2023-02-27): uvdevice: fix srbm_u3vcp_capability register size
It is 8 byte long according to wireshark u3v dissector.

Fixes #578