AravisProject/aravis #581: header files installation missing

mchalain, 2021-09-22:
**Describe the bug**
If the USB3 Vision is not present, the arvuv*.h are not installed.
The file arv.h and arvcamera.h require arvuvdevice.h,  arvuvinterface.h and arvuvstream.h.

**To Reproduce**
 - disable the usb support on meson.
 - build aravis library with ninja.
 - install aravis library and header files with ninja.
 - create a new project which use aravis library and retrieve the information on the installation with pkg-config.
 - write a C file which include arv.h
 - build the project
 fatal error: arvuvdevice.h: Aucun fichier ou dossier de ce type

**Expected behavior**
A binary of the project ;-)

**Camera description:**
Useless

**Platform description:**
 - Aravis version: 0.8
 - OS: Ubuntu
 - Hardware: x86_64

**Additional context**
Les Bauges :D

EmmanuelP, 2021-09-22:
Hi,

Thanks for the report.

`arvuv*` file inclusion is dependent on ARAVIS_HAS_USB in `arv.h`. Could you paste the complete build log of your project ? And the content of `arvfeatures.h` ?

Emmanuel.

> **Additional context**
> Les Bauges :D

:smile:

mchalain, 2021-09-23:
Hi,
Thank you, I found that the USB3Vision API is not protected inside arvcamera.h
I pushed a PR with a simple correction.

Thank you for your greate work

closed 2021-11-17 by EmmanuelP