AravisProject/aravis #582: Support for GainAbs register as part of arv_camera_*_gain()

hackel6, 2021-09-22:
In my experience, PixeLINK cameras utilize feature name "GainAbs" rather than either of "Gain" or "GainRaw" to hold their Float gain values. Rather than having to access this register directly using it's name, it would be much nicer to roll support for this into the existing arv_camera_*_gain() methods.

as an example:
```
> ./arv-tool-0.8 features
PixeLINK Inc.-PL-B959G - GEV Camera
<snip!>
     Category    : 'Error'
        Category    : 'IPEngineCamHeadSerialComLog'
        Category    : 'BlockDiscard'
    Category    : 'AnalogControls'
        Float       : 'GainAbs'
        Float       : 'Gamma'
        Float       : 'CameraGainAbsMax'
    Category    : 'GigEVisionTransportLayer'
        Integer     : 'PayloadSize'
        Integer     : 'GevVersionMajor'
        Integer     : 'GevVersionMinor'
<snip!>
```

I have something that I believe works alright and could probably put together a pull request pretty easily if this is deemed appropriate.

EmmanuelP, 2021-09-23:
Hi,

> In my experience, PixeLINK cameras utilize feature name "GainAbs" rather than either of "Gain" or "GainRaw" to hold their Float gain values.

GainAbs was deprecated in Genicam SFNC document at least 10 years ago... :(

> I have something that I believe works alright and could probably put together a pull request pretty easily if this is deemed appropriate.

Yes, please.

hackel6, 2021-09-23:
> GainAbs was deprecated in Genicam SFNC document at least 10 years ago... :(

That does not surprise me. Many of the cameras (and other hardware) that I support are past end-of-life.
> Yes, please.

I will work on it, thanks!

hackel6, 2021-09-24:
Looking back through older versions of the GenICam SFNC, I see now where it has been deprecated. I also note that this particular camera's XML reports using GenICam standard 1.0 in the header. Thanks for bringing this to my attention!
Our camera control application also supports Brightness (also known as BlackLevel). It appears our old PixeLINK cameras utilize their own "Brightness" feature:
```
	<Group Comment="PixeLINKFeatures">
		<Category Name="PixeLINKFeatures">
			<pFeature>AutoFrameRate</pFeature>
			<pFeature>FeaturesInquiry</pFeature>
			<pFeature>Brightness</pFeature>
			<pFeature>BrightnessMin</pFeature>
			<pFeature>BrightnessMax</pFeature>
```

and even our "newer" Basler cameras utilize the deprecated "BlackLevelRaw" register
`Integer     : 'BlackLevelRaw'`

I am interested in supporting "brightness" in the same way other features are supported in ArvCamera: via arv_camera_*_brightness() methods which will utilize register names "Brightness," "BrightnessAbs," "BlackLevel," and "BlackLevelRaw." Would you support this addition? It would allow me to streamline the upstream code to use a single API call rather than having to put the particular register name in a config file specific to the camera mdoel.
Again, I should be able to provide a pull request once i have it implemented.
Thanks! And I really appreciate all the hard work you've put into this project. It has allowed my organization to begin a long-awaited transition off of Windows XP for camera control.

EmmanuelP, 2021-09-24:
Hi,

> Would you support this addition?

It sounds like a nice addition, generally useful.

closed 2021-11-17 by EmmanuelP