AravisProject/aravis #593: Many errors(ARV_BUFFER_STATUS_TIMEOUT） occurd when CPU usage up to 70%

garfeng, 2021-10-21:
**Describe the bug**

When CPU usage is up to 70%, the function `arv_stream_pop_buffer` will get ARV_BUFFER_STATUS_TIMEOUT error.

[image]

Is there any way to capture image succeed no matter how much other thread use the CPU. 

For example, my computer has 8 cores, aravis uses 1 of it, and other thread only use the other 7.  

Or any other way ?

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Camera description:**
 - Manufacturer: OPT/Hikrobot
 - Model:
 - Interface Gige

**Platform description:**
 - Aravis version: 0.8
 - OS: Win 10
 - Hardware: x86_64

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2021-10-21:
> Unfortunately, there is no implementation for Windows, nor macOS.

Oh, that is not true. Sorry about that. @eudoxos has added a Windows implementation last july. But I don't remember how efficient it is.

garfeng, 2021-10-21:
@EmmanuelP, thank you for your answer.

I tried this 
>Could you rerun arv-camera-test with -d all, stop it with ctr+c when you have missing frames, and attach the debug output ?
 
Then I get this info:

I have set the frame rate to 5fps, so that only 30% of  the bandwidth was used.

``` 
[14:43:19.255] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 43 (last: 44)
[14:43:19.255] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =       43
                     packet_id    =     1079
                     data_size    =     8184
[14:43:19.255] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 43 (last: 44)
[14:43:19.255] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =       43
                     packet_id    =     1080
                     data_size    =     8184
[14:43:19.255] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 43 (last: 44)
[14:43:19.255] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =       43
                     packet_id    =     1081
                     data_size    =     8184
[14:43:19.255] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 43 (last: 44)
[14:43:19.255] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =       43
                     packet_id    =     1082
                     data_size    =      456
[14:43:19.315] 🅸 stream-thread> [GvStream::missing_packet_check] Maximum number of requests reached at dt = 60289, n_packet_requests = 1036 (1084 packets/frame), frame_id = 44
[14:43:19.474] 🆆 stream-thread> [GvStream::check_frame_completion] Timeout for frame 44 at dt = 220084
[14:43:19.474] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 44 (last: 44)
[14:43:19.475] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =       44
                     packet_id    =      150
                     data_size    =     8184
[14:43:19.475] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 44 (last: 44)
[14:43:19.475] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =       44
                     packet_id    =      151
                     data_size    =     8184
[14:43:19.475] 🅸 stream-thread> [GvStream::find_frame_data] Discard late frame 44 (last: 44)
[14:43:19.475] 🅸 sp> packet_type  =       ok (0x0000)
                     content_type = data-block (0x0003)
                     frame_id     =       44
                     packet_id    =      152
                     data_size    =     8184

```

I will try this later. 
> Oh, that is not true. Sorry about that. @eudoxos has added a Windows implementation last july. But I don't remember how efficient it is.

Thank you !

eudoxos, 2021-10-21:
The Windows "RT" (higher-priority) code is here: 
https://github.com/AravisProject/aravis/blob/1f9f4710c155ca2007878c372062e95cc530f5c3/src/arvrealtime.c#L327-L355
but in my case it did not seem to make much (any, actually, IIRC) difference, as I was more limited by low packet retention and timeouts which would simply not be met, and increasing those solved the packet drops for me: https://github.com/AravisProject/aravis/issues/546#issuecomment-888968324

The related PR for Windows thread priority is here https://github.com/AravisProject/aravis/pull/548 but as said, I never really needed that (and did not test with loaded CPU either). Improvements in that area welcome, I would say.

garfeng, 2021-10-22:
@EmmanuelP, thank you!
[arv-camera-test.log](https://github.com/AravisProject/aravis/files/7393643/arv-camera-test.log)

Here is the whole log file.

EmmanuelP, 2021-11-17:
Hi @garfeng 

Can we close this issue ?

EmmanuelP, 2021-12-16:
Lets close this issue. Please reopen if needed.

Thanks.

closed 2021-12-16 by EmmanuelP

eudoxos, 2021-12-17:
Once you make the next release including these changes, I will update the package in msys/mingw and will see the result. (no need to reopen, as far as I am concerned)

eudoxos, 2021-12-17:
PS @EmmanuelP please could you add the Windows label to the issue so that it can be found later? Thanks.

EmmanuelP, 2021-12-17:
> Once you make the next release including these changes, I will update the package in msys/mingw and will see the result. (no need to reopen, as far as I am concerned)

Aravis 0.8.20 was released yesterday, including these changes.