AravisProject/aravis #599: Slow rate of Aravis Fake Camera module

Jayawardenepura, 2021-11-07:
I`m working on Raspberry Pi setup which runs **arv-fake-gv-camera-0.8** instance and sends Test Pattern:

`arv-fake-gv-camera-0.8 -i eth1
`

But I faced the issue that the receiving part has the rate which is not more that 12 MiB/s. But I hope to see at least 100MiB/S. If I reduce the MTU (on both interfaces)  and packet size on sending part, the rate will slower (as expected). Also, if tune the FPS parameter, the rate is changed. So, I use the possible maximum - 1000 HZ.

I've checked the traffic (GVSP packets) via **wireshark** and I`m sure that RPi sends an honest size of the MTU.

```
$ ./src/arv-camera-test-0.8
Looking for the first available camera
vendor name            = Aravis
model name             = Fake
device serial number   = GV01
image width            = 640
image height           = 480
horizontal binning     = 1
vertical binning       = 1
exposure               = 10000 µs
gain                   = 0 dB
payload                = 307200 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
gv packet size         = 9000 bytes
 29 frames/s -    8.91 MiB/s
 34 frames/s -    10.4 MiB/s
 33 frames/s -    10.1 MiB/s
 34 frames/s -    10.4 MiB/s
 33 frames/s -    10.1 MiB/s
 33 frames/s -    10.1 MiB/s
 34 frames/s -    10.4 MiB/s
 33 frames/s -    10.1 MiB/s
 33 frames/s -    10.1 MiB/s
 34 frames/s -    10.4 MiB/s
 34 frames/s -    10.4 MiB/s
 34 frames/s -    10.4 MiB/s
 34 frames/s -    10.4 MiB/s
 35 frames/s -    10.8 MiB/s
^C 34 frames/s -    10.4 MiB/s
n_completed_buffers    = 501
n_failures             = 0
n_underruns            = 0
n_timeouts             = 0
n_aborteds             = 0
n_missing_frames       = 0
n_size_mismatch_errors = 0
n_received_packets     = 18537
n_missing_packets      = 0
n_error_packets        = 0
n_ignored_packets      = 0
n_resend_requests      = 0
n_resent_packets       = 0
n_resend_ratio_reached = 0
n_resend_disabled      = 0
n_duplicated_packets   = 0
n_transferred_bytes    = 154075536
n_ignored_bytes        = 0
```

**Camera description:**
 - Model: Aravis Fake Camera
 - Interface: GigE Vision (Ethernet)

**Platform description:**
 - Aravis version: 0.8.6
 - OS: GigE Vision Source - RPi, GigE Vision sink - Ubuntu 20 LTS
 - Hardware: Raspberry Pi 4B, Standard x86 PC with Gigabit Ethernet NIC
 - Cable: Cat 6e cable
 - Connection: Point to point (directly)

EmmanuelP, 2021-11-17:
Hi,

Did you check the CPU use on the raspberry ? No effort has been made to make the fake camera efficient yet, as it is mainly a testing tool for the aravis library.

Jayawardenepura, 2021-11-27:
Yes, I have checked that CPU load is less than 80%.
Maybe you can get some advice how I could improve the performance?

EmmanuelP, 2021-12-16:
You can dive into src/arvfakecamera.c and arvgvfakecamera.c. The code is quite simple, I'm not sure what is to be optimized.

You may try to create a static test pattern. That would use less CPU.

EmmanuelP, 2023-06-07:
I have just checked again. The performance is mainly limited by the CPU, and also the network interface on a raspberry pi 2 (about 4MB/s). So a static pattern would definitely improve things.