AravisProject/aravis #608: How to access ARV_GVBS_MANUFACTURER_INFORMATIONS

TIS-Edgar, 2021-12-08:
Hi,

I have to differentiate between camera models, which requires an identification of the internal type/model.

Aravis already has access to this information as they are sent in the discovery ack package.

The manufacturer information however are not explicitly included in the ArvGvInterfaceDeviceInfos (manual extraction ArvGvInterfaceDeviceInfos->discovery_data would still be possible).

Is it possible to reach these information or are they considered aravis internal?
If it is possible, how?

EmmanuelP, 2021-12-14:
Please test #611

TIS-Edgar, 2021-12-15:
Hi Emmanual,

Thanks for the work.

`arv_get_device_manufacturer_info(xyz)` returns NULL for me.

I tested by adding the line

    printf("%s -> %s\n", arv_get_device_model (i), arv_get_device_manufacturer_info(i));

to `update_device_list_cb` in arvviewer.c:1183 as a test.

The backend end has the correct string. Adding 

        printf("manufacturer info: %s\n", infos->manufacturer_info);

to arvgvinterface.c:253 shows that the string is as needed `@Type=4@Housing=33G@Model=IMX250A4_C@`.

If I can provide any additional information to help you with this please let me know.

EmmanuelP, 2021-12-15:
Oh right. It was poorly tested.

I have force pushed a fix.

TIS-Edgar, 2021-12-15:
No worries.

Tested and it works like a charm!

commit 91d8ee6e (2021-12-14): arv: add arv_get_device_manufacturer_info API

Fix #608
commit 6f0c9262 (2021-12-15): arv: add arv_get_device_manufacturer_info API

Fix #608
commit fa2598d3 (2021-12-15): arv: add arv_get_device_manufacturer_info API

Fix #608
PR #611 (2021-12-15): arv: add arv_get_device_manufacturer_info API
Fix #608
closed 2021-12-15 by EmmanuelP