AravisProject/aravis #609: Invalid Genicam data

miskinati, 2021-12-11:
**Describe**
Hi, i see camera in list but not playable on arv-viewer. All follow all instructions. all dependencies are installed in the readme, set mtu to 9k and  i do setcap command to executable in /usrlocal/bin. Also i set packet-support enabled in meson configure.  
i execute arv-viewer-0.8 with debug, i can't see any error but i start arv-camera-test-0.8 i see that error:   

> No camera found: Invalid Genicam data

i see camera data with tcpdumo or in wireshark. and play fake video.  
Guess  i'm new in the GigE, i do missing something, but i need reach somewhat. 

**Camera description:**
 - Manufacturer : Global Sensor Technology,Inc
 - Model : Gavin615A
 - Interface : net socket

**Platform description:**
 - Aravis version 0.8.x
 - OS: ubuntu 18.04 LTS
 - Hardware  x86_64

**Additional context**
i search GLib-CRITICAL error, everyone says just ignore it.

> ./arv-camera-test-0.8 --debug=all
Looking for the first available camera
[17:11:47.213] 🅸 interface> Found 0 USB3Vision device (among 9 USB devices)
[17:11:47.213] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[17:11:47.213] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.0.10 (192.168.0.255)
[17:11:47.213] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.1.162 (192.168.1.255)
[17:11:47.214] 🅸 interface> [GvInterface::discovery] Device 'Global Sensor Technology,Inc  -Gavin615A-06150001' found (interface 192.168.0.10) user_id 'Gavin615A' - MAC 'f8:1d:78:71:40:89'
[17:11:47.214] 🅸 device> [GvDevice::new] Interface address = 192.168.0.10
[17:11:47.214] 🅸 device> [GvDevice::new] Device address = 192.168.0.2
[17:11:47.214] 🅸 device> [GvDevice::load_genicam] xml url = 'Local: guide_gige_test.zip; 00010000; 06c5' at 0x200
(arv-camera-test-0.8:17617): GLib-CRITICAL **: 17:11:47.214: g_ascii_strcasecmp: assertion 's1 != NULL' failed
(arv-camera-test-0.8:17617): GLib-CRITICAL **: 17:11:47.214: g_file_get_contents: assertion 'filename != NULL' failed
[17:11:47.214] 🅸 device> [GvDevice::load_genicam] xml url = 'Local: guide_gige_test.zip; 00010000; 06c5' at 0x400
(arv-camera-test-0.8:17617): GLib-CRITICAL **: 17:11:47.214: g_ascii_strcasecmp: assertion 's1 != NULL' failed
(arv-camera-test-0.8:17617): GLib-CRITICAL **: 17:11:47.214: g_file_get_contents: assertion 'filename != NULL' failed
No camera found: Invalid Genicam data

> ./arv-viewer-0.8 --debug=all 
[17:15:06.018] 🅸 interface> Found 0 USB3Vision device (among 9 USB devices)
[17:15:06.018] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[17:15:06.018] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.0.10 (192.168.0.255)
[17:15:06.018] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.1.162 (192.168.1.255)
[17:15:06.018] 🅸 interface> [GvInterface::discovery] Device 'Global Sensor Technology,Inc  -Gavin615A-06150001' found (interface 192.168.0.10) user_id 'Gavin615A' - MAC 'f8:1d:78:71:40:89'
[17:15:09.122] 🅸 interface> Found 0 USB3Vision device (among 9 USB devices)
[17:15:09.122] 🅸 device> [GvDevice::new] Interface address = 192.168.0.10
[17:15:09.122] 🅸 device> [GvDevice::new] Device address = 192.168.0.2
[17:15:09.122] 🅸 device> [GvDevice::load_genicam] xml url = 'Local: guide_gige_test.zip; 00010000; 06c5' at 0x200
(arv-viewer-0.8:17778): GLib-CRITICAL **: 17:15:09.122: g_ascii_strcasecmp: assertion 's1 != NULL' failed
(arv-viewer-0.8:17778): GLib-CRITICAL **: 17:15:09.122: g_file_get_contents: assertion 'filename != NULL' failed
[17:15:09.122] 🅸 device> [GvDevice::load_genicam] xml url = 'Local: guide_gige_test.zip; 00010000; 06c5' at 0x400
(arv-viewer-0.8:17778): GLib-CRITICAL **: 17:15:09.122: g_ascii_strcasecmp: assertion 's1 != NULL' failed
(arv-viewer-0.8:17778): GLib-CRITICAL **: 17:15:09.122: g_file_get_contents: assertion 'filename != NULL' failed

[image]

EmmanuelP, 2021-12-12:
Hi,

Thanks for the report.

The issue is due to the leading spaces in the genicam data url. I don't know if it is allowed by GigEVision, but anyway this pull request should fix the issue:

https://github.com/AravisProject/aravis/pull/610

Could you please test it.

miskinati, 2021-12-13:
Hi, sorry for late reply, i try and it work! Thanks a lot..
Its works only without setcap permission, i set 
`sudo setcap cap_net_raw+ep ./arv-viewer-0.8 `  
it not work. i run ./arv-device-test without setcap

> Looking for the first available camera
vendor        = Global Sensor Technology,Inc  
model         = Gavin615A
device id     = 06150001
** (arv-device-test:7421): CRITICAL **: 15:06:11.931: arv_gc_integer_get_value: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed
sensor width  = 0
** (arv-device-test:7421): CRITICAL **: 15:06:11.932: arv_gc_integer_get_value: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed
sensor height = 0
image width   = 640 (max:640)
image height  = 512 (max:512)
payload size  = 655360 (0xa0000)
stream port = 59631 (59631)
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
Frame rate = 50 Hz
^Cstream port = 59631 (59631)
Processed buffers = 625
Failures          = 0
Underruns         = 0

if give setcap to  ./arv-device-test, 

> Looking for the first available camera
vendor        = Global Sensor Technology,Inc  
model         = Gavin615A
device id     = 06150001
** (arv-device-test:7511): CRITICAL **: 15:06:48.643: arv_gc_integer_get_value: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed
sensor width  = 0
** (arv-device-test:7511): CRITICAL **: 15:06:48.644: arv_gc_integer_get_value: assertion 'ARV_IS_GC_INTEGER (gc_integer)' failed
sensor height = 0
image width   = 640 (max:640)
image height  = 512 (max:512)
payload size  = 655360 (0xa0000)
stream port = 45801 (45801)
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
Frame rate = 0 Hz
^Cstream port = 45801 (45801)
Processed buffers = 0
Failures          = 0
Underruns         = 0

EmmanuelP, 2021-12-13:
When cap_net_raw+ep is enabled, aravis uses a packet socket, which seems to not work with your camera. Please open a new report for this issue.

closed 2021-12-13 by EmmanuelP