AravisProject/aravis #612: Failed to build Aravis for Arm64 by crosscompile

hankfumail, 2021-12-21:
**Describe the bug**
When I built Aravis for Arm64 on Unbuntu 18.04 of X86 , it says there is no file "X11/Xlib.h", "zlib.h",  "unicode/ucnv.h". 

**To Reproduce**
Steps to reproduce the behavior:
Environments: Unbuntu 18.04, Xilinx PetaLinux 2021.2, Xilinx ZynqMP Common image 2021.2, Aravis 0.8.20

I created an crosscompile configuration file. 

```
[binaries]
# /opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-gcc
c = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-gcc -sysroot=/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots'
cpp = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-g++ -sysroot=/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots'
ar = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-ar'
strip = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-strip'
pkgconfig = '/usr/bin/pkg-config'
cmake = '/opt/Xilinx/Vitis/2021.2/tps/lnx64/cmake-3.3.2/bin/cmake'
ld = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-ld  -sysroot=/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots'
#pcap-config = ''

[properties]
skip_sanity_check = true
sys_root = '/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots'
root = '/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux'

[host_machine]
system = 'linux'
cpu_family = 'aarch64'
cpu = 'aarch64'
endian = 'little'

[target_machine]
system = 'linux'
cpu_family = 'aarch64'
cpu = 'armv8a'
endian = 'little'
```

Then I execute the following commands to build Aravis. 

```
 meson arm64-build --reconfigure  --cross-file meson-cross-arm64.txt
 ninja -C arm64-build
```

I got the following error.

```
/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-gcc -Iviewer/arv-viewer-0.8.p -Iviewer -I../viewer -Isrc -I../src -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/libxml2 -I/usr/include/libusb-1.0 -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/gio-unix-2.0/ -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/atk-1.0 -I/usr/include/pixman-1 -I/usr/include/freetype2 -I/usr/include/libpng16 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/gstreamer-1.0 -I/usr/include/orc-0.4 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -Wduplicated-branches -Wimplicit-fallthrough -Wmisleading-indentation -Wstrict-prototypes -Wunused -Wunused-variable -Wdeclaration-after-statement -Wformat=2 -Wimplicit-function-declaration -Winit-self -Wmissing-include-dirs -Wmissing-prototypes -Wpointer-arith -Wformat-nonliteral -Wenum-conversion -Wmaybe-uninitialized -fvisibility=hidden -pthread '-DARAVIS_LOCALE_DIR="/usr/local/share/locale"' '-DARAVIS_GETTEXT="aravis-0.8"' -MD -MQ viewer/arv-viewer-0.8.p/arvviewer.c.o -MF viewer/arv-viewer-0.8.p/arvviewer.c.o.d -o viewer/arv-viewer-0.8.p/arvviewer.c.o -c ../viewer/arvviewer.c
cc1.real: warning: include location "/usr/include/glib-2.0" is unsafe for cross-compilation [-Wpoison-system-directori
In file included from ../viewer/arvviewer.c:34:
/usr/include/gtk-3.0/gdk/gdkx.h:30:10: fatal error: X11/Xlib.h: No such file or directory
   30 | #include <X11/Xlib.h>
      |          ^~~~~~~~~~~~
```

I aslo export variable "PKG_CONFIG_SYSROOT_DIR" by command "export  PKG_CONFIG_SYSROOT_DIR=/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots". Then I still got same error.

I can find those files in the sysroots directory.

```
> /opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux
> hankf@XSZGS4:/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux$ find -name "Xlib.h"
> ./usr/src/debug/libx11/1_1.6.12-r0/libX11-1.6.12/include/X11/Xlib.h
> hankf@XSZGS4:/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux$ find -name "libxml2"
> ./usr/src/debug/libxml2
> ./usr/lib/cmake/libxml2
> ./usr/lib/libxml2
> ./usr/share/licenses/libxml2
> ./usr/include/libxml2
> hankf@XSZGS4:/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux$ find -name "ucnv.h"
> ./usr/src/debug/icu/67.1-r0/icu/source/common/unicode/ucnv.h
> ./usr/include/unicode/ucnv.h
```

**Expected behavior**
I hope to compile Aravis for Arm64 of Xilinx MPSoC ZCU106 board. 

**Platform description:**
 - Aravis version: 0.8.20

build machine : Ubuntu 18.04 on X86 
host machine: Arm64 of Xilinx MPSoC ZCU106 board
target machine: Arm64 of Xilinx MPSoC ZCU106 board

EmmanuelP, 2021-12-21:
Hi,

```
pkgconfig = '/usr/bin/pkg-config'
```

Isn't there a pkg-config in `/opt/Xilinx` ? If so, I guess you should use this one.

EmmanuelP, 2021-12-21:
It looks like there is no need to use a cross compilation specific pkg-config.

Looking at meson documentation, you should use at least meson 0.54, and use `pkg_config_libdir` and `sysroot` properties to specify the cross root environment directories.

https://mesonbuild.com/Cross-compilation.html

hankfumail, 2021-12-22:
It seems that we need "pkg-config" because I always has the following error. It is weird that I did get this error yesterday.

```
Compiler for C supports arguments -fvisibility=hidden: YES 
Found Pkg-config: NO
Found CMake: /opt/Xilinx/Vitis/2021.2/tps/lnx64/cmake-3.3.2/bin/cmake (3.3.2)
WARNING: The version of CMake /opt/Xilinx/Vitis/2021.2/tps/lnx64/cmake-3.3.2/bin/cmake is 3.3.2 but version >=3.4 is required
Run-time dependency glib-2.0 found: NO 

meson.build:59:0: ERROR: Pkg-config binary for machine MachineChoice.HOST not found. Giving up.

A full log can be found at /proj/hankf/hankf/zte/gige_vision/aravis/arm64-build/meson-logs/meson-log.txt
```

Here is my cross-compile file.

```
# meson arm64-build --cross-file meson-cross-arm64.txt
# meson arm64-build --cross-file meson-cross-arm64.txt   --reconfigure
# ninja -C arm64-build

[binaries]
# /opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-gcc
c = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-gcc'
cpp = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-g++'
ar = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-ar'
strip = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-strip'
#pkgconfig = '/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux/usr/lib/pkgconfig'
cmake = '/opt/Xilinx/Vitis/2021.2/tps/lnx64/cmake-3.3.2/bin/cmake'
ld = '/opt/Xilinx/Vitis/2021.2/gnu/aarch64/lin/aarch64-linux/bin/aarch64-linux-gnu-ld'
#pcap-config = ''

[properties]
skip_sanity_check = true
sys_root = '/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux'
root = '/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux'

#pkg_config_libdir: casue problem glib.h/glibconfig.h: No such file or directory
#pkg_config_libdir = '/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux'
pkg_config_libdir = '/opt/Xilinx/peta/2021.2/sdk_zynqmp/sysroots/cortexa72-cortexa53-xilinx-linux/usr/lib/pkgconfig'

[host_machine]
system = 'linux'
cpu_family = 'aarch64'
cpu = 'aarch64'
endian = 'little'

[target_machine]
system = 'linux'
cpu_family = 'aarch64'
cpu = 'armv8a'
endian = 'little'
```

EmmanuelP, 2021-12-22:
```
system = 'linux'
cpu_family = 'aarch64'
cpu = 'aarch64'
endian = 'little'
```

You have said you are compiling on an x86 machine. Is that the right declaration for the host machine ?
Tha would explain why meson did not find pkg-config for HOST.

mshr-h, 2022-02-04:
Hi, we've successfully cross-compiled Aravis v0.8.19 for the zcu106 target.
Here's how we did it.

1. Install PetaLinux tools to `/opt/petalinux`
2. Create PetaLinux project to `/home/ubuntu/petalinux-wok/xilinx-zcu106-2021.1`
3. Source PetaLinux setting script in the terminal
    - `source /opt/petalinux/setting.sh`
4. Add dependent packages to the rootfs by `petalinux-config -c rootfs` command
```text
Filesystem Packages > libs > gobject-introspection > [*] gobject-introspection
Filesystem Packages > libs > gobject-introspection > [*] gobject-introspection-dev
Filesystem Packages > libs > libnotify > [*] libnotify
Filesystem Packages > libs > libnotify > [*] libnotify-dev
Filesystem Packages > misc > glib-2.0 > [*] glib-2.0
Filesystem Packages > misc > glib-2.0 > [*] glib-2.0-dev
Filesystem Packages > misc > glib-2.0 > [*] glib-2.0-utils
Filesystem Packages > misc > gstreamer1.0-meta-base > [*] gstreamer1.0-meta-base
Filesystem Packages > misc > gstreamer1.0-meta-base > [*] gstreamer1.0-meta-video
Filesystem Packages > misc > gstreamer1.0-meta-base > [*] gstreamer1.0-meta-video-dev
Filesystem Packages > misc > gstreamer1.0-meta-base > [*] gstreamer1.0-meta-base-dev
Filesystem Packages > multimedia > gstreamer1.0 > [*] gstreamer1.0
Filesystem Packages > multimedia > gstreamer1.0 > [*] gstreamer1.0-dev
Petalinux Package Groups > packagegroup-petalinux-gstreamer > [*] packagegroup-petalinux-gstreamer
```
5. Build and install petalinux SDK 
    - `petalinux-build --sdk && petalinux-package --sysroot`
    - cross compilers will be installed to `/home/ubuntu/petalinux-wok/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux`
    - sysroot will be installed to `/home/ubuntu/petalinux-wok/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/cortexa72-cortexa53-xilinx-linux/`
6. Source the environment setup script in the terminal
    - `. /home/ubuntu/petalinux-wok/xilinx-zcu106-2021.1/images/linux/sdk/environment-setup-cortexa72-cortexa53-xilinx-linux`
7. Install meson and ninja
    - `python3 -m pip install meson==0.59.0 ninja`
8. Create meson cross files to `/home/ubuntu/.local/share/meson/cross/cortexa72-cortexa53-xilinx-linux`
```
[binaries]
c         = '/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-gcc'
ar        = '/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-ar'
strip     = '/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-strip'
cpp       = '/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/x86_64-petalinux-linux/usr/bin/aarch64-xilinx-linux/aarch64-xilinx-linux-g++'
#cmake     = '/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/x86_64-petalinux-linux/usr/bin/cmake'
pkgconfig = '/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/x86_64-petalinux-linux/usr/bin/pkg-config'
#exe_wrapper = 'qemu' # A command used to run generated executables.
[properties]
sys_root      =            '/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/cortexa72-cortexa53-xilinx-linux/'
[built-in options]
c_args        = ['--sysroot=/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/cortexa72-cortexa53-xilinx-linux/']
cpp_args      = ['--sysroot=/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/cortexa72-cortexa53-xilinx-linux/']
c_link_args   = ['--sysroot=/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/cortexa72-cortexa53-xilinx-linux/']
cpp_link_args = ['--sysroot=/home/ubuntu/petalinux-work/xilinx-zcu106-2021.1/images/linux/sdk/sysroots/cortexa72-cortexa53-xilinx-linux/']
[host_machine]
system = 'linux'
cpu_family = 'aarch64'
cpu = 'armv8'
endian = 'little'
```
9. Clone and build aravis
    - `git clone https://github.com/AravisProject/aravis`
    - `cd aravis && git checkout 0.8.19`
    - `~/.local/bin/meson build --cross-file cortexa72-cortexa53-xilinx-linux`
    - `cd build && ninja`