AravisProject/aravis #613: SVC fxo545CXGE 10 GigE camera

roman-raisin, 2021-12-23:
**SVC fxo545CXGE 10 GigE camera is not working with Aravis.**

I'm trying to use Aravis (first time) with SVC fxo545CXGE 10 Gbit camera:
https://www.svs-vistek.com/en/industrial-cameras/svs-camera-detail.php?id=fxo545CXGE

Started with `arv-viewer`: camera is visible (format: BayerRG8), but it shows black screen (_0 fps, 0 image / 0errors_).

Then I tried to play with Features using `arv-tool` - successful.

Then I ran `arv-test` with the following result:
```
Testing 'SVS-VISTEK GmbH:fxo545CXGE'
Genicam:Load                        SUCCESS
Genicam:Schema                      IGNORED
Properties:SensorSizeReadout        SUCCESS
Properties:SensorSizeCheck          IGNORED
Properties:GainReadout              SUCCESS
Properties:ExposureTimeReadout      SUCCESS
MultipleAcquisitionA:BufferCheck    FAILURE 0/10
MultipleAcquisitionA:FrameRate      FAILURE
SoftwareTrigger:BufferCheck         FAILURE 0/5
MultipleAcquisitionB:BufferCheck    FAILURE 0/10
MultipleAcquisitionB:FrameRate      FAILURE
SingleAcquisition:BufferCheck       FAILURE
Chunks:NoSupport                    FAILURE
GigEVision:NStreamChannels          SUCCESS
GigEVision:StreamChannel            SUCCESS
```
**Camera description:**
 - Manufacturer: SVC
 - Model: fxo545CXGE
 - Interface [10GigE]

**Platform description:**
 - Aravis version 0.8.20 (cloned master, commit 0c9a3de0)
 - OS: [Ubuntu 20.04 LTS]
 - Hardware [Intel 10 Gbit Ethernet Controller X540]

**Additional context**
My camera is working well with the proprietary SVCamKit SDK from SVC.
My build of Aravis is operational, I tested it with Basler USB camera (`arc-viewer works, I have a live picture).

I tried to investigate the case and found that `arv_stream_timeout_pop_buffer()` is always returning `NULL`.

It causes _MultipleAcquisitionA:BufferCheck_, _SoftwareTrigger:BufferCheck_, _MultipleAcquisitionB:BufferCheck_ and _SingleAcquisition:BufferCheck_ to fail in `arv-test`.

I tried to go deeper, but I couldn't find `arv_stream_get_instance_private()` function neither declaration nor definition.

I can attach the result of `arv-test --debug=all:4` if it could help. I could also attach the tree of camera features in some form.

I'll be glad to contribute in supporting SVC 10GigE cameras if I could.

EmmanuelP, 2021-12-24:
Hi,

Thanks for your interest in Aravis.

> I can attach the result of arv-test --debug=all:4 if it could help. I could also attach the tree of camera features in some form.

Please attach the output of `arv-test-0.8 -d all -t Single*` and the feature tree.

roman-raisin, 2021-12-24:
> Please attach the output of `arv-test-0.8 -d all -t Single*` and the feature tree.

Hi Emmanuel,

Thank you for reaching me back.

<details>
  <summary>arv-test result</summary>

  Actually I have two cameras on two 10 GBit interfaces, so I have limited my testing to one of them.
  
  ```bash
  $ arv-test-0.8 -a 192.168.100.100 -d all -t Single*
  
[19:51:49.545] 🅸 interface> Found 0 USB3Vision device (among 11 USB devices)
[19:51:49.546] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[19:51:49.546] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.16.60.3 (172.16.255.255)
[19:51:49.546] 🅸 interface> [GvDiscoverSocket::new] Add interface 10.42.0.1 (10.42.0.255)
[19:51:49.546] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.100.1 (192.168.100.255)
[19:51:49.546] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.101.1 (192.168.101.255)
[19:51:49.546] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.16.32.1 (172.16.32.255)
[19:51:49.546] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.16.136.1 (172.16.136.255)
[19:51:49.546] 🅸 interface> [GvInterface::discovery] Device 'SVS-VISTEK GmbH-fxo545CXGE-107444' found (interface 192.168.101.1) user_id 'Left-101' - MAC 'ac:4f:fc:01:6f:e8'
[19:51:49.546] 🅸 interface> [GvInterface::discovery] Device 'SVS-VISTEK GmbH-fxo545CXGE-107445' found (interface 192.168.100.1) user_id 'Right-100' - MAC 'ac:4f:fc:01:6f:e9'
Found 2 devices
[19:51:50.548] 🅸 interface> Found 0 USB3Vision device (among 11 USB devices)
[19:51:50.548] 🅸 device> [GvDevice::new] Interface address = 192.168.101.1
[19:51:50.548] 🅸 device> [GvDevice::new] Device address = 192.168.101.101
[19:51:50.549] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:fxo545CXGE_v1.7.2_b2990.xml;10000;4C114' at 0x200
[19:51:50.549] 🅸 device> [GvDevice::load_genicam] Xml address = 0x10000 - size = 0x4c114 - fxo545CXGE_v1.7.2_b2990.xml
[19:51:50.757] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSDoNotFragment'
[19:51:50.757] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[19:51:50.757] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[19:51:50.758] 🅸 device> [GvDevice::new] Device endianness = big
[19:51:50.758] 🅸 device> [GvDevice::new] Packet resend     = yes
[19:51:50.758] 🅸 device> [GvDevice::new] Write memory      = yes
[19:51:50.758] 🅸 device> [GvDevice::new] Legacy endianness handling = no
Testing 'SVS-VISTEK GmbH:fxo545CXGE'
[19:51:50.759] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[19:51:50.759] 🅸 device> [GvDevice::auto_packet_size] No GevSCPSFireTestPacket feature found
[19:51:50.760] 🅸 device> [GvStream::stream_new] Packet size = 576 byte(s)
[19:51:50.761] 🅸 stream> [GvStream::stream_new] Destination stream port = 59694
[19:51:50.761] 🅸 stream> [GvStream::stream_new] Source stream port = 0
[19:51:50.761] 🅸 stream> [GvStream::loop] Standard socket method
[19:51:51.763] 🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;           0;           0
                         -------------
                         >=      2000;           0;           0;           0
                         <          0;           0;           0;           0
                         min         :         n/a;         n/a;         n/a
                         max         :         n/a;         n/a;         n/a
                         last max at :           0;           0;           0
                         counter     :           0:           0:           0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_failures             = 0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_timeouts             = 0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_aborteds             = 0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_missing_frames       = 0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_received_packets     = 0
[19:51:51.763] 🅸 stream> [GvStream::finalize] n_missing_packets      = 0
[19:51:51.764] 🅸 stream> [GvStream::finalize] n_error_packets        = 0
[19:51:51.764] 🅸 stream> [GvStream::finalize] n_ignored_packets      = 0
[19:51:51.764] 🅸 stream> [GvStream::finalize] n_resend_requests      = 0
[19:51:51.764] 🅸 stream> [GvStream::finalize] n_resent_packets       = 0
[19:51:51.764] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
[19:51:51.764] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 0
[19:51:51.764] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
[19:51:51.764] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 0
[... 579 lines cut ...]
            EnumEntry   : 'UserSet8'
            EnumEntry   : 'UserSet7'
            EnumEntry   : 'UserSet6'
            EnumEntry   : 'UserSet5'
            EnumEntry   : 'UserSet4'
            EnumEntry   : 'UserSet3'
            EnumEntry   : 'UserSet2'
            EnumEntry   : 'UserSet1'
            EnumEntry   : 'Default'
    Category    : 'Debug'
        Command     : 'RegisterRefresh'
        Integer     : 'RegisterAddress'
        Integer     : 'RegisterValue_dec'
        Integer     : 'RegisterValue_hex'
        Command     : 'StreamingStatsRefresh'
        Integer     : 'StreamingStatsImageAll'
        Integer     : 'StreamingStatsImageDrop'
        Integer     : 'StreamingStatsImageSent'
        Integer     : 'StreamingStatsResendPacketSum'
        Integer     : 'StreamingStatsResendPacketUnavail'
        Integer     : 'StreamingStatsResendPacketNotYetAvail'
        Integer     : 'StreamingStatsResendPacketOverflow'
        Integer     : 'StreamingStatsResendPacketRequests'
```
</details

roman-raisin, 2021-12-26:
I disable one of my two 10 Gbit interfaces, and ran the test again (with the same result).
I also realised how to get a GenICam XML-like tree with `arc-tool genicam`, attaching the result
[fxo545CXGE.genicam.txt](https://github.com/AravisProject/aravis/files/7777125/fxo545CXGE.genicam.txt)
.

roman-raisin, 2021-12-26:
I ran `tcpdump` while running `arv-test` and noticed:

1. I see no correspondence between the _Destination stream port_ mentioned in `arv-test` and the _UDP port_ in `tcpdump`. As I understand, that means there is no stream from camera;
2. _Source stream port = 0_ displayed by `arv-test` seems to be wrong.

<details>
  <summary>arv-test</summary>

```bash
$ arv-test-0.8 -d all -t Single*
# ...ripped...
🅸 interface> [GvInterface::discovery] Device 'SVS-VISTEK GmbH-fxo545CXGE-107445' found (interface 192.168.100.1) user_id 'Right-100' - MAC 'ac:4f:fc:01:6f:e9'
Found 1 device
🅸 interface> Found 0 USB3Vision device (among 11 USB devices)
🅸 device> [GvDevice::new] Interface address = 192.168.100.1
🅸 device> [GvDevice::new] Device address = 192.168.100.100
🅸 device> [GvDevice::load_genicam] xml url = 'Local:fxo545CXGE_v1.7.2_b2990.xml;10000;4C114' at 0x200
🅸 device> [GvDevice::load_genicam] Xml address = 0x10000 - size = 0x4c114 - fxo545CXGE_v1.7.2_b2990.xml
🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSDoNotFragment'
🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
🅸 device> [GvDevice::new] Device endianness = big
🅸 device> [GvDevice::new] Packet resend     = yes
🅸 device> [GvDevice::new] Write memory      = yes
🅸 device> [GvDevice::new] Legacy endianness handling = no
Testing 'SVS-VISTEK GmbH:fxo545CXGE'
🅸 device> [GvDevice::create_stream] Number of stream channels = 1
🅸 device> [GvDevice::auto_packet_size] No GevSCPSFireTestPacket feature found
🅸 device> [GvStream::stream_new] Packet size = 576 byte(s)
🅸 stream> [GvStream::stream_new] Destination stream port = 44685
🅸 stream> [GvStream::stream_new] Source stream port = 0
🅸 stream> [GvStream::loop] Standard socket method
🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                     0;           0;           0;           0
          -------------
          >=      2000;           0;           0;           0
          <          0;           0;           0;           0
          min         :         n/a;         n/a;         n/a
          max         :         n/a;         n/a;         n/a
          last max at :           0;           0;           0
          counter     :           0:           0:           0
🅸 stream> [GvStream::finalize] n_completed_buffers    = 0
🅸 stream> [GvStream::finalize] n_failures             = 0
🅸 stream> [GvStream::finalize] n_underruns            = 0
🅸 stream> [GvStream::finalize] n_timeouts             = 0
🅸 stream> [GvStream::finalize] n_aborteds             = 0
🅸 stream> [GvStream::finalize] n_missing_frames       = 0
🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
🅸 stream> [GvStream::finalize] n_received_packets     = 0
🅸 stream> [GvStream::finalize] n_missing_packets      = 0
🅸 stream> [GvStream::finalize] n_error_packets        = 0
🅸 stream> [GvStream::finalize] n_ignored_packets      = 0
🅸 stream> [GvStream::finalize] n_resend_requests      = 0
🅸 stream> [GvStream::finalize] n_resent_packets       = 0
🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
🅸 stream> [GvStream::finalize] n_resend_disabled      = 0
🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
🅸 stream> [GvStream::finalize] n_transferred_bytes    = 0
🅸 stream> [GvStream::finalize] n_ignored_bytes        = 0
🅸 stream> [Stream::finalize] Flush 1 buffer[s] in input queue
🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue
SingleAcquisition:BufferCheck       FAILURE
```
</details>

<details>
  <summary>tcpdump</summary>

```bash
$ sudo tcpdump -i enp3s0f0
[... 1248 lines cut ...]
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 16
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 12
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 12
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 16
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 16
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 16
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 12
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 12
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 12
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 12
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 12
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 12
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 12
IP 192.168.100.1.38591 > 192.168.100.100.3956: UDP, length 16
IP 192.168.100.100.3956 > 192.168.100.1.38591: UDP, length 12
```
</details>

durnezj, 2022-02-02:
Hi, any update on this issue, did you resolve it and if so, how?
I am running into the exact same issue with a NET Gmbh camera.

EmmanuelP, 2022-02-02:
What could help is a wireshark capture of a successful session using the proprietary SDK. Make sure to select a small image size and a slow framerate in order to minimize the capture file size.

durnezj, 2022-02-03:
I have no possibility of running a proprietary SDK for the moment. The cameras are already integrated into a linux system and I only have access to them through Linux with an Aravis 0.8.20 install.
The only observations I can offer are:
- My camera is a NET GmbH GP4136C.
- running arv-tool on an earlier version like 0.8.0 gives the error: `Failed to open device 'NET GmbH-GP4136C-15212037003': Invalid Genicam data`, but is detected on 0.8.20 .

- When I (in order) set auto packet size, trigger a Software Trigger and acquire a buffer (which is always NULL, nothing in the GError about it), the follwing debug output is shown:
```
[15:05:56.322] 🅸 device> [GvDevice::new] Interface address = 10.0.0.1
[15:05:56.322] 🅸 device> [GvDevice::new] Device address = 10.0.0.2
[15:05:56.421] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:gigepro_1.12.008.00.xml.zip;FFC10000;0000AD5A?SchemaVersion=1.1.0' at 0x200
[15:05:56.421] 🅸 device> [GvDevice::load_genicam] Xml address = 0xffc10000 - size = 0xad5a - gigepro_1.12.008.00.xml.zip
[15:05:56.638] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[15:05:56.672] 🅸 device> [GvDevice::new] Device endianness = big
[15:05:56.672] 🅸 device> [GvDevice::new] Packet resend     = yes
[15:05:56.672] 🅸 device> [GvDevice::new] Write memory      = yes
[15:05:56.672] 🅸 device> [GvDevice::new] Legacy endianness handling = no
Found camera 'GP4136C'
[15:05:56.675] 🅸 device> [GvDevice::auto_packet_size] Try packet size = 8177
[15:05:56.676] 🅸 device> [GvDevice::auto_packet_size] Packet size set to 8177 bytes
[15:05:56.680] 🅸 device> [GvDevice::create_stream] Number of stream channels = 1
[15:05:56.682] 🅸 device> [GvDevice::auto_packet_size] Current packet size check successfull (8177 bytes)
[15:05:56.683] 🅸 device> [GvStream::stream_new] Packet size = 8177 byte(s)
[15:05:56.684] 🅸 stream> [GvStream::stream_new] Destination stream port = 58343
[15:05:56.684] 🅸 stream> [GvStream::stream_new] Source stream port = 0
[15:05:56.696] 🅸 stream> [GvStream::loop] Packet socket method
[15:05:57.673] 🅳 device> [GvDevice::Heartbeat] Ack value = 2
[15:06:06.732] 🅸 stream>     bins    ;frame_retent; packet_time;inter_packet
                                    0;           0;           0;           0
                         -------------
                         >=      2000;           0;           0;           0
                         <          0;           0;           0;           0
                         min         :         n/a;         n/a;         n/a
                         max         :         n/a;         n/a;         n/a
                         last max at :           0;           0;           0
                         counter     :           0:           0:           0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_failures             = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_timeouts             = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_aborteds             = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_missing_frames       = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_received_packets     = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_missing_packets      = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_error_packets        = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_ignored_packets      = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_resend_requests      = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_resent_packets       = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_resend_disabled      = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_transferred_bytes    = 0
[15:06:06.732] 🅸 stream> [GvStream::finalize] n_ignored_bytes        = 0
[15:06:06.732] 🅸 stream> [Stream::finalize] Flush 1 buffer[s] in input queue
[15:06:06.732] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue
```

The line that seems to be off is `[15:05:56.684] 🅸 stream> [GvStream::stream_new] Source stream port = 0`, as Romain-raisin also has. Do you have any idea in what direction to look to solve this problem?

durnezj, 2022-02-28:
@EmmanuelP It might not be the camera after all, I got it working on my personal PC instead of the embedded PC the camera is connected to. But then I am really confused as to where the difference could be between the PC where it works and where it doesn't. They have the same aravis version.

roman-raisin, 2022-05-10:
@EmmanuelP 

# Update: Aravis is able to work with SVS fxo545CXGE cameras on Intel Ubuntu 20.04 machine.

## What has changed:
1. I rebuilt and reinstalled Aravis to more recent version (3c8ec492);
2. I updated cameras firmware from `b2990` to recent `b3038` version;

## What's new to my experiment ?
I mentioned that Aravis doesn't work after proprietary SDK. This could be the real cause of our problem.

## Report from console

```bash
# After cold start
> cat /etc/os-release | grep VERSION_ID
VERSION_ID="20.04"
> uname -a
Linux rr-ubu20 5.13.0-40-generic #45~20.04.1-Ubuntu SMP Mon Apr 4 09:38:31 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
> sudo ip link set enp3s0f0 mtu 9400
> ip link show enp3s0f0
4: enp3s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 9400 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether a0:36:9f:20:d3:a4 brd ff:ff:ff:ff:ff:ff
 
> arv-test-0.8 -t Single*
Found 1 device
Testing 'SVS-VISTEK GmbH:fxo545CXGE'
SingleAcquisition:BufferCheck       SUCCESS 

> arv-viewer-0.8 
> # I have seen the picture, no errors

# Now running the proprietary SDK's viewer
> SVCapture64 
HeartbeatRate 3000 
# I have seen the video from camera

> arv-test-0.8 -t Single*
Found 1 device
Testing 'SVS-VISTEK GmbH:fxo545CXGE'
SingleAcquisition:BufferCheck       FAILURE 
> arv-viewer-0.8 
> # I saw 0 images and increasing error counter
 
> # Now I restart camera cold (physical power off/on)
> arv-test-0.8 -t Single*
Found 1 device
Testing 'SVS-VISTEK GmbH:fxo545CXGE'
SingleAcquisition:BufferCheck       SUCCESS 
```

## Coming soon

Build Aravis for NVidia Xavier AGX (Ubuntu 18.04, aarch64) and test with my camera.

roman-raisin, 2022-05-10:
# Test result on NVidia Xavier AGX (Ubuntu 18.04)
## The build prerequisites
To build Aravis on Jetson AGX running Ubuntu 18.04.6, I had to update `meson` to more recent version, but it depends on `python 3.7+`, while system python is 3.6. So I installed Anaconda Mini to provide that environment.
```bash
jetson:~$ meson --version
0.45.1
# Too old
jetson:~$ ninja --version
1.8.2
jetson:~$ python3 --version
Python 3.6.9
# too old for Meson

jetson:~$ ./Miniconda3-latest-Linux-aarch64.sh
# success
(base) jetson:~$ python3 --version
Python 3.9.7

(base) jetson:~$ conda install meson
# success
(base) jetson:~$ meson --version
0.59.0
(base) jetson:~$ ninja --version
1.10.2
```
After that I could make a normal build of Aravis (the same version, 3c8ec492) .

## Test results
Aravis works with fxo545CXGE camera on Jetson Xavier AGX

```bash
jetson:~$ cat /etc/os-release | grep VERSION_ID
VERSION_ID="18.04"
jetson:~$ uname -a
Linux rogue-1 4.9.253-tegra #1 SMP PREEMPT Thu Feb 24 11:50:52 EST 2022 aarch64 aarch64 aarch64 GNU/Linux

jetson:~$ sudo ip link set eth1 mtu 9000
jetson:~$ sudo ip link show eth1
4: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 00:0c:8b:b4:11:dc brd ff:ff:ff:ff:ff:ff

jetson:~$ arv-test-0.8 -t Single*
Found 1 device
Testing 'SVS-VISTEK GmbH:fxo545CXGE'
SingleAcquisition:BufferCheck       SUCCESS 

# Viewer test (in Desktop GUI)
jetson:~$ arv-viewer-0.8 
# I saw the picture
```
[image]
[image]

Later on I'll investigate the issue with proprietary SDK, to realize what changed in camera properties by comparing feature trees on cold camera start and after running that SDK.

EmmanuelP, 2022-05-11:
> This solution seems stable enough and saturates my gbit link. I will now try and understand the differences between the modes 'standard socket' and 'packet socket'

Packet socket is a more efficient method of receiving stream packets from the device. The issue is, for some devices, aravis reads 0 as the source port, which is incorrect. Packet socket has an input filter that not only checks for the right host ip and port, but also the right device ip and port. I guess we should remove the source port from the input packet filtering.

EmmanuelP, 2023-06-07:
> Packet socket is a more efficient method of receiving stream packets from the device. The issue is, for some devices, aravis reads 0 as the source port, which is incorrect. Packet socket has an [...]

That should be fixed for a while now by 6c2894e106f21037bba7458365f308fc63ad3381 .

I guess we can close this issue now.

closed 2023-06-07 by EmmanuelP