AravisProject/aravis #614: Chunk data is never found with Lucid TRI204S-C

contradict, 2021-12-29:
**Describe the bug**
Attempting to use any chunk data from a Lucid TRI204S-C always fails with "Chunk 0xXXXXXXXX not found"

**To Reproduce**
An update of the program from #289 shows the problem:

```c++
#include "arv.h"
#include <iostream>
#include <stdint.h>
#include <vector>

using namespace std;

int main()
{
    ArvInterface interface;
    arv_update_device_list();
    uint32_t n_devices = arv_get_n_devices();

    if (n_devices == 0)
    {
        cout<<"No GenICam cameras available"<<endl;
        return(-1);
    }
    ArvDevice* p_device;
    ArvCamera* p_cam = arv_camera_new(arv_get_device_id(0), NULL);

    p_device = arv_camera_get_device(p_cam);

    arv_camera_stop_acquisition(p_cam, NULL);

    cout<<"Device vendor: "<<arv_camera_get_vendor_name(p_cam, NULL)<<endl;
    cout<<"Device model: "<<arv_camera_get_model_name(p_cam, NULL)<<endl;

    //cout<<"Device family: "<<arv_device_get_string_feature_value(p_device, "DeviceFamilyName")<<endl;
    cout<<"Device version: "<<arv_device_get_string_feature_value(p_device, "DeviceVersion", NULL)<<endl;
    cout<<"Firmware version: "<<arv_device_get_string_feature_value(p_device, "DeviceFirmwareVersion", NULL)<<endl;
    //cout<<"Serial number: "<<arv_device_get_string_feature_value(p_device, "DeviceSerialNumber")<<endl;
    //cout<<"Sensor Description: "<<arv_device_get_string_feature_value(p_device, "SensorDescription")<<endl;

    cout<<"Aravis library "<<ARAVIS_MAJOR_VERSION<<"."<<ARAVIS_MINOR_VERSION<<"."<<ARAVIS_MICRO_VERSION<<endl;

    arv_device_set_string_feature_value (p_device, "TriggerSelector", "AcquisitionStart", NULL);
    arv_device_set_string_feature_value (p_device, "TriggerMode", "Off", NULL);
    arv_device_set_string_feature_value (p_device, "TriggerSelector", "FrameStart", NULL);
    arv_device_set_string_feature_value (p_device, "TriggerActivation", "RisingEdge", NULL);
    arv_device_set_string_feature_value (p_device, "TriggerSource", "Software", NULL);
    arv_device_set_string_feature_value (p_device, "TriggerMode", "On", NULL);

    uint32_t n_formats;
    int64_t* formats = arv_camera_dup_available_pixel_formats(p_cam, &n_formats, NULL);
    vector<uint32_t> desired_formats;
    desired_formats.push_back(ARV_PIXEL_FORMAT_BAYER_RG_12_PACKED);
    desired_formats.push_back(ARV_PIXEL_FORMAT_BAYER_RG_12);
    desired_formats.push_back(ARV_PIXEL_FORMAT_BAYER_RG_16);
    desired_formats.push_back(ARV_PIXEL_FORMAT_MONO_12_PACKED);
    desired_formats.push_back(ARV_PIXEL_FORMAT_MONO_12);
    desired_formats.push_back(ARV_PIXEL_FORMAT_MONO_16);
    bool format_set = false;
    for(int i = 0; i < desired_formats.size(); i++)
    {
        for(int j = 0; j < n_formats; j++)
        {
            if(formats[j] == desired_formats[i])
            {
                arv_camera_set_pixel_format(p_cam, desired_formats[i], NULL);
                format_set = true;
                break;
            }
        }
        if(format_set)
        {
            break;
        }
    }
    if(!format_set)
    {
        cout<<"Failed to set pixel format, available pixel formats are:";
        for(int j = 0; j < n_formats; j++)
        {
            cout<<"\t"<<formats[j];
        }
        cout<<endl;
    }

    cout<<"Pixel format: "<<arv_camera_get_pixel_format_as_string(p_cam, NULL)<<endl;

    //Enable embedded info
    arv_camera_set_chunks(p_cam, "ExposureTime Gain Timestamp", NULL);
    ArvChunkParser* p_chunk_parser = arv_camera_create_chunk_parser(p_cam);

    arv_device_set_string_feature_value (p_device, "BalanceWhiteAuto", "Off", NULL);
    arv_device_set_string_feature_value (p_device, "BalanceRatioSelector", "Blue", NULL);
    arv_device_set_float_feature_value (p_device, "BalanceRatio", 1.633f, NULL);
    arv_device_set_string_feature_value (p_device, "BalanceRatioSelector", "Red", NULL);
    arv_device_set_float_feature_value (p_device, "BalanceRatio", 1.716f, NULL);

    arv_device_set_float_feature_value(p_device, "BlackLevel", 0., NULL);
    arv_device_set_boolean_feature_value(p_device, "IspEnable", false, NULL);
    arv_device_set_boolean_feature_value(p_device, "GammaEnable", false, NULL);
    arv_device_set_string_feature_value(p_device, "AdcBitDepth", "Bit12", NULL);

    int32_t width, height;
    arv_camera_get_sensor_size(p_cam, &width, &height, NULL);
    cout<<"Sensor size: "<<width<<"x"<<height<<endl;

    int32_t payload = arv_camera_get_payload(p_cam, NULL);
    cout << "Payload: "<<payload<<endl;

    ArvStream* p_stream = arv_camera_create_stream(p_cam, NULL, NULL, NULL);
    arv_stream_push_buffer(p_stream, arv_buffer_new(payload, NULL));
    arv_stream_push_buffer(p_stream, arv_buffer_new(payload, NULL));
    arv_camera_start_acquisition(p_cam, NULL);

    arv_camera_set_gain_auto(p_cam, ARV_AUTO_OFF, NULL);
    arv_camera_set_exposure_time_auto(p_cam, ARV_AUTO_OFF, NULL);

    arv_device_set_float_feature_value(p_device, "AutoExposureTimeUpperLimit", 5000, NULL);
    arv_device_set_float_feature_value(p_device, "AutoExposureExposureTimeUpperLimit", 5000, NULL);

    arv_camera_software_trigger(p_cam, NULL);

    ArvBuffer* p_buf = arv_stream_pop_buffer(p_stream);
    ArvBufferStatus status = arv_buffer_get_status(p_buf);
    GError *err = NULL;
    if(status == ARV_BUFFER_STATUS_SUCCESS)
    {
        ArvPixelFormat pixel_format = arv_buffer_get_image_pixel_format(p_buf);
        cout<<"ChunkExposureTime: "<<arv_chunk_parser_get_float_value(p_chunk_parser, p_buf, "ChunkExposureTime", &err)<<endl;
        if(err) {
            cout<<"Error: "<<err->message<<endl;
            g_error_free(err);
            err = NULL;
        }
        cout<<"ChunkGain: "<<arv_chunk_parser_get_float_value(p_chunk_parser, p_buf, "ChunkGain", &err)<<endl;
        if(err) {
            cout<<"Error: "<<err->message<<endl;
            g_error_free(err);
            err = NULL;
        }
        cout<<"ChunkTimestamp: "<<arv_chunk_parser_get_integer_value(p_chunk_parser, p_buf, "ChunkTimestamp", &err)<<endl;
        if(err) {
            cout<<"Error: "<<err->message<<endl;
            g_error_free(err);
            err = NULL;
        }
     }
    arv_stream_push_buffer(p_stream, p_buf);

    arv_camera_stop_acquisition(p_cam, NULL);
    g_object_unref(p_stream);
    g_object_unref(p_cam);
    return 0;
}
```
Results in the output
```
$ g++ -I ~/src/aravis/src -I ~/src/aravis/build/src -I /usr/include/glib-2.0 -I /usr/lib/x86_64-linux-gnu/glib-2.0/include chunk.cpp -L ~/src/aravis/build/src -laravis-0.8 -lglib-2.0 -lgobject-2.0
$ LD_LIBRARY_PATH=~/src/aravis/build/src ./a.out
Device vendor: Lucid Vision Labs
Device model: TRI204S-C
Device version: RQQ51
Firmware version: 1.63.0.0
Aravis library 0.8.21
Pixel format: BayerRG12Packed
Sensor size: 4504x4504
Payload: 30429072
ChunkExposureTime: 0
Error: [ArvGcPort::read] Chunk 0x4c564c08 not found
ChunkGain: 0
Error: [ArvGcPort::read] Chunk 0x4c564c09 not found
ChunkTimestamp: 0
Error: [ArvGcPort::read] Chunk 0x4c564c0b not found
```

**Expected behavior**
I hoped to retrieve the values for the specified chunks.

**Additional context**

I used the Python interface to poke at this a little bit and noticed that the format of the buffer data seems off, there appear to be 4 extra 0 bytes at the end that throw off the chunk parser. When requesting an image with these three chunks enabled, the last 48 bytes of `buffer.get_data()` are:
```
4c564c00 01d04f60 ad412602 4c564c08 00000004 00000000 4c564c09 00000004 00000000 4c564c0b 00000004 00000000
```
Which looks like the chunk ids for Image, Exposure Time, Gain (value zero??), and Timestamp (also zero?), plus an extra 4 zero bytes.

contradict, 2021-12-29:
It turns out this is only a problem with the Timestamp chunk. Other chunks do not have this problem, but enabling the Timestamp chunk causes the extra 4 bytes to be appended, no matter what other chunks are included. This is probably not fixable in Aravis, but I'll leave it here in case anyone else stumbles across the problem.

EmmanuelP, 2023-06-07:
Thanks for the follow up.

closed 2023-06-07 by EmmanuelP