AravisProject/aravis #616: Improve performance of GigeVision stream receiving loop

lyodaom, 2022-01-04:
**Is your feature request related to a problem? Please describe.**
We have a server that drops some frames due to the receiving loop not executing fast enough.

**Describe the solution you'd like**
A big portion of the time is spent in the function call to `g_socket_receive` (line [here](https://github.com/AravisProject/aravis/blob/master/src/arvgvstream.c#L836)) which uses the libc `recv` function and this one relies on the `recvfrom` system call, considering how expensive system calls are (specially since spectre/meltdown patches), I suggest using `g_socket_receive_messages` instead, which is based on the `recvmmsg` function, this way we can receive many messages in one system call (this requires to change the rest of the loop to consider many messages instead of just one).
The only downside is having to require GIO version greater or equal to 2.48, but I suppose that's not a big issue.

lyodaom, 2022-01-04:
is there any load test available that could be used to compare the global performance after a code change in aravis?
 I think it would be a useful thing to have

EmmanuelP, 2022-01-04:
Hi,

Thanks for your interest in Aravis. Any improvements in performance is welcome.

As for the load test, you can use the camera simulator `arv-fake-gv-camera-0.8`. You can also set a small packet size on the receiving side, that will help to see how your changes make Aravis behave.

closed 2022-01-12 by EmmanuelP