AravisProject/aravis #627: Failed to connect to camera

Shorlaks, 2022-02-09:
**Describe the bug**
Aravis finds the camera but unable to connect.
I ran arv-camera-test with debug option.
```
pi@raspberrypi:~/aravis-0.8.20/build/src $ ./arv-camera-test-0.8 --debug=all
Looking for the first available camera
[16:05:40.361] 🅸 interface> Found 0 USB3Vision device (among 6 USB devices)
[16:05:40.363] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[16:05:40.364] 🅸 interface> [GvDiscoverSocket::new] Add interface 10.0.0.64 (10.255.255.255)
[16:05:40.369] 🅸 interface> [GvInterface::discovery] Device 'GigECamSim-6b:6e:05:05:22:c1' found (interface 10.0.0.64) user_id '' - MAC '6b:6e:05:05:22:c1'
[16:05:40.370] 🅸 device> [GvDevice::new] Interface address = 10.0.0.64
[16:05:40.370] 🅸 device> [GvDevice::new] Device address = 10.0.0.22
[16:05:40.871] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[16:05:40.872] 🆆 device> [GvDevice::read_memory] Ack reception timeout
[16:05:40.873] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:gigecamsim.xml; F000; 2B39' at 0x200
[16:05:40.873] 🅸 device> [GvDevice::load_genicam] Xml address = 0xf000 - size = 0x2b39 - gigecamsim.xml
[16:05:40.891] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceVendorName'
[16:05:40.892] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceModelName'
[16:05:40.892] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceVersion'
[16:05:40.893] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceManufacturerInfo'
[16:05:40.894] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[16:05:40.894] 🅸 genicam> [Gc::set_default_node_data] Add 'GevStreamChannelCount'
[16:05:40.895] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPHostPort'
[16:05:40.896] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSDoNotFragment'
[16:05:40.897] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[16:05:40.898] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCDA'
[16:05:40.901] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCSP'
[16:05:40.902] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[16:05:41.403] 🆆 device> [GvDevice::write_register] Ack reception timeout
[16:05:41.403] 🆆 device> [GvDevice::write_register] Ack reception timeout
[16:05:41.406] 🆆 device> [GvDevice::take_control] Can't get control access
[16:05:41.407] 🅸 device> [GvDevice::new] Device endianness = little
[16:05:41.407] 🅸 device> [GvDevice::new] Packet resend     = no
[16:05:41.407] 🅸 device> [GvDevice::new] Write memory      = yes
[16:05:41.407] 🅸 device> [GvDevice::new] Legacy endianness handling = yes
Failed to read the current device configuration: GigEVision read_memory error (invalid-access)
[16:05:41.409] 🆆 device> [GvDevice::leave_control] Can't relinquish control access
```
**Platform description:**
 - Aravis version 8.20
 - OS: Raspbian 64bit, Ubuntu 20.04 64bit
 - Hardware: RaspberryPi4

What seems to be the problem?

Thanks,
Shorlak

EmmanuelP, 2022-02-09:
Hi,

> Failed to read the current device configuration: GigEVision read_memory error (invalid-access)

Aravis tried to read a feature not available on your device. Could you rerun `arv-camera-test` with `--debug=3` ?

Does the viewer work ?

Shorlaks, 2022-02-09:
Hi,
`--debug=3` throws `Invalid debug selection`.

The viewer recognizes the camera, but when I hit **play** nothing happens.

Shorlaks, 2022-02-10:
Hi @EmmanuelP 
Here is the log, it's quite long so I upload it as a `.txt` file:
[log.txt](https://github.com/AravisProject/aravis/files/8041064/log.txt)

Thanks,
Shorlaks

EmmanuelP, 2022-02-14:
I have pushed a path to aravis main branch that should help to identify the missing feature. Could you try the current main branch, and attach a new log.txt file ?

Thanks.

Shorlaks, 2022-02-15:
Hi @EmmanuelP 
I've built the project from main branch, here is the log file:
[log.txt](https://github.com/AravisProject/aravis/files/8069664/log.txt)

correct me if I'm wrong but it seems like the GeniCam protocol is not implanted properly on the camera and is not responding to the discovery command.
also this time I can see the camera in the arv-viewer but the `play` button is disabled.
On the previous build I was able to hit `play` but nothing happened.

adrianca88, 2022-02-21:
Hello,

We have the same problem. However, it is not a problem with an invalid command, because it fails only sometimes. 

In fact, it works if we do a small change into _send_cmd_and_receive_ack  function to retry that command for io_data->gvcp_n_retries. 

Actually the code only retries it once because of:
  - packet_type == ARV_GVCP_PACKET_TYPE_ERROR
  - expected_answer == TRUE  
  - command_error != ARV_GVCP_ERROR_NONE

Here you can see a capture with debug info:

[image]

Do you know why are we getting this problem? We have review every camera and network parameters and it seems to be ok.

Can we create a pull request with this change or could be some collateral issues with that change?

Thanks so much in advance.

EmmanuelP, 2022-02-21:
Hi Adrian,

> Can we create a pull request with this modification or could be some collateral issues with that change?

Please go ahead. We will discuss about your patch there.

adrianca88, 2022-02-21:
done! 

https://github.com/AravisProject/aravis/pull/637