AravisProject/aravis #628: Receive images continuously using ArvStream in software trigger mode

GJChen919, 2022-02-10:
**Describe the bug**
Receive images continuously using ArvStream in software trigger mode

**To Reproduce**
I receives images with `ArvStream` based on the sample code - `arvexample.c` and I try to change the trigger source - software in the below codes:
```
arv_device_set_string_feature_value(dev, "AcquisitionMode", "Continuous", nullptr);
arv_device_set_string_feature_value(dev, "TriggerSelector", "AcquisitionStart", nullptr);
arv_device_set_string_feature_value(dev, "TriggerMode", "Off", nullptr);
arv_device_set_string_feature_value(dev, "TriggerSelector", "FrameStart", nullptr);
arv_device_set_string_feature_value(dev, "TriggerSource", "Software", nullptr);
arv_device_set_string_feature_value(dev, "TriggerActivation", "RisingEdge", nullptr);
arv_device_set_string_feature_value(dev, "TriggerMode", "On", nullptr);
std::cout << arv_device_get_string_feature_value(dev, "TriggerSelector", nullptr) << std::endl;
std::cout << arv_device_get_string_feature_value(dev, "TriggerSource", nullptr) << std::endl;
std::cout << arv_device_get_string_feature_value(dev, "TriggerMode", nullptr) << std::endl;
arv_camera_set_frame_rate(camera, 4.0, NULL);
```
But I still receives many frames after `AcquisitionStart` with no other operation.

**Expected behavior**
Unless I call `arv_camera_software_trigger or TriggerSoftware(arv_device_execute_command)`, it shouldn't comes any images generally. 

**Camera description:**
 - Cognex, Lucid, Hikvision
 - CIC-5000R, TRI245S-M, MV-CH250-21GM-F-NF
 - GigE

**Platform description:**
 - Aravis 0.8.20
 - OS: Windows 10 64bit
 - Hardware Intel I7

**Additional context**
My code is
[main_isu.zip](https://github.com/AravisProject/aravis/files/8039336/main_isu.zip)
 in the attatched

EmmanuelP, 2022-02-10:
Hi,

```
...
arv_camera_set_frame_rate(camera, 4.0, NULL);
```

This function actually changes the trigger settings, and makes the camera emit 4 images per second. You should remove this call if you want to trigger the image acquisition using `arv_camera_software_trigger()`.

Cheers.

closed 2022-02-10 by EmmanuelP

GJChen919, 2022-02-22:
@EmmanuelP Very thanks, it solved.