AravisProject/aravis #632: ArvGcCategory objects do not seem to have DisplayName/ToolTip/Description

TIS-Edgar, 2022-02-15:
Hi Emmanuel,

**Describe the bug**

Using functions like
arv_gc_feature_node_get_display_name 
or
arv_gc_feature_node_get_description

on category nodes returns NULL and not what the genicam document contains.

**To Reproduce**

A small c program that reproduces the problem can be found here:
https://gist.github.com/TIS-Edgar/a8d29750948c4a2fb53a05bb25b6bdaa

**Expected behavior**

DisplayName, ToolTip, Description, etc should be accessible.

**Camera description:**

Seems to be reproducable with all TIS cameras. GigE and USB3.
Verified with DMK 33UX273, DFK 33GP1300 and DFK 33GP1300.

**Platform description:**
 - current master (08dc0ba0)
 - OS: Debian Testing
 - Hardware: amd64

If you need any additional information, 
please let me know.

As always, thank you in advance!

EmmanuelP, 2022-02-15:
Hi Edgar,

Thanks for the report. Please test #633.

TIS-Edgar, 2022-02-15:
Hi Emmanuel,

That seems to fix everything.

Thank you very much!

PR #633 (2022-02-15): gccategory: accept all property nodes
Fixes #632
commit 182ba799 (2022-02-15): gccategory: accept all property nodes

Fixes #632
closed 2022-02-15 by EmmanuelP