AravisProject/aravis #638: arv_gc_feature_node_is_locked does not work for nodes with boolean 'pIsLocked' targets.

TIS-Edgar, 2022-02-23:
**Describe the bug**
`arv_gc_feature_node_is_locked` does not work for nodes that contain pIsLocked targets.

**To Reproduce**

Cameras offering genicam like the following exhibit wrong behavior.
```
	<Boolean Name="IrisAuto" NameSpace="Custom">
		<DisplayName>Auto Iris</DisplayName>
		<pValue>IrisAutoRegister</pValue>
		<OnValue>1</OnValue>
		<OffValue>0</OffValue>
	</Boolean>

	<Integer Name="Iris" NameSpace="Custom">
		<pIsLocked>IrisAuto</pIsLocked>
		<pValue>IrisRegister</pValue>
		<Min>0</Min>
		<Max>255</Max>
	</Integer>
```
Sample code:

```
ArvGcFeatureNode* iris_node = ...; // Iris node
GError* err = NULL;
gboolean is_locked = arv_gc_feature_node_is_locked(iris_node, &err);

if (is_locked == FALSE) // this will always be true!
```

**Expected behavior**

`arv_gc_feature_node_is_locked` should check pIsLocked.

**Camera description:**
Reproducible with all TIS cameras

**Platform description:**
Tested on amd64 Debian Testing and Linux Mint

If you require any additional information, please let me know.
As always, thank you in advance!

TIS-Edgar, 2022-02-23:
Updated description. Only pIsLocked bools are not correctly handled.

TIS-Edgar, 2022-02-23:
arv_gc_feature_node_is_locked uses arv_gc_property_node_get_int64 to get a value from the pointed to locked node. This function does not return a value (it returns an error) for boolean nodes.
https://github.com/AravisProject/aravis/blob/main/src/arvgcpropertynode.c#L378

EmmanuelP, 2022-02-23:
Hi Edgar,

Please try https://github.com/AravisProject/aravis/pull/639

TIS-Edgar, 2022-02-23:
Hi Emmanuel,

I can confirm, that #639 seems to solve this issue.

Thank you!

closed 2022-02-23 by EmmanuelP