AravisProject/aravis #641: ERROR: Program(s) ['g-ir-scanner'] not found or not executable

alexvatti, 2022-02-25:
yocto+poky+zeus: cross compiliton / build for arm:
Program g-ir-scanner found: NO
src/meson.build:297:2: ERROR: Program(s) ['g-ir-scanner'] not found or not executable

he Meson build system
Version: 0.51.2
/build/tmp/work/cortexa7t2hf-neon-mx7d-poky-linux-gnueabi/aravis/1.0-r0/git
Build dir: /build/tmp/work/cortexa7t2hf-neon-mx7d-poky-linux-gnueabi/aravis/1.0-r0/build
Build type: cross build
Project name: aravis
Project version: 0.8.20
C compiler for the build machine: gcc (gcc 7.5.0 "gcc (Ubuntu 7.5.0-3ubuntu118.04) 7.5.0")
C++ compiler for the build machine: g++ (gcc 7.5.0 "g++ (Ubuntu 7.5.0-3ubuntu118.04) 7.5.0")
C compiler for the host machine: arm-poky-linux-gnueabi-gcc -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 -Wformat -Wformat-security -Werror=format-security --sysroot=/media/WD4TBSSD/jenkins/jenkins-build/aravis-build-test/JADAK-Piccolo.YoctoPiccolo/build/tmp/work/cortexa7t2hf-neon-mx7d-poky-linux-gnueabi/aravis/1.0-r0/recipe-sysroot (gcc 9.2.0 "arm-poky-linux-gnueabi-gcc (GCC) 9.2.0")
C++ compiler for the host machine: arm-poky-linux-gnueabi-g++ -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 -Wformat -Wformat-security -Werror=format-security --sysroot=/media/WD4TBSSD/jenkins/jenkins-build/aravis-build-test/JADAK-Piccolo.YoctoPiccolo/build/tmp/work/cortexa7t2hf-neon-mx7d-poky-linux-gnueabi/aravis/1.0-r0/recipe-sysroot (gcc 9.2.0 "arm-poky-linux-gnueabi-g++ (GCC) 9.2.0")
Host machine cpu family: arm
Host machine cpu: arm
Target machine cpu family: arm
Target machine cpu: arm
Build machine cpu family: x86_64
Build machine cpu: x86_64
Compiler for C supports arguments -Wduplicated-branches: YES
Compiler for C supports arguments -Wimplicit-fallthrough: YES
Compiler for C supports arguments -Wmisleading-indentation: YES
Compiler for C supports arguments -Wstrict-prototypes: YES
Compiler for C supports arguments -Wunused: YES
Compiler for C supports arguments -Wunused-variable: YES
Compiler for C supports arguments -Wdeclaration-after-statement: YES
Compiler for C supports arguments -Wformat=2: YES
Compiler for C supports arguments -Wimplicit-function-declaration: YES
Compiler for C supports arguments -Winit-self: YES
Compiler for C supports arguments -Wmissing-include-dirs: YES
Compiler for C supports arguments -Wmissing-prototypes: YES
Compiler for C supports arguments -Wpointer-arith: YES
Compiler for C supports arguments -Wformat-nonliteral: YES
Compiler for C supports arguments -Wenum-conversion: NO
Compiler for C supports arguments -Wmaybe-uninitialized: YES
Compiler for C supports arguments -fvisibility=hidden: YES
Found pkg-config: /media/WD4TBSSD/jenkins/jenkins-build/aravis-build-test/JADAK-Piccolo.YoctoPiccolo/build/tmp/work/cortexa7t2hf-neon-mx7d-poky-linux-gnueabi/aravis/1.0-r0/recipe-sysroot-native/usr/bin/pkg-config (0.29.2)
Run-time dependency glib-2.0 found: YES 2.60.7
Run-time dependency gobject-2.0 found: YES 2.60.7
Run-time dependency gio-2.0 found: YES 2.60.7
Run-time dependency libxml-2.0 found: YES 2.9.9
Run-time dependency zlib found: YES 1.2.11
Found CMake: NO
Run-time dependency libusb-1.0 found: NO (tried pkgconfig and cmake)
Library m found: YES
Found pkg-config: /media/WD4TBSSD/jenkins/jenkins-build/aravis-build-test/JADAK-Piccolo.YoctoPiccolo/build/tmp/work/cortexa7t2hf-neon-mx7d-poky-linux-gnueabi/aravis/1.0-r0/recipe-sysroot-native/usr/bin/pkg-config (0.29.2)
Configuring arvapi.h using configuration
Configuring arvfeatures.h using configuration
Configuring arvversion.h using configuration
Program g-ir-scanner found: NO

src/meson.build:297:2: ERROR: Program(s) ['g-ir-scanner'] not found or not executable

Can you help me to resolve the issue

alexvatti, 2022-02-25:
SRCBRANCH="main"
SRC_URI = "git://git@github.com/AravisProject/aravis.git;protocol=ssh;branch=${SRCBRANCH}"

SRCREV="6c57c65b55d467ffa9ef635162c5a60711289613"

S = "${WORKDIR}/git"

inherit meson pkgconfig
EXTRA_OEMESON = "-Dintrospection=enabled -Dgst-plugin=enabled"

DEPENDS=" \
        gstreamer1.0 \
        gstreamer1.0-plugins-base \
        libxml2 \
        glib-2.0 \
        glib-2.0-native \
        gtk-doc \
        gobject-introspection \
        intltool-native \
"

EmmanuelP, 2022-02-25:
Hi,

I'm not used to cross compilation, but I guess you need `g-ir-scanner` on the build machine. Try to install the `gobject-introspection` package on it.

alexvatti, 2022-02-27:
**Case 1:** 
Without enable  - "introspection"
Build goes fine and but we need aravis-typelib

**Case 2:** 
For that I enabled:
EXTRA_OEMESON = "-Dintrospection=enabled"

**Getting follow error:**
Program g-ir-scanner found: NO
src/meson.build:297:2: ERROR: Program(s) ['g-ir-scanner'] not found or not executable

**Source code:**
Branch = "main"
with commit id: "6c57c65b55d467ffa9ef635162c5a60711289613"