AravisProject/aravis #648: memory_read Timeout on Imperx IP67-C1911

mlanting, 2022-03-28:
**Describe the bug**
When I attempt to set or even just read the pixel format from a C1911, I'm getting a memory_read Timeout error and the reported pixel format is null

[ArvDevice::set_status] Status changed ('[ArvDevice::read_memory] Timeout')

**To Reproduce**
Steps to reproduce the behavior:

code snippet:
```
std::string device_id = device;
camera_ = arv_camera_new(device_id.c_str(), NULL);
ArvPixelFormat format = arv_camera_get_pixel_format (camera_, NULL);
```

In terminal, `export ARV_DEBUG=camera:3,device:3'

Program outputs `[ArvDevice::set_status] Status changed ('[ArvDevice::read_memory] Timeout')` after attempting to read pixel format and returns NULL.

**Expected behavior**
arv_camera_get_pixel_format should return the default pixel format for the C1911 (0x01080009)

**Camera description:**
 - Manufacturer: Imperx
 - Model: IP67-C1911
 - Interface: GigE

**Platform description:**
 - Aravis version: 0.8.20
 - OS: Ubuntu 18.04
 - Hardware: x86_64

**Additional context**
Program is running as a ROS node. It was written for an older Imperx camera using Aravis 0.6 and I had to update to 0.8 for the C1911's since the chunk parameter functions in 0.6 are integer based but the newer camera treats chunking as a boolean (as does Aravis 0.8). I made minimal changes to our original ROS node while updating to 0.8 (basically just added NULL as a parameter to most of the function calls to ignore the new GERROR params).

EmmanuelP, 2022-03-29:
Please attach the output of:

```
arv-tool-0.8 -d all control PixelFormat
```

mlanting, 2022-03-29:
```
[13:53:14.964] 🅸 interface> Found 0 USB3Vision device (among 3 USB devices)
[13:53:14.964] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[13:53:14.964] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.57.100 (192.168.255.255)
[13:53:14.966] 🅸 interface> [GvInterface::discovery] Device 'Imperx, Inc-P67-C1911C-RC0000000-00310347' found (interface 192.168.57.100) user_id '310347' - MAC '08:ed:02:10:06:b2'
[13:53:14.966] 🅸 interface> [GvInterface::discovery] Device 'Imperx, Inc-P67-C1911C-RC0000000-00310346' found (interface 192.168.57.100) user_id '310346' - MAC '08:ed:02:10:06:b1'
Imperx, Inc-P67-C1911C-RC0000000-00310346 (192.168.57.200)
[13:53:15.968] 🅸 interface> Found 0 USB3Vision device (among 3 USB devices)
[13:53:15.968] 🅸 device> [GvDevice::new] Interface address = 192.168.57.100
[13:53:15.968] 🅸 device> [GvDevice::new] Device address = 192.168.57.200
[13:53:15.970] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:IpxGev_CheetahPregius_1.6.8.zip;30210000;a01c' at 0x200
[13:53:15.970] 🅸 device> [GvDevice::load_genicam] Xml address = 0x30210000 - size = 0xa01c - IpxGev_CheetahPregius_1.6.8.zip
[13:53:16.101] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[13:53:16.115] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[13:53:16.115] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[13:53:16.115] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[13:53:16.116] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[13:53:16.116] 🅸 genicam> [Gc::set_default_node_data] Add 'GevStreamChannelCount'
[13:53:16.116] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[13:53:16.116] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[13:53:16.119] 🅸 device> [GvDevice::new] Device endianness = big
[13:53:16.119] 🅸 device> [GvDevice::new] Packet resend     = yes
[13:53:16.119] 🅸 device> [GvDevice::new] Write memory      = yes
[13:53:16.119] 🅸 device> [GvDevice::new] Legacy endianness handling = no
PixelFormat read error: GigEVision read_memory error (invalid-access)
Imperx, Inc-P67-C1911C-RC0000000-00310347 (192.168.57.201)
[13:53:16.127] 🅸 interface> Found 0 USB3Vision device (among 3 USB devices)
[13:53:16.127] 🅸 device> [GvDevice::new] Interface address = 192.168.57.100
[13:53:16.127] 🅸 device> [GvDevice::new] Device address = 192.168.57.201
[13:53:16.129] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:IpxGev_CheetahPregius_1.6.8.zip;30210000;a01c' at 0x200
[13:53:16.129] 🅸 device> [GvDevice::load_genicam] Xml address = 0x30210000 - size = 0xa01c - IpxGev_CheetahPregius_1.6.8.zip
[13:53:16.260] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[13:53:16.273] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[13:53:16.273] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[13:53:16.273] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[13:53:16.274] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[13:53:16.274] 🅸 genicam> [Gc::set_default_node_data] Add 'GevStreamChannelCount'
[13:53:16.274] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[13:53:16.274] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[13:53:16.278] 🅸 device> [GvDevice::new] Device endianness = big
[13:53:16.278] 🅸 device> [GvDevice::new] Packet resend     = yes
[13:53:16.278] 🅸 device> [GvDevice::new] Write memory      = yes
[13:53:16.278] 🅸 device> [GvDevice::new] Legacy endianness handling = no
PixelFormat read error: GigEVision read_memory error (invalid-access)
```

romainreignier, 2022-05-02:
It looks like I have the same issue on an Imperx POE-C4110C camera:

```
./src/arv-tool-0.8 -d all control PixelFormat
[17:29:05.626] 🅸 interface> Found 0 USB3Vision device (among 12 USB devices)
[17:29:05.627] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1 (127.0.0.1)
[17:29:05.627] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.87.24 (192.168.87.255)
[17:29:05.627] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.0.100 (192.168.0.255)
[17:29:05.627] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.122.1 (192.168.122.255)
[17:29:05.627] 🅸 interface> [GvDiscoverSocket::new] Add interface 169.254.193.20 (169.254.193.255)
[17:29:05.629] 🅸 interface> [GvInterface::discovery] Device 'Imperx, Inc-POE-C4110C-RC0000000-50960031' found (interface 169.254.193.20) user_id '' - MAC '08:ed:02:10:08:35'
Imperx, Inc-POE-C4110C-RC0000000-50960031 (169.254.193.23)
[17:29:06.630] 🅸 interface> Found 0 USB3Vision device (among 12 USB devices)
[17:29:06.630] 🅸 device> [GvDevice::new] Interface address = 169.254.193.20
[17:29:06.631] 🅸 device> [GvDevice::new] Device address = 169.254.193.23
[17:29:06.632] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:IpxGev_CheetahPregius_1.6.8.zip;30210000;a01c' at 0x200
[17:29:06.633] 🅸 device> [GvDevice::load_genicam] Xml address = 0x30210000 - size = 0xa01c - IpxGev_CheetahPregius_1.6.8.zip
[17:29:06.761] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[17:29:06.796] 🅸 genicam> [Gc::set_default_node_data] Add 'DeviceID'
[17:29:06.796] 🅸 genicam> [Gc::set_default_node_data] Add 'GevStreamChannelCount'
[17:29:06.796] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[17:29:06.796] 🅸 genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[17:29:06.800] 🅸 device> [GvDevice::new] Device endianness = big
[17:29:06.800] 🅸 device> [GvDevice::new] Packet resend     = yes
[17:29:06.800] 🅸 device> [GvDevice::new] Write memory      = yes
[17:29:06.800] 🅸 device> [GvDevice::new] Legacy endianness handling = no
PixelFormat read error: [PixelFormat] [PixelFormat] [BayerBG12Packed] [PixelFormatBayerBG12pAvailExpr] [PixelFormatBayerBG12pAvailExpr] [IsBayerSensor] [IsBayerSensor] [SensorTypeValExpr] [SensorTypeValExpr] [IsPolarizationAvailableExpr] [IsPolarizationAvailableExpr] [RGS_Cam_InfoReg] GigEVision read_memory error (invalid-access)
```

The arv-viewer shows a blank pixel format combo box:
[image]

I am using the latest git version: 3c8ec492c1cf8fae28ea40c839c91cd96b1dd583

EmmanuelP, 2022-05-02:
Hi,

Please send me privately at emmanuel.pacaud@free.fr the Genicam data obtained using `arv-tool-0.8 genicam`.

romainreignier, 2022-05-02:
Hi @EmmanuelP 
I have just send you an email.

I have also noticed that the output of `arv-tool-0.8 values` is quite short:
```
./src/arv-tool-0.8 values
Imperx, Inc-POE-C4110C-RC0000000-50960031 (169.254.193.23)
Category    : 'Root'
    Category    : 'DeviceControl'
        StringReg    : [RO] 'DeviceVendorName' = 'Imperx, Inc'
        StringReg    : [RO] 'DeviceModelName' = 'POE-C4110C-RC0000000'
        StringReg    : [RO] 'DeviceManufacturerInfo' = 'Support: 1-561-989-0006'
        StringReg    : [RO] 'DeviceVersion' = '1.0.0'
        StringReg    : [RO] 'DeviceSerialNumber' = '50960031'
        StringReg    : [RW] 'DeviceUserID' = ''
        Integer      : [RO] 'DeviceSFNCVersionMajor' = 2
        Integer      : [RO] 'DeviceSFNCVersionMinor' = 3
        Integer      : [RO] 'DeviceSFNCVersionSubMinor' = 0
        Command      : [WO] 'DeviceReset'
        Command      : [WO] 'CameraHeadReset'
        Category    : 'TemperatureControl'
    Category    : 'VersionInfo'
        Integer      : [RO] 'FirmwareImage' = 10
        Integer      : [RO] 'CameraHeadFirmwareVersion' = 65792
        Integer      : [RO] 'FamilyID' = 15
        Integer      : [RO] 'XmlVersion' = 67080
        Integer      : [RO] 'GevFirmwareSwVersion' = 16983470
        Integer      : [RO] 'GevFirmwareHwVersion' = 16786862
    Category    : 'ImageFormatControl'
        Integer      : [RO] 'WidthMax' = 4112
        Integer      : [RO] 'HeightMax' = 3008
        Integer      : [RW] 'Width' = 4112
        Integer      : [RW] 'Height' = 3008
        Integer      : [RW] 'OffsetX' = 0
        Integer      : [RW] 'OffsetY' = 0
        Enumeration  : [RO] 'PixelSize' = 'Bpp8'
        Enumeration  : [RO] 'PixelColorFilter' = 'BayerRG'
        Enumeration  : [RW] 'TestPattern' = 'Off'
        Category    : 'SlaveAOI'
            Integer      : [RW] 'SAOI_Width' = 4112
            Integer      : [RW] 'SAOI_Height' = 3008
            Integer      : [RW] 'SAOI_OffsetX' = 0
            Integer      : [RW] 'SAOI_OffsetY' = 0
            Enumeration  : [RW] 'SAOI_Mode' = 'Disable'
    Category    : 'AcquisitionControl'
        Enumeration  : [RW] 'AcquisitionMode' = 'Continuous'
        Command      : [WO] 'AcquisitionStart'
        Command      : [WO] 'AcquisitionStop'
        Command      : [WO] 'AcquisitionAbort'
        Integer      : [RW] 'AcquisitionFrameCount' = 1
        Command      : [WO] 'TriggerSoftware'
    Category    : 'GainControl'
    Category    : 'AutoGainAndAutoExposureAndAutoIris'
        Category    : 'Control'
        Category    : 'Status'
    Category    : 'DataCorrection'
    Category    : 'WhiteBalance'
    Category    : 'Strobe'
        Category    : 'OUT1'
        Category    : 'OUT2'
    Category    : 'PulseGenerator'
    Category    : 'TransportLayerControl'
        Integer      : [RO] 'PayloadSize' = 12368896
        Category    : 'GigEVision'
            Integer      : [RO] 'GevMACAddress' = 9814034876469
            Enumeration  : [RW] 'GevGVSPExtendedIDMode' = 'Off'
            Boolean      : [RO] 'GevCurrentIPConfigurationLLA' = true
            Boolean      : [RW] 'GevCurrentIPConfigurationDHCP' = true
            Boolean      : [RW] 'GevCurrentIPConfigurationPersistentIP' = false
            Integer      : [RO] 'GevCurrentIPAddress' = 2852045079
            Integer      : [RO] 'GevCurrentSubnetMask' = 4294901760
            Integer      : [RO] 'GevCurrentDefaultGateway' = 0
            Integer      : [RW] 'GevPersistentIPAddress' = 0
            Integer      : [RW] 'GevPersistentSubnetMask' = 0
            Integer      : [RW] 'GevPersistentDefaultGateway' = 0
            Integer      : [RO] 'GevLinkSpeed' = 1000
            StringReg    : [RO] 'GevFirstURL' = 'Local:IpxGev_CheetahPregius_1.6.8.zip;30210000;a01c'
            StringReg    : [RO] 'GevSecondURL' = ''
            Enumeration  : [RW] 'GevCCP' = 'ControlAccess'
            Integer      : [RO] 'GevPrimaryApplicationSocket' = 35762
            Integer      : [RO] 'GevPrimaryApplicationIPAddress' = 2852045076
            Integer      : [RW] 'GevMCPHostPort' = 0
            Integer      : [RW] 'GevMCDA' = 0
            Integer      : [RW] 'GevMCTT' = 0
            Integer      : [RW] 'GevMCRC' = 0
            Integer      : [RO] 'GevMCSP' = 49152
            Boolean      : [RO] 'GevSCCFGUnconditionalStreaming' = false
            Integer      : [RW] 'GevSCPHostPort' = 0
            Command      : [RW] 'GevSCPSFireTestPacket'
            Boolean      : [RW] 'GevSCPSDoNotFragment' = false
            Integer      : [RW] 'GevSCPD' = 0
            Integer      : [RW] 'GevSCDA' = 0
            Integer      : [RO] 'GevSCSP' = 3011
            Integer      : [RW] 'GevSCPSPacketSize' = 7168
            Integer      : [RW] 'GevHeartbeatTimeout' = 3000
            Integer      : [RO] 'GevTimestampTickFrequency' = 100000000
            Integer      : [RO] 'GevTimestampTickFrequencyHigh' = 0
            Integer      : [RO] 'GevTimestampTickFrequencyLow' = 100000000
            Command      : [WO] 'GevTimestampControlLatch'
            Command      : [WO] 'GevTimestampControlReset'
            Integer      : [RO] 'GevTimestampValue' = 0
            Integer      : [RO] 'GevTimestampValueHigh' = 0
            Integer      : [RO] 'GevTimestampValueLow' = 0
    Category    : 'UserSetControl'
        Enumeration  : [RW] 'UserSetSelector' = 'Default'
        Command      : [WO] 'UserSetLoad'
        Command      : [WO] 'UserSetSave'
        Enumeration  : [RW] 'UserSetDefault' = 'Default'
    Category    : 'EventControl'
    Category    : 'SpecialFeatures'
        Integer      : [RW] 'TLParamsLocked' = 0
```

With classic values like `PixelFormat` and `ExposureTime` missing.

romainreignier, 2022-05-05:
Small update, we have noticed that replacing `_use_legacy_endianness_mechanism (port, length)` by `length == 4` here:
https://github.com/AravisProject/aravis/blob/3c8ec492c1cf8fae28ea40c839c91cd96b1dd583/src/arvgcport.c#L218-L220
allows to read the missing features and set the PixelFormat.

This camera is also lacking the `TriggerSelector` feature so `arvcamera.c` needs some changed to make the usage of this feature optional.

Some work is needed make these changes generic enough.

I have pushed some changes on this branch:
https://github.com/romainreignier/aravis/tree/imperx_support

romainreignier, 2022-05-05:
With these changes, `arv-camera-test` works:
```
./src/arv-camera-test-0.8 --duration 5
Looking for the first available camera
vendor name            = Imperx, Inc
model name             = POE-C4110C-RC0000000
device serial number   = 50960031
image width            = 4112
image height           = 3008
gain                   = 5 dB
payload                = 12368896 bytes
gv n_stream channels   = 1
gv current channel     = 0
gv packet delay        = 0 ns
gv packet size         = 7168 bytes
  7 frames/s -    86.6 MiB/s
 10 frames/s -     124 MiB/s
  9 frames/s -     111 MiB/s
 10 frames/s -     124 MiB/s
  9 frames/s -     111 MiB/s
n_completed_buffers    = 45
n_failures             = 0
n_underruns            = 0
n_timeouts             = 0
n_aborteds             = 0
n_missing_frames       = 0
n_size_mismatch_errors = 0
n_received_packets     = 79040
n_missing_packets      = 0
n_error_packets        = 0
n_ignored_packets      = 0
n_resend_requests      = 0
n_resent_packets       = 0
n_resend_ratio_reached = 0
n_resend_disabled      = 0
n_duplicated_packets   = 0
n_transferred_bytes    = 563475164
n_ignored_bytes        = 0
```

romainreignier, 2022-05-05:
To get some success on `arv-test` I had to add a 1 sec sleep time before calling `arv_camera_create_stream` in the `MultipleAcquisition` and `SoftwareTrigger` tests.

```
./src/arv-test-0.8
Found 1 device
Testing 'Imperx, Inc:POE-C4110C-RC0000000'
Genicam:Load                        SUCCESS
Genicam:Schema                      IGNORED
Properties:SensorSizeReadout        FAILURE [SensorWidth] Not found
Properties:SensorSizeCheck          IGNORED
Properties:GainReadout              SUCCESS
Properties:ExposureTimeReadout      SUCCESS
MultipleAcquisitionA:BufferCheck    SUCCESS 10/10
MultipleAcquisitionA:FrameRate      SUCCESS 9.52 Hz
SoftwareTrigger:BufferCheck         SUCCESS 5/5
MultipleAcquisitionB:BufferCheck    SUCCESS 10/10
MultipleAcquisitionB:FrameRate      SUCCESS 5.07 Hz
SingleAcquisition:BufferCheck       SUCCESS
Chunks:NoSupport                    FAILURE
GigEVision:NStreamChannels          SUCCESS
GigEVision:StreamChannel            SUCCESS
```