AravisProject/aravis #651: Linking glib failed, msys2 package

wooseokyourself, 2022-03-30:
I installed aravis as msys2 package, the information is [here](https://packages.msys2.org/base/mingw-w64-aravis).   
I construct simple C++ project which include the aravis with meson,  and following is my `meson.build` file.

```
project('tutorial', 'cpp')
cpp = meson.get_compiler('cpp')

dep0 = cpp.find_library('glib-2.0', dirs: 'C:/msys64/mingw64')
dep1 = cpp.find_library('libxml2', dirs: 'C:/msys64/mingw64')
dep2 = cpp.find_library('aravis-0.8', dirs: 'C:/msys64/mingw64')

cflags = include_directories('c:/msys64/mingw64/include/aravis-0.8', 'c:/msys64/mingw64/include/glib-2.0', 'C:/msys64/mingw64/lib/glib-2.0/include')

executable('demo', 'main.cpp', include_directories : cflags, dependencies : [dep0, dep1, dep2])
```

.. and when I compile the project, it fails with following prints.
```
[2/2] Linking target demo.exe
FAILED: demo.exe
"c++"  -o demo.exe demo.exe.p/main.cpp.obj "-Wl,--allow-shlib-undefined" "-Wl,--start-group" "C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../lib/libglib-2.0.dll.a" "C:/msys64/mingw64/bin/../lib/gcc/x./lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../lib/libxml2.dll.a" "C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../lib/libaravis-0.8.dll.a" "-Wl,--subsystem,console" "-lkernel32" "-luser32di32" lwpoo" "-lgdi32" "-lwinspool" "-lshell32" "-lole32" "-loleaut32" "-luuid" "-lcomdlg32" "-ladvapi32" "-Wl,--end-group"
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: demo.exe.p/main.cpp.obj:c:/msys64/mingw64/include/aravis-0.8/arvbuffer.h:90: undefined reference to `g_type_check_instanceck_instance_is_a'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: demo.exe.p/main.cpp.obj:c:/msys64/mingw64/include/aravis-0.8/arvcamera.h:39: undefined reference to `g_type_check_instanceck_instance_is_a'
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: demo.exe.p/main.cpp.obj:C:\Users\woose\example\builddir/../main.cpp:26: undefined reference to `g_object_unref'       '
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/11.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: demo.exe.p/main.cpp.obj: in function `main':
C:\Users\woose\example\builddir/../main.cpp:30: undefined reference to `g_object_unref'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```

I've couldn't figure it out what I missed. Anyone help?

EmmanuelP, 2022-03-30:
Hi,

glib is in fact several libraries: glib, gobject and gio.

You should add a dependency on gobject: `dep3 = cpp.find_library('gobject-2.0', dirs: 'C:/msys64/mingw64')`

EmmanuelP, 2022-03-30:
By the way, if you have pkgconfig available, it is better to use the dependency function. Have a look at the [aravis-c-examples:](https://github.com/AravisProject/aravis-c-examples/blob/main/meson.build):

https://github.com/AravisProject/aravis-c-examples/blob/ebcec6280945a368c36d791cace569b1b7ea1d52/meson.build#L1-L14

closed 2022-03-30 by EmmanuelP

wooseokyourself, 2022-03-30:
It's working now! Sorry for silly questions and thanks.