AravisProject/aravis #652: arv-camera-test do not handle control-lost properly in USB Async mode

romainreignier, 2022-03-30:
**Describe the bug**
When running `arv-camera-test` with the option `-s async`, the `control-lost` callback is only called after a `Ctrl+C`

Program output:

```
  0 frame/s  -       0 MiB/s
^C  0 frame/s  -       0 MiB/s
n_completed_buffers    = 98
n_failures             = 0
n_underruns            = 6468697324
n_transferred_bytes    = 1222839296
n_ignored_bytes        = 0
Control lost
```

And the program uses 100% of the CPU.

**To Reproduce**

1. Start `arv-camera-test-0.8 -s async`
2. Unplug the USB cable
3. Wait but the program never stops and it uses 100% of one CPU core

**Expected behavior**

1. Start `arv-camera-test-0.8 -s async`
2. Unplug the USB cable
3. arv-camera-test exits properly with the output `Control lost`

In USB sync mode, it works properly.

**Camera description:**
 - Manufacturer: Allied Vision
 - Model: Alvium 1800U
 - Interface: USB3

**Platform description:**
 - Aravis version: git version, branch main, commit: b303b11ecc
 - OS: Ubuntu 18.04
 - Hardware: x86_64

**Additional context**
In my application, I have noticed that in USB Async mode, in case of a control-lost event, the buffer memory is not released while the memory is freed in USB Sync mode and Ethernet mode.
I have then tried to reproduce the issue with `arv-camera-test` and then I have noticed that the `control-lost` was never called and unplugging the camera makes the process using the CPU a lot.

romainreignier, 2022-03-30:
From a perf capture, it seems that the CPU usage is spent in the `arv_stream_pop_input_buffer` function:
[image]

EmmanuelP, 2022-03-30:
Indeed.

There is several issues in the async variant:

* we don't signal a control-lost if an error is detected in submit_transfer
* if an error happens, buffers popped from the pool may never return to the pool, and leaked.
* the hash table keeping track of the transfer contexts is using a buffer pointer as the hash value. If a buffer is deallocated and a new one is created at the same memory location, the transfer context will still continue to use incorrectly configured bulk transfers.

romainreignier, 2022-03-30:
The status `-4` (`LIBUSB_ERROR_NO_DEVICE`) is returned from `libusb_submit_transfer()`:

```
[15:57:17.598] 🆆 stream-thread> Payload transfer failed: transfer->status = 5
[15:57:17.598] 🆆 stream-thread> Payload transfer failed: transfer->status = 5
[15:57:17.598] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.598] 🆆 stream-thread> Payload transfer failed: transfer->status = 5
[15:57:17.598] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.598] 🆆 stream-thread> Payload transfer failed: transfer->status = 5
[15:57:17.598] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.598] 🆆 stream-thread> Leader transfer failed: transfer->status = 5
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> Trailer transfer failed: transfer->status = 5
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> Payload transfer failed: transfer->status = 5
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> Payload transfer failed: transfer->status = 5
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> Payload transfer failed: transfer->status = 5
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[15:57:17.599] 🆆 stream-thread> Payload transfer failed: transfer->status = 5
[15:57:17.599] 🆆 stream-thread> libusb_submit_transfer failed (-4)
```

From my understand of the following code:
https://github.com/AravisProject/aravis/blob/d817b638e857d0b09fa34eab337674e75204791a/src/arvuvstream.c#L344-L368

A success and a failed transfer both return the function so the thread keep running, trying to pop a buffer that will never come because the transfer failed.

Did I miss something?

romainreignier, 2022-03-30:
Oh ok, I have just seen your comment.

In the sync mode, I have noticed that I really need to spend less than 2 ms in the buffer callback otherwise I start getting `ARV_BUFFER_STATUS_SIZE_MISMATCH` errors and the stream never comes back. Is it expected?
Note that this happen in the case when I do not limit the bandwidth manually and the camera seems to adapt the framerate to send as much images as possible per second.

EmmanuelP, 2022-03-30:
> * if an error happens, buffers popped from the pool may never return to the pool, and leaked.
>
> * the hash table keeping track of the transfer contexts is using a buffer pointer as the hash value. If [...]

Reported in #654 and #655

EmmanuelP, 2022-04-01:
Fixed by #653

closed 2022-04-01 by EmmanuelP