AravisProject/aravis #654: Possible buffer leak in uv_stream in async mode

EmmanuelP, 2022-03-30:
It is not completely clear if  arv_uv_stream_trailer_cb is always called for every pulled buffer from the buffer pool.

https://github.com/AravisProject/aravis/blob/d817b638e857d0b09fa34eab337674e75204791a/src/arvuvstream.c#L174-L220

romainreignier, 2022-03-30:
I am not sure if it is related to this issue, but disconnecting the camera in async mode leads to memory leaks:

```
valgrind --tool=memcheck --leak-check=yes --suppressions=../tests/aravis.supp ./src/arv-camera-test-0.8 -s async
```
Returns the following summary:

> ==7624== LEAK SUMMARY:
> ==7624==    **definitely lost: 5,005 bytes in 56 blocks**
> ==7624==    **indirectly lost: 49,476,144 bytes in 9 blocks**
> ==7624==      possibly lost: 12,368,928 bytes in 2 blocks
> ==7624==    still reachable: 50,112 bytes in 274 blocks
> ==7624==                       of which reachable via heuristic:
> ==7624==                         length64           : 1,072 bytes in 22 blocks
> ==7624==                         newarray           : 1,648 bytes in 23 blocks
> ==7624==         suppressed: 52,649 bytes in 533 blocks

While in sync mode I have:

> ==7513== LEAK SUMMARY:
> ==7513==    **definitely lost: 5 bytes in 1 blocks**
> ==7513==    **indirectly lost: 0 bytes in 0 blocks**
> ==7513==      possibly lost: 32 bytes in 1 blocks
> ==7513==    still reachable: 50,103 bytes in 273 blocks
> ==7513==                       of which reachable via heuristic:
> ==7513==                         length64           : 1,072 bytes in 22 blocks
> ==7513==                         newarray           : 1,648 bytes in 23 blocks
> ==7513==         suppressed: 52,649 bytes in 533 blocks

Or, with [heaptrack](https://github.com/KDAB/heaptrack):
Async:

> total runtime: 5.57s.
> calls to allocation functions: 93992 (16880/s)
> temporary memory allocations: 28016 (5031/s)
> peak heap memory consumption: 624.45M
> peak RSS (including heaptrack overhead): 635.56M
> **total memory leaked: 53.55M**

Sync:

> total runtime: 5.55s.
> calls to allocation functions: 1851278 (333383/s)
> temporary memory allocations: 28310 (5098/s)
> peak heap memory consumption: 625.26M
> peak RSS (including heaptrack overhead): 635.30M
> **total memory leaked: 4.07M**

With the heaptrack flamegraph in async mode, we can see that the leaked memory is from the `arv_buffer_new`:
[image]

But in the logs, the buffers are said to be flushed at the end:
```
[19:46:48.347] 🆆 stream-thread> Control lost
libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
[19:46:48.347] 🆆 stream-thread> libusb_submit_transfer failed (-4)
  0 frame/s  -       0 MiB/s - 1 error
n_completed_buffers    = 57
n_failures             = 1
n_underruns            = 0
n_transferred_bytes    = 721710080
n_ignored_bytes        = 0
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::read_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🆆 device> [UvDevice::write_memory] Command sending error: Not connected
[19:46:48.914] 🅸 stream> [UvStream::finalize] n_completed_buffers    = 57
[19:46:48.914] 🅸 stream> [UvStream::finalize] n_failures             = 1
[19:46:48.914] 🅸 stream> [UvStream::finalize] n_underruns            = 0
[19:46:48.914] 🅸 stream> [UvStream::finalize] n_transferred_bytes    = 721710080
[19:46:48.914] 🅸 stream> [UvStream::finalize] n_ignored_bytes        = 0
[19:46:48.914] 🅸 stream> [Stream::finalize] Flush 48 buffer[s] in input queue
[19:46:48.914] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue
```

romainreignier, 2022-03-31:
I could not test it in the `arv-camera-test` before the control-lost patch applied but I have seen the leak in my application before.

EmmanuelP, 2022-03-31:
Should be fixed by dacb545e3d7085a3dd116707391f4c1b33b26f01

romainreignier, 2022-03-31:
I will try it tomorrow.
Thanks!

romainreignier, 2022-04-01:
Either in `arv-camera-test` and in my application, the memory consumed stays constant across disconnections:
[image]
Great! :1st_place_medal:

EmmanuelP, 2022-04-01:
Fixed by #653

closed 2022-04-01 by EmmanuelP