AravisProject/aravis #656: Evaluator returns wrong address for BalanceRatio

TIS-Edgar, 2022-03-31:
**Describe the bug**

Setting white balance via BalanceRatio/BalanceRatioRaw seems to change all wb channel to the same value.

The evaluator returns the wrong address for `BalanceRatio`/`BalanceRatioRaw`.
Reduced output of the sample below with `ARV_DEBUG=gvcp:3,evaluator:3`:
```
[09:12:30.352] 🅳 evaluator> [Evaluator::set_int64_variable] ADDR_GREEN = 0x4c525443cc0000c0
[09:12:30.352] 🅳 evaluator> [Evaluator::set_int64_variable] ADDR_BLUE = 0x4c525443cc0000d0
[09:12:30.352] 🅳 evaluator> [Evaluator::set_int64_variable] ADDR_RED = 0x4c525443cc0000b0
[09:12:30.352] 🅳 evaluator> [Evaluator::set_int64_variable] SEL = 0x2
[09:12:30.352] 🅳 evaluator> [Evaluator::evaluate_as_int64] Expression = '(SEL = 0) ? ADDR_RED : ((SEL = 1) ? ADDR_BLUE : ADDR_GREEN)'
[09:12:30.352] 🅳 evaluator> (var) SEL = 2
[09:12:30.352] 🅳 evaluator> (int64) 0
[09:12:30.352] 🅳 evaluator> (operator) =
[09:12:30.352] 🅳 evaluator> (var) ADDR_RED = 5.49955e+18
[09:12:30.352] 🅳 evaluator> (var) SEL = 2
[09:12:30.352] 🅳 evaluator> (int64) 1
[09:12:30.352] 🅳 evaluator> (operator) =
[09:12:30.352] 🅳 evaluator> (var) ADDR_BLUE = 5.49955e+18
[09:12:30.352] 🅳 evaluator> (var) ADDR_GREEN = 5.49955e+18
[09:12:30.352] 🅳 evaluator> (operator) :
[09:12:30.352] 🅳 evaluator> (operator) ?
[09:12:30.352] 🅳 evaluator> (operator) :
[09:12:30.352] 🅳 evaluator> (operator) ?
[09:12:30.352] 🅳 evaluator> [Evaluator::evaluate] Result = (int64) 4c525443cc000000
[09:12:30.353] 🅳 evaluator> [Evaluator::set_int64_variable] ADDR_GREEN = 0x4c525443cc0000c0
[09:12:30.353] 🅳 evaluator> [Evaluator::set_int64_variable] ADDR_BLUE = 0x4c525443cc0000d0
[09:12:30.353] 🅳 evaluator> [Evaluator::set_int64_variable] ADDR_RED = 0x4c525443cc0000b0
[09:12:30.353] 🅳 evaluator> [Evaluator::evaluate_as_int64] Expression = '(SEL = 0) ? ADDR_RED : ((SEL = 1) ? ADDR_BLUE : ADDR_GREEN)'
[09:12:30.353] 🅳 evaluator> (var) SEL = 2
[09:12:30.353] 🅳 evaluator> (int64) 0
[09:12:30.353] 🅳 evaluator> (operator) =
[09:12:30.353] 🅳 evaluator> (var) ADDR_RED = 5.49955e+18
[09:12:30.353] 🅳 evaluator> (var) SEL = 2
[09:12:30.353] 🅳 evaluator> (int64) 1
[09:12:30.353] 🅳 evaluator> (operator) =
[09:12:30.353] 🅳 evaluator> (var) ADDR_BLUE = 5.49955e+18
[09:12:30.353] 🅳 evaluator> (var) ADDR_GREEN = 5.49955e+18
[09:12:30.353] 🅳 evaluator> (operator) :
[09:12:30.353] 🅳 evaluator> (operator) ?
[09:12:30.353] 🅳 evaluator> (operator) :
[09:12:30.353] 🅳 evaluator> (operator) ?
[09:12:30.353] 🅳 evaluator> [Evaluator::evaluate] Result = (int64) 4c525443cc000000
```
The values are in hex as i modified the debug output to see the real addresses are they can be seen in wireshark.
The evaluator should return ADDR_GREEN/ADDR_BLUE/ADDR_RED. Instead it returns 4c525443cc000000, the address of gain.
Changing BalanceRatioSelector does not change the output.

**To Reproduce**
Steps to reproduce the behavior:

Set `BalanceRatioSelector` and set `BalanceRatioRaw` to a random value.
The other balance ratio channels will appear to have the same value.

A small program for reproduction can be found here:
https://gist.github.com/TIS-Edgar/5387b38652f757ab71925e0246effde2

**Expected behavior**
The evaluator should return the correct address so that white balance can be correctly set.

**Camera description:**
 - This seems to affect all TIS usb 33 cameras
 - GigE cameras behave correctly

**Platform description:**
 - aravis main HEAD
 - OS: Debian Testing, Ubuntu 18.04
 - x86_64

If you require any additional information, please let me know.

EmmanuelP, 2022-03-31:
Thanks Edgar,

Please test https://github.com/AravisProject/aravis/pull/657

It was a typo unnoticed for several years.

TIS-Edgar, 2022-03-31:
Hi Emmanuel,

#657 fixes the problem.

Thank you very much!

EmmanuelP, 2022-03-31:
Thanks for the report.

commit 1a35c406 (2022-03-31): misc: fix parameter type of arv_value_set_int64

A typo never noticed for several years. Led to loss of information.

Fix #656
closed 2022-03-31 by EmmanuelP