AravisProject/aravis #661: Feature need: Configure or set CTI File

garfeng, 2022-04-06:
**Is your feature request related to a problem? Please describe.**

The manufacturer (HikRobots) provides a net card.

 When I connect my Gige camera from their net card, the aravis can not detect it. But aravis works properly when camera is connected  to the default Ethernet. 

 HikRobots software MVS (Select CTI File) : 

[image]

[image]

Then, the MVS detected the net card.

[image]

and the camera is listed to the net card
[image]

**Describe the solution you'd like**

May you provide a api for us to set the CTI file. Or any other solutions. 

**Describe alternatives you've considered**

Any way for me to select CTI file.

**Additional context**

EmmanuelP, 2022-04-06:
Hi @garfeng 

Aravis uses its own transport implementation, with no support for an external one.

I have little interest in trying to make Aravis use proprietary transport implementations. CTIs mainly exist because manufacturers artificially limit their implementation to their own devices, while Aravis is manufacturer agnostic.

That being said, if someone proposes a patch that implements this support, as long as it optional, and implemented in Aravis as ArvDevice, ArvInterface and  ArvStream subclasses, I will gladly accept it.

closed 2022-04-06 by EmmanuelP

garfeng, 2022-04-06:
Thank you very much  @EmmanuelP !

I researched the SDK provided by the manufacturer today and feel that if I use it, it will be difficult to switch to other brands of hardware in the future
That is, Aravis is better for me, reliable and open source.
Thank you very much again for your excellent work !