AravisProject/aravis #663: Aravis arv_camera_stop_acquisition followed by arv_camera_start_acquisition Fails

kim7604, 2022-04-20:
We are using aravis-0.8.20 library.  When we issue a arv_camera_stop_acquisition, acquisition stops successfully.  We then call arv_camera_start_acquisition a few seconds later but we do not always get video frames.  If I enable aravis debugging, export ARV_DEBUG=stream:3,device:3, then I see the below when arv_camera_start_acquisition

[11:46:48.712] ?¸ stream> [UvStream::finalize] n_completed_buffers    = 2017
[11:46:48.712] ?¸ stream> [UvStream::finalize] n_failures             = 0
[11:46:48.712] ?¸ stream> [UvStream::finalize] n_underruns            = 104
[11:46:48.712] ?¸ stream> [UvStream::finalize] n_transferred_bytes    = 4182620628
[11:46:48.713] ?¸ stream> [UvStream::finalize] n_ignored_bytes        = 215269920
[11:46:48.713] ?¸ stream> [Stream::finalize] Flush 0 buffer[s] in input queue
[11:46:48.713] ?¸ stream> [Stream::finalize] Flush 0 buffer[s] in output queue
[11:46:50.892] ?¸ device> [UvDevice::new] GUID    = 247910000045
[11:46:50.901] ?¸ device> [UvDevice::new] Using control endpoint 2, interface 0
[11:46:50.901] ?¸ device> [UvDevice::new] Using data endpoint 1, interface 2
[11:46:50.903] ?¸ device> Get genicam
[11:46:50.905] ?¸ device> MANUFACTURER_NAME =        'Smartek'
[11:46:50.912] ?¸ device> MAX_DEVICE_RESPONSE_TIME = 0x0000012c
[11:46:50.912] ?¸ device> DEVICE_CAPABILITY        = 0x0000000000000b01
[11:46:50.912] ?¸ device> SRBM_ADDRESS =             0x000000000001b27c
[11:46:50.912] ?¸ device> MANIFEST_TABLE_ADDRESS =   0x000000000002b268
[11:46:50.919] ?¸ device> U3VCP_CAPABILITY =         0x00000001
[11:46:50.920] ?¸ device> MAX_CMD_TRANSFER =         0x00000400
[11:46:50.920] ?¸ device> MAX_ACK_TRANSFER =         0x00000400
[11:46:50.920] ?¸ device> SIRM_OFFSET =              0x000000000002b238
[11:46:50.940] ?¸ device> SIRM_INFO =                  0x02000000
[11:46:50.940] ?¸ device> SIRM_CONTROL =               0x00000000
[11:46:50.940] ?¸ device> SIRM_REQ_PAYLOAD_SIZE =      0x00000000001fa400
[11:46:50.940] ?¸ device> SIRM_REQ_LEADER_SIZE =       0x00000400
[11:46:50.940] ?¸ device> SIRM_REQ_TRAILER_SIZE =      0x00000400
[11:46:50.941] ?¸ device> SIRM_MAX_LEADER_SIZE =       0x00000400
[11:46:50.941] ?¸ device> SIRM_PAYLOAD_SIZE =          0x00100000
[11:46:50.941] ?¸ device> SIRM_PAYLOAD_COUNT =         0x00000001
[11:46:50.941] ?¸ device> SIRM_TRANSFER1_SIZE =        0x000fa400
[11:46:50.941] ?¸ device> SIRM_TRANSFER2_SIZE =        0x00000000
[11:46:50.941] ?¸ device> SIRM_MAX_TRAILER_SIZE =      0x00000400
[11:46:50.945] ?¸ device> MANIFEST_N_ENTRIES =       0x0000000000000001
[11:46:50.946] ?¸ device> MANIFEST ENTRY
                         00000000 00 00 00 03 00 04 00 01 cc 34 01 00 00 00 00 00  .........4......
                         00000010 58 79 00 00 00 00 00 00 00 00 00 00 00 00 00 00  Xy..............
                         00000020 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
                         00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
[11:46:50.946] ?¸ device> genicam address =          0x00000000000134cc
[11:46:50.946] ?¸ device> genicam size    =          0x0000000000007958
[11:46:51.388] ?¸ device> zip file =                 UC2_CMOS.xml
[11:46:52.058] ?¸ stream> SIRM_INFO             = 0x02000000
[11:46:52.059] ?¸ stream> SIRM_REQ_PAYLOAD_SIZE = 0x00000000001fa400
[11:46:52.059] ?¸ stream> SIRM_REQ_LEADER_SIZE  = 0x00000400
[11:46:52.059] ?¸ stream> SIRM_REQ_TRAILER_SIZE = 0x00000400
[11:46:52.059] ?¸ stream> Required alignment    = 4
[11:46:52.070] ?¸ stream> SIRM_PAYLOAD_SIZE     = 0x00100000
[11:46:52.070] ?¸ stream> SIRM_PAYLOAD_COUNT    = 0x00000001
[11:46:52.070] ?¸ stream> SIRM_TRANSFER1_SIZE   = 0x000fa400
[11:46:52.070] ?¸ stream> SIRM_TRANSFER2_SIZE   = 0x00000000
[11:46:52.070] ?¸ stream> SIRM_MAX_LEADER_SIZE  = 0x00000400
[11:46:52.070] ?¸ stream> SIRM_MAX_TRAILER_SIZE = 0x00000400

We run any USB3Vision camera and we observe the same issue.  Most recently with Smarttek Vision UCC4113C.
We are running linux x64 on a Snapdragon820 processor.

We have tried version 0.8.21 of Aravis and still see the issue.

Thanks,
Kim

EmmanuelP, 2022-04-22:
Hi,

Could you please provide the shortest sample code that exhibits this issue ? That would help a lot.

Thanks.

kim7604, 2022-04-25:
Hi Emmanual,  this is a smaller piece of our code exhibiting the issue:

EmmanuelP, 2022-04-27:
Hi,

I don't know the motivation for the aravisCameraTask, but be careful when you are using Aravis objects from different threads. Aravis API is not tread safe, except for a small set of functions.

Also keep in mind the new-buffer callback is called from the Aravis stream receiving thread. So for example the call to `arv_camera_stop_acquisition` should be avoided, or at least protected by mutexes.

I'm closing this issue as it is not obvious wether Aravis is at fault here.

You may reopen the issue if you think there is something wrong in Aravis, but please attach a small example that can produce an executable and allow us to reproduce the issue.

Cheers.

closed 2022-04-27 by EmmanuelP

ban1080, 2022-06-09:
I have a very similar issue when using gstreamer. Usually after 8 stop-starts the framerate begins to deteriorate rapidly.

EmmanuelP, 2022-06-10:
> I have a very similar issue when using gstreamer. Usually after 8 stop-starts the framerate begins to deteriorate rapidly.

Hi @ban1080 

Please open a new issue, this one is not about the GStreamer plugin.