AravisProject/aravis #668: Arv-viewer works, but not gstreamer

BrendanTDY, 2022-04-28:
Hi!  I have a Dalsa M1450 camera, and it works fine with the arv-viewer application, but not in gstreamer.  I tried searching the internal data stream error + aravis, but didnt find anything helpful.  Would appreciate any assistance you could provide. 

**Camera description:**
 - Manufacturer: Teledyne Dalsa
 - Model: M1450
 - 1 GigE

**Platform description:**
 - Aravis version: aravis-0.8.10
 - Ubuntu/L4T: (Linux sales-nano 4.9.140-tegra 1 SMP PREEMPT Wed Mar 13 00:32:22 PDT 2019 aarch64 aarch64 aarch64 GNU/Linux)
 - Nvidia Jetson Nano 4GB

**Output from: gst-launch-1.0 -vvv -e aravissrc**
ubuntu@sales-nano:~$ sudo ./aravisexport.sh; gst-launch-1.0 -vvv -e aravissrc
nvbuf_utils: Could not get EGL display connection
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstAravis:aravis0.GstPad:src: caps = video/x-raw, format=(string)GRAY8, width=(int)728, height=(int)544, framerate=(fraction)30/1
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming stopped, reason not-linked (-1)
EOS on shutdown enabled -- waiting for EOS after Error
Waiting for EOS...
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Interrupt while waiting for EOS - stopping pipeline...
Execution ended after 0:00:09.798714780
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ... 

**Output from: arv-camera-test-0.8 -d all**
ubuntu@sales-nano:~/Downloads$ arv-camera-test-0.8 -d all
Looking for the first available camera
[15:52:22.074] 🅸 interface> Found 0 USB3Vision device (among 6 USB devices)
[15:52:22.075] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1
[15:52:22.075] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.1.55
[15:52:22.075] 🅸 interface> [GvDiscoverSocket::new] Add interface 192.168.55.1
[15:52:22.075] 🅸 interface> [GvDiscoverSocket::new] Add interface 172.17.0.1
[15:52:22.076] 🅸 interface> [GvInterface::discovery] Device 'Teledyne DALSA-Nano-M1450-S1158235' found (interface 192.168.1.55) user_id 'S1158235' - MAC '00:01:0d:c3:4f:8d'
[15:52:22.076] 🅸 device> [GvDevice::new] Interface address = 192.168.1.55
[15:52:22.076] 🅸 device> [GvDevice::new] Device address = 192.168.1.69
[15:52:22.076] 🅸 device> [GvDevice::load_genicam] xml url = 'Local:TeledyneDALSA_Nano-IMX273_Mono_1.6M_e0d244cf_10CA18.0004.zip;b000;15325' at 0x200
[15:52:22.076] 🅸 device> [GvDevice::load_genicam] Xml address = 0xb000 - size = 0x15325 - TeledyneDALSA_Nano-IMX273_Mono_1.6M_e0d244cf_10CA18.0004.zip
[15:52:22.111] 🅸 device> [GvDevice::load_genicam] Zipped xml data
[15:52:22.118] 🅸 dom> [Genicam::create_element] Unknown tag (pAlias)
[15:52:22.119] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.121] 🅸 dom> [Genicam::create_element] Unknown tag (pError)
[15:52:22.121] 🅸 dom> [Genicam::create_element] Unknown tag (pValueCopy)
[15:52:22.122] 🅸 dom> [Genicam::create_element] Unknown tag (pError)
[15:52:22.131] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.142] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.145] 🅸 dom> [Genicam::create_element] Unknown tag (pBlockPolling)
[15:52:22.148] 🅸 dom> [Genicam::create_element] Unknown tag (pBlockPolling)
[15:52:22.149] 🅸 dom> [Genicam::create_element] Unknown tag (pBlockPolling)
[15:52:22.153] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.159] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.161] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.170] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.171] 🅸 dom> [Genicam::create_element] Unknown tag (pBlockPolling)
[15:52:22.178] 🅸 dom> [Genicam::create_element] Unknown tag (pBlockPolling)
[15:52:22.180] 🅸 dom> [Genicam::create_element] Unknown tag (pBlockPolling)
[15:52:22.182] 🅸 dom> [Genicam::create_element] Unknown tag (pBlockPolling)
[15:52:22.184] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.188] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[15:52:22.190] 🅸 dom> [GcFeature::set_attribute] Unknown attribute 'Comment'
[... 82 lines cut ...]
-------------
>=  8000;       0
<      0;       0
min     ;    3302
max     ;    6223
last max
at:     ;       8
Counter =      332
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_completed_buffers    = 332
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_failures             = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_timeouts             = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_aborteds             = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_underruns            = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_missing_frames       = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_size_mismatch_errors = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_received_packets     = 90636
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_missing_packets      = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_error_packets        = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_ignored_packets      = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_resend_requests      = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_resent_packets       = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_resend_ratio_reached = 0
[15:52:30.364] 🅸 stream> [GvStream::finalize] n_duplicated_packets   = 0
[15:52:30.364] 🅸 stream> [Stream::finalize] Flush 50 buffer[s] in input queue
[15:52:30.364] 🅸 stream> [Stream::finalize] Flush 0 buffer[s] in output queue

EmmanuelP, 2022-04-29:
Hi,

aravissrc is a gstreamer src, useless alone. You need a complete pipeline with a sink.

This is the simplest pipeline that will display a video stream:

```
gst-launch-10. aravissrc ! videoconvert ! autovideosink
```

closed 2022-04-29 by EmmanuelP

BrendanTDY, 2022-04-29:
Hi Emmanuel.  Thanks for your prompt message and my apologized for the oversite.    I tried the pipeline you suggested and still getting the same internal data stream error.  Could you kindly advise?  

ubuntu@sales-nano:~$ gst-launch-1.0 -v -e --gst-plugin-path=/usr/local/lib/ aravissrc ! videoconvert ! autovideosink

(gst-plugin-scanner:6042): GStreamer-WARNING **: 10:43:36.099: Failed to load plugin '/usr/local/lib/python2.7/dist-packages/cv2/python-2.7/cv2.so': libopencv_ml.so.4.5: cannot open shared object file: No such file or directory
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstAravis:aravis0.GstPad:src: caps = video/x-raw, width=(int)728, height=(int)544, framerate=(fraction)1/1, format=(string)GRAY8
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)728, height=(int)544, framerate=(fraction)1/1, format=(string)GRAY8
/GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = video/x-raw, width=(int)728, height=(int)544, framerate=(fraction)1/1, format=(string)GRAY8
/GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0.GstGhostPad:sink: caps = video/x-raw, width=(int)728, height=(int)544, framerate=(fraction)1/1, format=(string)GRAY8
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:sink: caps = video/x-raw, width=(int)728, height=(int)544, framerate=(fraction)1/1, format=(string)GRAY8
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstAravis:aravis0:
streaming stopped, reason not-negotiated (-4)
EOS on shutdown enabled -- waiting for EOS after Error
Waiting for EOS...
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Interrupt while waiting for EOS - stopping pipeline...
Execution ended after 0:00:04.781750406
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...