AravisProject/aravis #670: meson test failed on ppc64 architecture

matwey, 2022-05-09:
**Describe the bug**
When aravis is compiled at `pcc64` the tests are failed as the following:

```
[   96s] + /usr/bin/meson test -C ppc64-suse-linux --num-processes 8 --print-errorlogs
[   97s] ninja: Entering directory `/home/abuild/rpmbuild/BUILD/aravis-0.8.21/ppc64-suse-linux'
[   97s] ninja: no work to do.
[   97s] 1/9 aravis:main / evaluator      OK              0.13s
[   97s] 2/9 aravis:main / buffer         OK              0.12s
[   97s] 3/9 aravis:main / misc           OK              0.11s
[   97s] 4/9 aravis:main / dom            OK              0.09s
[   97s] 5/9 aravis:main / genicam        FAIL            0.19s   killed by signal 6 SIGABRT
[   97s] >>> /home/abuild/rpmbuild/BUILD/aravis-0.8.21/ppc64-suse-linux/tests/genicam
[   97s] ――――――――――――――――――――――――――――――――――――― ✀  ―――――――――――――――――――――――――――――――――――――
[   97s] stdout:
[   97s] # random seed: R02S6327792775a67f8047456bbedd27eb22
[   97s] 1..16
[   97s] # Start of genicam tests
[   97s] ok 1 /genicam/value-type
[   97s] ok 2 /genicam/integer
[   97s] ok 3 /genicam/boolean
[   97s] ok 4 /genicam/float
[   97s] ok 5 /genicam/enumeration
[   97s] ok 6 /genicam/swissknife
[   97s] ok 7 /genicam/converter
[   97s] ok 8 /genicam/register
[   97s] ok 9 /genicam/string
[   97s] ok 10 /genicam/url
[   97s] ok 11 /genicam/mandatory
[   97s] Bail out! ERROR:../tests/genicam.c:1135:chunk_data_test: assertion failed (float_value == 1.1): (-1.54234871e-180 == 1.1)
[   97s] stderr:
[   97s] **
[   97s] ERROR:../tests/genicam.c:1135:chunk_data_test: assertion failed (float_value == 1.1): (-1.54234871e-180 == 1.1)
[   97s] ――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――――
[   97s] 
[   97s] 6/9 aravis:python / fake.py      OK              0.64s
[   98s] 7/9 aravis:main / fake           OK              1.12s
[   98s] 8/9 aravis:python / exception.py OK              1.18s
[  105s] 9/9 aravis:network / fakegv      OK              8.77s
[  105s] 
[  105s] Summary of Failures:
[  105s] 
[  105s] 5/9 aravis:main / genicam        FAIL            0.19s   killed by signal 6 SIGABRT
[  105s] 
[  105s] 
[  105s] Ok:                 8   
[  105s] Expected Fail:      0   
[  105s] Fail:               1   
[  105s] Unexpected Pass:    0   
[  105s] Skipped:            0   
[  105s] Timeout:            0   

```

**To Reproduce**
Compile aravis and run test as
```
/usr/bin/meson test
```

**Expected behavior**
Test are passed

**Camera description:**
N/A

**Platform description:**
 - 0.8.21
 - openSUSE Tumbleweed
 - ppc64

EmmanuelP, 2022-05-11:
Please test #673

matwey, 2022-05-12:
It works, thank you.

commit 277aab41 (2022-05-12): tests: fix ChunkFloat test on big endian platform

Fixes #670
closed 2022-05-12 by EmmanuelP